FreeCAD/FreeCAD #17021: TaskAttacher: Does not take into account the attaching object position

PaddleStroke, 2024-10-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If the object with the attachment extension is not at the same level than the attachment support, then currently the placement is wrong : 
[image]
(in the above example, Part has a non-identity placement which is not taken into account for the LCS object)

This is because the attachment task is not setting the attachment support correctly. It puts only the selection subObject instead of the full path.
This can be manually edited to be correct : 
[image]

The attachment task should take into account the position of the attaching object and of the attached object so that the placement is correct.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-05:
@FreeCAD/cad-advisory-group FYI

luzpaz, 2024-10-05:
Weird that the version info shows '0.22' ?

pierreporte, 2024-10-05:
Fixing this is definitely needed.

closed 2024-12-13 by chennes