FreeCAD/FreeCAD #17022: Part: Attachment engines do not support App::Links

PaddleStroke, 2024-10-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]
You can select a link in the view and task attacher is ok with that. But it offers you no mode than deactivated.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaiwas, 2024-10-05:
I don't know if we're talking about the same thing, but I have attached the link. 

https://github.com/user-attachments/assets/0d7acb32-fcc7-42b0-8c12-0b08210ca115

kaiwas, 2024-10-05:
But if we do it through the command, not through the properties - everything works as it should.

https://github.com/user-attachments/assets/7a0ed528-fb7f-44bd-9383-813e78ac14a0

closed 2024-12-13 by chennes