FreeCAD/FreeCAD #17023: Buttons under "New File" on the Start page have wrong color in dark mode

nilclass, 2024-10-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I just checked out 1.0.0RC2 - Love the dark mode! :heart:

Spotted this small issue on the start page:

[image]

The text is hardly legible, probably the background of those buttons should be dark as well?

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Assembly4 0.12.5
  * Help 1.0.3
  * fcgear 1.0.0
  * fasteners 0.4.24
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-05:
I think this is the same as: https://github.com/FreeCAD/FreeCAD/issues/15756
@MisterMakerNL FYI

closed 2024-10-05 by maxwxyz

MisterMakerNL, 2024-10-05:
This issue goes away after reboot, the real issue is that it doesn't get applied. 
This was fixed in previous start-page but came back in the revised start-page.