FreeCAD/FreeCAD #17025: File totally freeze

erwanyves, 2024-10-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[branches.zip](https://github.com/user-attachments/files/17266204/branches.zip)
While designint some parts, the file totally freezes and it is not possible to modify. Dependency graph seems to be correct. 

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * Assembly3.backup1707390127.209102 0.12.0
  * fasteners.backup1707390182.6606832 0.4.71
  * Nodes 0.1.36
  * sheetmetal.backup1708444688.6412804 0.4.1
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

erwanyves, 2024-10-05:
[modele gauche bug.zip](https://github.com/user-attachments/files/17267204/modele.gauche.bug.zip)
More precisely, this part freeze.

luzpaz, 2024-10-06:
@pieterhijma is this the multitransform bug that you were mentioning before ?

FlachyJoe, 2024-10-06:
The error is the DressUp features inherit of FeatureAddSub but don't compute an AddSubShape.
A DressUp can be additive and subtractive in the same time so it shouldn't be used with the current FeatureTransformed implementation.

erwanyves, 2024-10-08:
Yes, it is. I have rebuilt on a fresh file the same part, and problem occurs when the multitransform is applied on the part

maxwxyz, 2024-10-09:
does it work in the stable 0.21.2 version?

erwanyves, 2024-10-10:
I do not try it on the 0.21.2 version, sorry

FlachyJoe, 2024-10-10:
If you transforms the _dent_ and next the _entrée dent_ in another multi-transform, it works.
Maybe related to boolean precision issues : https://github.com/FreeCAD/FreeCAD/pull/17119

pieterhijma, 2024-10-10:
> @pieterhijma is this the multitransform bug that you were mentioning before ?

Seems to be unrelated. FreeCAD/OCC seem to hit something that is very compute-intensive. 

A couple of stacktraces on `branches.FCStd` from the compute-intensive part:

<details>
<summary>stacktrace 1 branches.FCStd</summary>
<pre>
#0  __GI___lll_lock_wake (futex=futex@entry=0x7fffa45fea40, private=<optimized out>) at lowlevellock.c:64
#1  0x00007fffef6a8452 in lll_mutex_unlock_optimized (mutex=0x7fffa45fea40) at pthread_mutex_unlock.c:43
#2  __GI___pthread_mutex_unlock_usercnt (mutex=0x7fffa45fea40, decr=<optimized out>) at pthread_mutex_unlock.c:68
#3  0x00007fffa6339e79 in ?? () from /usr/lib/libTKBO.so.7.8
#4  0x00007fffa63433f7 in ?? () from /usr/lib/libTKBO.so.7.8
#5  0x00007fffa44aa517 in OSD_ThreadPool::performJob(opencascade::handle<Standard_Failure>&, OSD_ThreadPool::JobInterface*, int) () from /usr/lib/libTKernel.so.7.8
#6  0x00007fffa44aaf3b in OSD_ThreadPool::Launcher::run(OSD_ThreadPool::JobInterface&) () from /usr/lib/libTKernel.so.7.8
#7  0x00007fffa44aaf86 in OSD_ThreadPool::Launcher::perform(OSD_ThreadPool::JobInterface&) () from /usr/lib/libTKernel.so.7.8
#8  0x00007fffa63b2e75 in ?? () from /usr/lib/libTKBO.so.7.8
#9  0x00007fffa6341c1e in BOPAlgo_PaveFiller::PerformEE(Message_ProgressRange const&) () from /usr/lib/libTKBO.so.7.8
#10 0x00007fffa6336180 in BOPAlgo_PaveFiller::PerformInternal(Message_ProgressRange const&) () from /usr/lib/libTKBO.so.7.8
#11 0x00007fffa632da43 in BOPAlgo_PaveFiller::Perform(Message_ProgressRange const&) () from /usr/lib/libTKBO.so.7.8
#12 0x00007fffa62d5d15 in BRepAlgoAPI_BuilderAlgo::IntersectShapes(NCollection_List<TopoDS_Shape> const&, Message_ProgressRange const&) () from /usr/lib/libTKBO.so.7.8
#13 0x00007fffa62d62dd in BRepAlgoAPI_BooleanOperation::Build(Message_ProgressRange const&) () from /usr/lib/libTKBO.so.7.8
#14 0x00007fffa7c3e2d0 in Part::TopoShape::makeElementBoolean (this=0x7fffffff9b30, maker=0x7fffa7d0160a "CUT", shapes=std::vector of length 68, capacity 128 = {...}, op=0x7fffa7d0160a "CUT", tolerance=0) at /home/pieter/current/fpa/freecad-main/src/Mod/Part/App/TopoShapeExpansion.cpp:5743
#15 0x00007fffa7c2f231 in Part::TopoShape::makeElementCut (this=0x7fffffff9b30, shapes=std::vector of length 68, capacity 128 = {...}, op=0x0, tol=0) at /home/pieter/current/fpa/freecad-main/src/Mod/Part/App/TopoShapeExpansion.cpp:4098
#16 0x00007fff44694199 in PartDesign::Transformed::execute (this=0x55555ee13b00) at /home/pieter/current/fpa/freecad-main/src/Mod/PartDesign/App/FeatureTransformed.cpp:337
#17 0x00007ffff3c7bd86 in App::DocumentObject::recompute (this=0x55555ee13b00) at /home/pieter/current/fpa/freecad-main/src/App/DocumentObject.cpp:162
#18 0x00007fffa7899a5b in Part::Feature::recompute (this=0x55555ee13b00) at /home/pieter/current/fpa/freecad-main/src/Mod/Part/App/PartFeature.cpp:106
#19 0x00007fff4462986a in PartDesign::Feature::recompute (this=0x55555ee13b00) at /home/pieter/current/fpa/freecad-main/src/Mod/PartDesign/App/Feature.cpp:72
</pre>
</details>

<details>
<summary>stacktrace 2 branches.FCStd</summary>
<pre>
#0  0x00007fffa6345b4f in BOPAlgo_PaveFiller::ReduceIntersectionRange(int, int, int, int, double&, double&) () from /usr/lib/libTKBO.so.7.8
#1  0x00007fffa634b632 in BOPAlgo_PaveFiller::PerformEF(Message_ProgressRange const&) () from /usr/lib/libTKBO.so.7.8
#2  0x00007fffa6336240 in BOPAlgo_PaveFiller::PerformInternal(Message_ProgressRange const&) () from /usr/lib/libTKBO.so.7.8
#3  0x00007fffa632da43 in BOPAlgo_PaveFiller::Perform(Message_ProgressRange const&) () from /usr/lib/libTKBO.so.7.8
#4  0x00007fffa62d5d15 in BRepAlgoAPI_BuilderAlgo::IntersectShapes(NCollection_List<TopoDS_Shape> const&, Message_ProgressRange const&) () from /usr/lib/libTKBO.so.7.8
#5  0x00007fffa62d62dd in BRepAlgoAPI_BooleanOperation::Build(Message_ProgressRange const&) () from /usr/lib/libTKBO.so.7.8
#6  0x00007fffa7c3e2d0 in Part::TopoShape::makeElementBoolean (this=0x7fffffff9b30, maker=0x7fffa7d0160a "CUT", shapes=std::vector of length 68, capacity 128 = {...}, op=0x7fffa7d0160a "CUT", tolerance=0) at /home/pieter/current/fpa/freecad-main/src/Mod/Part/App/TopoShapeExpansion.cpp:5743
#7  0x00007fffa7c2f231 in Part::TopoShape::makeElementCut (this=0x7fffffff9b30, shapes=std::vector of length 68, capacity 128 = {...}, op=0x0, tol=0) at /home/pieter/current/fpa/freecad-main/src/Mod/Part/App/TopoShapeExpansion.cpp:4098
#8  0x00007fff44694199 in PartDesign::Transformed::execute (this=0x55555ee13b00) at /home/pieter/current/fpa/freecad-main/src/Mod/PartDesign/App/FeatureTransformed.cpp:337
#9  0x00007ffff3c7bd86 in App::DocumentObject::recompute (this=0x55555ee13b00) at /home/pieter/current/fpa/freecad-main/src/App/DocumentObject.cpp:162
#10 0x00007fffa7899a5b in Part::Feature::recompute (this=0x55555ee13b00) at /home/pieter/current/fpa/freecad-main/src/Mod/Part/App/PartFeature.cpp:106
#11 0x00007fff4462986a in PartDesign::Feature::recompute (this=0x55555ee13b00) at /home/pieter/current/fpa/freecad-main/src/Mod/PartDesign/App/Feature.cpp:72
</pre>
</details>

<details>
<summary>stacktrace 3 branches.FCStd</summary>
<pre>
#0  0x00007fffa635f11a in BOPAlgo_PaveFiller::GetEFPnts(int, int, NCollection_List<IntSurf_PntOn2S>&) () from /usr/lib/libTKBO.so.7.8
#1  0x00007fffa634f968 in BOPAlgo_PaveFiller::PerformFF(Message_ProgressRange const&) () from /usr/lib/libTKBO.so.7.8
[... cut ...]
#9  0x00007ffff3c7bd86 in App::DocumentObject::recompute (this=0x55555ee13b00) at /home/pieter/current/fpa/freecad-main/src/App/DocumentObject.cpp:162
#10 0x00007fffa7899a5b in Part::Feature::recompute (this=0x55555ee13b00) at /home/pieter/current/fpa/freecad-main/src/Mod/Part/App/PartFeature.cpp:106
#11 0x00007fff4462986a in PartDesign::Feature::recompute (this=0x55555ee13b00) at /home/pieter/current/fpa/freecad-main/src/Mod/PartDesign/App/Feature.cpp:72
</pre>
</details>

The document object for this last one is: `"branches#MultiTransform"`.

erwanyves, 2024-10-10:
> #0  __GI___lll_lock_wake ***@***.***=0x7fffa45fea40, private=) at lowlevellock.c:64
> #2  __GI___pthread_mutex_unlock_usercnt (mutex=0x7fffa45fea40, decr=) at pthread_mutex_unlock.c:68
> #5  0x00007fffa44aa517 in OSD_ThreadPool::performJob(opencascade::handle&, OSD_ThreadPool::JobInterface*, int) () from /usr/lib/libTKernel.so.7.8
> #12 0x00007fffa62d5d15 in BRepAlgoAPI_BuilderAlgo::IntersectShapes(NCollection_List const&, Message_ProgressRange const&) () from /usr/lib/libTKBO.so.7.8
> #4  0x00007fffa62d5d15 in BRepAlgoAPI_BuilderAlgo::IntersectShapes(NCollection_List const&, Message_ProgressRange const&) () from /usr/lib/libTKBO.so.7.8
> #0  0x00007fffa635f11a in BOPAlgo_PaveFiller::GetEFPnts(int, int, NCollection_List&) () from /usr/lib/libTKBO.so.7.8

I have rebuilt the part,  do the multitransform only with dent and replace
the draft by a substractive operation.  The main problem was I was unable
to edit the previous file without crash.

FlachyJoe, 2024-10-10:
> I was unable to edit the previous file without crash.

If you can open it, right-click the document object in the tree view and check Skip Recompute then delete the faulty feature and uncheck.

erwanyves, 2024-10-10:
Thank you for the trick!!

luzpaz, 2024-11-23:
This ticket needs to be renamed. Proposals ?

closed 2025-08-16 by github-actions[bot]