FreeCAD/FreeCAD #17026: SIGSEGV when trying to launch FreeCAD 1.0 rc2 app image

straszydlo, 2024-10-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Whenever I try to launch the FreeCAD app image, I get SIGSEGV and the app crashes.

Image used: `FreeCAD_1.0.0RC2-conda-Linux-x86_64-py311.AppImage`

System info:
```
System:
  Kernel: 6.5.0-41-generic x86_64 bits: 64 compiler: N/A Desktop: Cinnamon 5.8.4 tk: GTK 3.24.33
    wm: muffin dm: LightDM Distro: Linux Mint 21.2 Victoria base: Ubuntu 22.04 jammy
CPU:
  Info: 6-core model: 11th Gen Intel Core i5-11400H bits: 64 type: MT MCP arch: Tiger Lake rev: 1
    cache: L1: 480 KiB L2: 7.5 MiB L3: 12 MiB
Graphics:
  Device-1: Intel TigerLake-H GT1 [UHD Graphics] vendor: Acer Incorporated ALI driver: i915
    v: kernel ports: active: eDP-1 empty: DP-1,DP-2 bus-ID: 00:02.0 chip-ID: 8086:9a68
  OpenGL: renderer: Mesa Intel UHD Graphics (TGL GT1) v: 4.6 Mesa 23.2.1-1ubuntu3.1~22.04.2
    direct render: Yes
```

Full log:
[freecad_1_rc2.log](https://github.com/user-attachments/files/17266286/freecad_1_rc2.log)

### Full version info

```shell
FreeCAD 1.0.0 Revision: 38806 (Git)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-10-05:
Could apply the _About_ info of the version that works and try the latest version? Quite possible that you have an outdated workbench or a macro installed.

If so, the solution is to delete the configuration and try from scratch.

straszydlo, 2024-10-05:
@xtemp09 you were right, I had an outdated workbench. After applying the updates the rc2 image works like a charm. Thanks a lot!

closed 2024-10-05 by straszydlo