FreeCAD/FreeCAD #17027: Crashes instantly when try to ESC the 2D polyline draft

andzejsp, 2024-10-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Start drafting in 2D with poly line, press left mouse on the draft region to draw stuff, then press right click then press ESC and it crashes.

Demonstration
https://streamable.com/totept
https://github.com/user-attachments/assets/d9e1b7ae-96f3-4c9a-8220-3d85663a164b

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Assembly4 0.50.2
  * Glass
  * IconThemes
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-10-06:
Confirmed. Note that dragging the mouse while pressing Esc is the cause of the probe,

FEA-eng, 2024-10-06:
@maxwxyz Looks serious (especially since the crash occurs even on Windows) and is easy to reproduce. Blocker?

maxwxyz, 2024-10-06:
sure

AIRCAP, 2024-10-06:
i have not managed to reproduce this on ubuntu linux - i tried to follow the video closely -  could this be a windows specific issue?

andzejsp, 2024-10-07:
> i have **not** managed to reproduce this on ubuntu linux - i tried to follow the video closely - could this be a windows specific issue?
> 
> ```
> OS: Ubuntu 20.04.6 LTS (XFCE/xfce)
> Word size of FreeCAD: 64-bit
> Version: 1.0.0RC2.38806 (Git) AppImage
> Build type: Release
> Branch: (HEAD detached at 1.0rc2)
> Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
> Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
> Locale: English/United States (en_US)
> Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
> Installed mods: 
>   * CurvedShapes 1.0.10
>   * CfdOF 1.27.10
>   * Curves 0.6.48
>   * AirPlaneDesign 0.4.1
> ```
> 
> do you hold the mouse button down while dragging or is it a short click? if i click short, the right-click-menu appears, and ESC closes the menu, then i can continue drawing the polyline if i right click and hit ESC while the button is still clicked, then it will end the polyline, and only then the right click menu appears with some delay. but no crash in either case

well.. today i booted up my machine and tried to replicate and i cant replicate this no more.. seems like it was maybe afterinstall bug? Maybe some kind of dependency didnt load?

Or maybe i need to warm up my machine and see if it still persists later. Will test it after few hours later.

But the sequence was this > left click lots of times then right click and release and then move mouse off that mouse menu and press ESC.

Roy-043, 2024-10-07:
Reproducing can be tricky:
1. The task panel must be docked.
2. Start the Draft_Wire command.
3. Click a few points.
4. Right-click to bring up the context menu.
5. Move the cursor to the very left edge of the menu (so that none of the menu options is highlighted).
6. Make a fast move with the mouse the left. Do not hold down mouse buttons, but press Esc during the move.

I have meanwhile found the cause and will submit a PR to fix this.

zbynekwinkler, 2024-10-07:
It is very tricky to reproduce indeed. But finally I could reproduce it as well. What I don't understand is why it crashes - there is a guard for nonnull `ActiveDialog`. It is set to null when the reject is called for the first time. It seems it should survive repeated calls to `reject` thus the fix in #17064 should not be needed.

[image]

Roy-043, 2024-10-07:
@zbynekwinkler Thanks for your comment. You have given me an idea for an  improvement of my PR.

This is not just about the task panel. There are 3D view callbacks that have to be ended. Are you aware of that?

zbynekwinkler, 2024-10-07:
@Roy-043 I am just really starting orient myself in the code base so I am taking the opportunity to learn while may be helping to fix some crashes.

PR #17064 (2024-10-07): Draft: Stop draftToolBar.escape() executing multiple times
Fixes #17027.

The `DraftTaskPanel.reject()` function can be called multiple times if Esc is pressed during a fast mouse move. We need to prevent multiple calls to `draftToolBar.escape()` as this will lead to multiple calls to `gui_utils.end_all_events()` which results in a crash.
commit f92ecfef (2024-10-07): Draft: Stop draftToolBar.escape() executing multiple times

Fixes #17027.

The `DraftTaskPanel.reject()` function can be called multiple times if Esc is pressed during a fast mouse move. We need to prevent multiple calls to `draftToolBar.escape()` as this will lead to multiple calls to `gui_utils.end_all_events()` which results in a crash.
closed 2024-10-07 by yorikvanhavre