FreeCAD/FreeCAD #17028: Sketcher construction lines disappear

alfrix, 2024-10-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Happens on X11 and Wayland, and on a system with QT5 (Debian) and one with QT6 (Manjaro)
The video shows 1.0 RC2, but it happened on RC1 too

https://github.com/user-attachments/assets/be6cc7f7-0a48-4f21-a697-09f7ccf979ad

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Argentina (es_AR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CurvedShapes 1.0.9
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-05:
@benj5378 you've had a Wayland issue in spreadsheet lately, can you reproduce?
@alfrix does it work in the stable 0.21.2?

benj5378, 2024-10-05:
Sounds a bit weird that it should happen on X11, then we would have tons of bug reports already. For Wayland, it sounds very similar to https://github.com/coin3d/coin/pull/531#issuecomment-2385726371.

> However, if one tries to create a rectangle in FreeCAD Sketcher, SIGSEGV happen.
> 
> Vertices and distance constraints work fine, but arcs, rectangles and lines causes a crash.

I want to see the terminal output/report view, do we get bunch of errors? I got
```
Coin error in SoGroup::removeChild(): tried to remove non-existent child 0x561a5d7621f0 (Annotation)
```

at91rm9200, 2024-10-05:
Hello,

I can confirm that there are issues with the construction lines. In the sketcher, I first created a rectangle in the sketch. Then I changed the geometry to construction mode and draw another rectangle. In the screencast you can see, that there are no lines visible around the rectangle.

Regards, Bernd.
```
OS: Ubuntu 18.04.4 LTS (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

https://github.com/user-attachments/assets/bccac842-9804-4028-ad79-92ed9525a2cf

benj5378, 2024-10-05:
@at91rm9200, are you on X11 or Wayland? `echo $XDG_SESSION_TYPE` in terminal

at91rm9200, 2024-10-05:
I am on X11.

benj5378, 2024-10-05:
In X11, so weird. I can't reproduce it on my X11 device... Does Coin version play a role?

@at91rm9200 and @alfrix that could reproduce runs Coin 4.0.3. My Wayland device having a similar issue runs Coin 4.0.3.

My X11 device where I cannot reproduce runs Coin 4.0.0

@maxwxyz, blocker?

alfrix, 2024-10-06:
> @alfrix does it work in the stable 0.21.2?

0.21 works correctly

other platforms that i have tested show the same issue:
```
OS: Manjaro Linux (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: makepkg
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: Spanish/Argentina (es_AR)
Installed mods: 
  * Curves 0.3.0
  * ThreadProfile 1.81.0
  * CurvedShapes 1.0.5
```

also same issue under X11:

```
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: makepkg
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: Spanish/Argentina (es_AR)
Installed mods: 
  * Curves 0.3.0
  * ThreadProfile 1.81.0
  * CurvedShapes 1.0.5

```

so i don't think it's something particular with any distro since Manjaro is Arch based and it's quite different to Debian
Or QT since it happens on both 5 and 6
Or Wayland, as it happens on X11 too

> How did you test for X11?
@benj5378 log out and select X11 on the SDDM greeter or equivalent lockscreen

hyarion, 2024-10-06:
I've tested RC2 on macOS, which uses coin 4.0.3 and I don't see this bug.
```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * A2plus 0.4.68
  * lattice2 1.0.0
  * CurvedShapes 1.0.10
  * sheetmetal 0.4.26
  * Curves 0.6.6
```

benj5378, 2024-10-06:
> I've tested RC2 on macOS, which uses coin 4.0.3 and I don't see this bug.

Hmmm, combination of Linux and Coin 4.0.3?

at91rm9200, 2024-10-06:
Hello,

I checked RC1, which shows the problem on my desktop computer too:
```
OS: Ubuntu 18.04.4 LTS (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```
The latest stable version 0.21.2 on my desktop computer is ok:
```
OS: Ubuntu 18.04.4 LTS (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
```
Then I checked RC1 and RC2 on my laptop computer and both versions do not show the problem:
```
OS: Ubuntu 18.04.6 LTS (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```
The difference between my desktop computer and the laptop computer is the Linux version:
Desktop: Ubuntu 18.04.4 LTS (MATE/mate)
Laptop: Ubuntu 18.04.6 LTS (MATE/mate)

Perhaps another minor detail: The construction lines in the RC-versions are dashed, whereas they were solid in the stable version.

Regards, Bernd.

benj5378, 2024-10-06:
I just spend ages compiling Coin 4.0.3 on Linux. I can't reproduce, so that doesn't seem to be the error...

maxwxyz, 2024-10-06:
I'll label it as blocker for now so the issue can be discussed at least.

benj5378, 2024-10-07:
Related to #14414?

Syres916, 2024-10-07:
IMHO, quite simply FreeCAD 1.0 shouldn't ship with the dashed lines _enabled_ as default (for any of the different geometries) until there's a reliable way (possibly in the First Time Start Widget) to determine if the user's setup is compatible with them. I'm a little surprised @obelisk79 hasn't raised it as I can replicate using EndeavourOS

```
OS: EndeavourOS (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38881 (Git)
Build type: Release
Branch: main
Hash: ab44bd1f7ba3c3008f3fd8ef6bdb61d11d1a82e1
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/breeze
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * FreeCAD-themes 2024.7.24
```

benj5378, 2024-10-08:
Very unlikely, but related to Pivy? Please do `sudo apt remove python3-pivy` and check if the issue persists.

Syres916, 2024-10-09:
@benj5378 I tried `sudo pacman -Rcns python-pivy` on EndeavourOS and it made no difference other than adding a couple of warnings entering Sketcher Edit Mode so I reverted the change.

benj5378, 2024-10-09:
> I checked RC1, which shows the problem on my desktop computer too:
> Then I checked RC1 and RC2 on my laptop computer and both versions do not show the problem

@at91rm9200, so you did not compile yourself on any of the devices? You used the Conda build? Because if you used Conda build, and this issues shows on some devices, and others not... then I'm honestly as lost as I can be

benj5378, 2024-10-09:
As far as my knowledge, no color for hovering Sketcher geometry can be set in Preferences. Never have I changed it. Still somehow, right now when testing, the color when hovering was blue... after resetting preferences, they were the normal white color...

Very unlikely, but being lost and just wanting to rule everything out... can someone remove their preference pack and settings by resetting all their preferences?

at91rm9200, 2024-10-10:
I always used the provided AppImages:
```
FreeCAD-0.21.2-Linux-x86_64.AppImage.
FreeCAD_1.0.0RC1-conda-Linux-x86_64-py311.AppImage. 
FreeCAD_1.0.0RC2-conda-Linux-x86_64-py311.AppImage 
```

> Please do sudo apt remove python3-pivy and check if the issue persists.

I'm afraid of breaking something on the machines.

Syres916, 2024-10-10:
> As far as my knowledge, no color for hovering Sketcher geometry can be set in Preferences.

It can definitely be set in Preferences, the parameter is _BaseApp/Preferences/View/DeactivatedConstrDimColor_ and can be found in `Preferences` > `Sketcher` > `Appearance` > `Constraint colors` > `Deactivated constraint` , it's 127,127,127 by default as per https://github.com/FreeCAD/FreeCAD/pull/14763/files

Syres916, 2024-10-10:
@benj5378 I think in this instance a picture would have helped clarify what you are getting at, if I now understand you are talking about Preselect Color https://wiki.freecad.org/Preferences_Editor#Selection which is a yellow like color by default and in FreeCAD Dark changes to a blue color. I'm still completely lost as to what this has got to do with the dashed lines not showing on certain distros but I'll leave it with you.

benj5378, 2024-10-10:
I'm still completely lost as to what this has got to do with the dashed lines not showing on certain distros but I'll leave it with you.

@Syres916, say, for some reason, there was a bug such that the color in preferences was changed to transparent. Would explain why it's only seen on some devices

alfrix, 2024-10-10:
found the cause, it's commit b140feabaf78e6f384df182bcf87c109eb749cef

bisecting was tedious, i had to constantly reapply the compile fixes for python 3.12 and the sequencer issue, and each test took more than an hour to build even with most of the workbenches disabled

benj5378, 2024-10-10:
@alfrix, thank you for your investigation! Do you want to make patch, or should I? I would have to make you test some different patches and debugging, as I can't reproduce the issue myself.

PaddleStroke, 2024-10-11:
This commit is the one that introduced the dashed lines in sketcher.
I cannot reproduce this error either so I can't do much to find the issue.
Note that other recent bugs seemed to be linked to Coin & linux : https://github.com/FreeCAD/FreeCAD/issues/15706
So there might be some larger problem with some version of coin + linux ? @wwmayer does that ring any bell to you?

wwmayer, 2024-10-11:
> does that ring any bell to you?

No, it doesn't.

alfrix, 2024-10-11:
i fixed it with a bunch of reverts: https://github.com/alfrix/FreeCAD/commit/421c26c1b735861dfefb668f1a7db74c1c1a6474

i would assume it's probably something with the sublayer system

benj5378, 2024-10-11:
@alfrix, are you on Discord? Would help making a patch easier. Please tag me in #development with \@bensay on FreeCAD Discord server

Footleg, 2024-10-15:
I have found in some of my sketches that external geometry lines stop being drawn when I go back into a sketch to edit it after making other edits in my parts. The lines still existing as far as the sketcher is concerned, but are not longer rendered and so I cannot select them or generate new constraints using them. This was on Windows 10 with 1.0 RC2. I have not worked out how to reproduce it, but have seen it in a couple of different parts now. I am wondering if this is the same issue? Is there a nightly I can test against my case where this issue has been addressed?

alfrix, 2024-10-16:
> I have found in some of my sketches that external geometry lines stop being drawn when I go back into a sketch to edit it after making other edits in my parts. The lines still existing as far as the [...]

if it's the same issue then setting all the lines as solid instead of dashed fixes the issue, like this:
[image]

i've marked as needs restart but closing edit mode and re-entering should suffice, please report back if it fixes the issue for you.

hyarion, 2024-10-16:
This smells like a draw call optimization issue or maybe a driver issue.

* What happens if the view is rotated?
* Are they only missing in some orientations?
* What happens after they go missing if you remove the normal white lines, do they appear again? 

I think the next step is to use RenderDoc (or similar tool) to capture a frame where the lines has disappeared and see if coin tries to render the geometry at all in the frame.

Footleg, 2024-10-16:
[Plate Bracket.zip](https://github.com/user-attachments/files/17402070/Plate.Bracket.zip)
Changing the appearance to a solid line style did not make the sketch render. I have attached my part so others can investigate. Open Sketch003 and see if you see the sketch lines. I don't on Windows 10. It appears that the part is being drawn over the sketch, hiding all the lines in it.

obelisk79, 2024-10-16:
I am unable to reproduce this behavior. It is likely a dependency version issue of some sort. Tested dev  branch and RC2 appimage.

OS: EndeavourOS (KDE/plasma/wayland)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38984 (Git)
Build type: Release
Branch: main
Hash: 9806a463246f5a0daf5d87b323ab8c06a97feba2
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Installed mods: 
  * Curves 0.6.47
  * OpenTheme 2024.9.1
  * sheetmetal 0.4.25

obelisk79, 2024-10-16:
Also, to help rule out possible driver issues, it may be worth enabling this preference, restarting and retesting for the behavior:
[image]

obelisk79, 2024-10-16:
@Footleg I think in your case you need to enable the view section function to let your sketch geometry not be hidden by the model as I see the lines just fine. I think you'll also likely want to enable the preference shown in the image below to avoid having to toggle this view frequently:
[image]
[image]

Footleg, 2024-10-16:
OK, my case appears to be that I needed to use section view mode (a feature I have not come across before in FreeCAD). So I think we can rule out my case as 'not a bug'. Thanks for the detailed information to help me understand it.

kadet1090, 2024-10-16:
So, from @hyarion dive into coin it seems that it might be caused by the `glLineStipple` call that was deprecated in OpenGL 3.1. From quick glance at docs it also seems that `glLineWidth` is not guaranteed to support anything else than 1:
 
> Not all widths can be supported when line antialiasing is enabled. If an unsupported width is requested, the nearest supported width is used. Only width 1 is guaranteed to be supported; others depend on the implementation. Likewise, there is a range for aliased line widths as well. To query the range of supported widths and the size difference between supported widths within the range, call glGet with arguments GL_ALIASED_LINE_WIDTH_RANGE, GL_SMOOTH_LINE_WIDTH_RANGE, and GL_SMOOTH_LINE_WIDTH_GRANULARITY. 

While I highly doubt that it is common to not support drivers are probably not tested with such things in mind and so can contain bugs. For please check if the issue is reproducible on software OpenGL (https://github.com/FreeCAD/FreeCAD/issues/17028#issuecomment-2417924454). Please also attach driver information so we can check if this is driver issue. I don't experience such issue on nvidia propertiary drivers.

#### What we do next?
The call is deprecated, and so is probably like bigger half of coin. For 1.1 we should update Coin3D to not use such deprecated features and use for example shader based implementation of line rendering, like one done here: https://stackoverflow.com/questions/6017176/gllinestipple-deprecated-in-opengl-3-1

alfrix, 2024-10-17:
I've tried with Software OpenGL, and construction lines get hidden
All the AntiAliasing options including None, construction get hidden
line widths set to 1, hidden
transparent objects to Back-face, hidden

it will only start working if you set the line to solid

also the "reset preferences" on the corner doesn't reset the line type
and line smoothing AA, seems very broken in my machine not rendering transparency correctly

i'm using the appimage of RC2 for testing
```
OS: Debian GNU/Linux trixie/sid (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Argentina (es_AR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CurvedShapes 1.0.9
```
driver is the amd r600 still maintained by Gert Wollny
```
OpenGL core profile version string: 4.5 (Core Profile) Mesa 24.2.4-1
OpenGL core profile shading language version string: 4.50
OpenGL version string: 4.5 (Compatibility Profile) Mesa 24.2.4-1
OpenGL shading language version string: 4.50
```

for those that do not have discord, i've found that doing this:
```diff
--- a/src/Mod/Sketcher/Gui/EditModeGeometryCoinManager.cpp
+++ b/src/Mod/Sketcher/Gui/EditModeGeometryCoinManager.cpp
@@ -665,13 +665,13 @@ void EditModeGeometryCoinManager::createEditModeCurveInventorNodes()
             sep->addChild(editModeScenegraphNodes.CurvesCoordinate[i][t]);
 
             if (geometryLayerParameters.isConstructionSubLayer(t)) {
-                sep->addChild(editModeScenegraphNodes.CurvesConstructionDrawStyle);
+                sep->addChild(editModeScenegraphNodes.CurvesDrawStyle);
             }
             else if (geometryLayerParameters.isInternalSubLayer(t)) {
-                sep->addChild(editModeScenegraphNodes.CurvesInternalDrawStyle);
+                sep->addChild(editModeScenegraphNodes.CurvesDrawStyle);
             }
             else if (geometryLayerParameters.isExternalSubLayer(t)) {
-                sep->addChild(editModeScenegraphNodes.CurvesExternalDrawStyle);
+                sep->addChild(editModeScenegraphNodes.CurvesDrawStyle);
             }
             else {
                 sep->addChild(editModeScenegraphNodes.CurvesDrawStyle);
```

makes the construction and other lines work even if dashed, of course you lose the ability to set them with different settings.

hyarion, 2024-10-19:
According to Qt's documentation, setting software opengl may not change the driver. If qt fails to find a software driver, it will still draw in hardware accelerated mode.

I think we need to get a frame recorded using renderdoc or similar to continue investigating this issue. I can assist on a video link if needed.

benj5378, 2024-10-19:
> According to Qt's documentation, setting software opengl may not change the driver. If qt fails to find a software driver, it will still draw in hardware accelerated mode.

Might be possible to use [QOpenGLFunctions::glGetString](https://doc.qt.io/qt-6/qopenglfunctions.html#glGetString) to check driver. https://stackoverflow.com/a/36336632

alfrix, 2024-10-24:
i have tried a lot of things inside freecad code without any luck

i've tried to run renderdoc, (the appimage needs to be extracted), but it complains about only working with core profiles, but enabling core profile doesn't seem to work? maybe freecad needs compatibility mode, that would make renderdoc not usable for this case, idk for sure not an expert on this

@Syres916 since you are affected, can you share what GPU are you using, maybe this is an AMD thing, or directly an issue with the r600g driver (HD3xxx to HD6xxx)

hyarion, 2024-11-02:
So what are we doing about this issue? I see two possibilities:
* We can disable dashed lines for mesa amd r600g drivers
* Decide that this card as too uncommon/old to add a workaround for

The workaround seems to change all dashed styles to be continuous lines instead.

What do you think?

alfrix, 2024-12-10:
BTW opened an issue in mesa:
https://gitlab.freedesktop.org/mesa/mesa/-/issues/12106

gitdocke, 2025-01-21:
I'm running Freecad v1.0 Flatpak in Manjaro KDE and having the same problem with invisible construction geometry.
But, I discovered a particularity of this bug that may help the devs to fix it.
The construction geometry is shown fine as long as no regular geometry is introduced, or it is already present in the sketch. In other words, a mix in the sketch of both geometries will turn invisible the construction one.
It might be this is not about video drivers but tinkering with layers.

hyarion, 2025-01-21:
@gitdocke may I ask which gpu you use?

gitdocke, 2025-01-21:
@hyarion 
Sure.

[image]

gitdocke, 2025-02-05:
Reporting.
I just updated Manjaro. Among the many new packages there is a Mesa one. So, I started FreeCAD to see if the construction geometry problem went away and.... voila! It is fixed indeed.
Screenshots speak by themselves.
Thanks devs for attention and patience.

[image]

[image]

benj5378, 2025-02-05:
@gitdocke you have the same old graphics card as all the others also experiencing this issue. Can you send a link to the new meta package? I want to see its changelog

gitdocke, 2025-02-05:
@benj5378 
I have no idea how to get that link and I'm not sure if what you refer to as metapackage is the info shown in the screenshots, if so, i can prepare screenshots showing the full lists.

[image]

[image]

[image]

benj5378, 2025-02-05:
If you downgrade the mesa package, does the issue reappear? @gitdocke

gitdocke, 2025-02-06:
@benj5378
I did it last night and everything went wrong.
Immediately after downgrading mesa I was unable to launch any program or even to restart the machine from the launcher menu (perhaps systemd was wrecked); so, I hard reset but when booting again, Plasma was failing to load and my computer was bricked.
To make it short, I'm back after a fresh Manjaro install and I regret to say, I will never ever downgrade a package again.
Sorry.

[image]

lorn10, 2025-02-06:
@gitdocke Just FYI, I also had my "hard lessons" with Kubuntu and different Mesa builds. However, there exist in most cases a (super) trivial resolution for problems in conjunction with broken graphics drivers. Just boot up the system to the command line (CLI) by adding a number "**3**" to the GRUB boot parameter. After that you should be able to login with username and password and **undo** the change you made to the installation. In my case, to go back to the default "stock" Mesa version I just had to purge / delete the installed Mesa devel PPA with `sudo ppa-purge ppa:oibaf/graphics-drivers`. This will revert all Mesa files back to the original state and also remove the PPA itself. That procedure helped me until now quite several times and I never had to reinstall the whole OS from scratch.

Whatever, this might be different on Manjaro and other non-Debian based distros but the underlying principle to fix the problem from the CLI should be the same. In this situation here you just had to figure out how to "upgrade" the Mesa build again from the terminal to the working version or to set everything Mesa related back to the default "stock" variant. 😉

alfrix, 2025-02-06:
I can confirm as Not fixed on 24.2.8 and 24.3.4

the 7xxx series gpu (souther islands), which should use the radeonsi driver and not r600g

benj5378, 2025-02-07:
@alfrix can you USB boot manjaro? Or @gitdocke USB boot debian with old mesa driver or something? Not so important, just curious...

alfrix, 2025-02-07:
> [@alfrix](https://github.com/alfrix) can you USB boot manjaro? Or [@gitdocke](https://github.com/gitdocke) USB boot debian with old mesa driver or something? Not so important, just curious...

24.3.4 was tested with manjaro
24.2.8 was tested with debian-testing (my main)

24.3.4 will be in debian testing in 2-3 days (if nothing bad is found that can hold it)

lorn10, 2025-02-07:
> the 7xxx series gpu (souther islands), which should use the radeonsi driver and not r600g

This correct for the 7**7**xx series Radeon HD GPU (and newer). However, the Radeon HD 7460G is still TeraScale based and uses the `r600` Mesa driver. 😉

gitdocke, 2025-02-08:
@lorn10 
Thanks. I'll figure the method out in Manjaro but I'm not much confident of success, I'm taking into account that the package manager did a very mickey mouse job when downgrading and I suspect it might be not much different when trying to fix a broken system is at hand.

@benj5378
All right, I did the following tests with the package "FreeCAD_weekly-builds-40041-conda-Linux-x86_64-py311.AppImage" and USB booting different Manjaro versions with also different mesa ones.
As follows:
1) manjaro-kde-22.0.1-230124-linux61.iso with mesa 22.3.3-3 ---> No problem
2) manjaro-kde-23.1.2-240102-linux66.iso with mesa 1:23.1.9-1 ---> No problem
3) manjaro-kde-24.0.3-240702-linux69.iso with mesa 1:24.1.1-1 ---> PROBLEM DETECTED (see pic)
4) manjaro-kde-24.1.2-241104-linux610.iso with mesa 1:24.2.4-0.1 ---> No problem
5) manjaro-kde-24.2.1-241216-linux612.iso with mesa 1:24.3.4-1 ---> No problem
  
[image]

FEA-eng, 2025-10-21:
@alfrix Can you confirm if this issue is still relevant in the newest available weekly build?

alfrix, 2025-10-23:
> @alfrix Can you confirm if this issue is still relevant in the newest available weekly build?

I don't have that machine on hand, but I don't think it would be fixed, since it's part of a driver that barely gets mantened (which is awesome that it still being patched considering how old is the hardware)

I think this issue can be closed since making a workaround inside FreeCAD code for a series of ancient cards, is not something I would do, nevertheless the users affected have a way to fix it, you just need to set all the lines to solid and not dashed