FreeCAD/FreeCAD #17031: Elements disappear when switching to strict IFC mode with only Building element present (and no site)

semhustej, 2024-10-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When there is Building object created in the file which does not contain a Site object, walls (and also other objects) disappear when users switches to strict IFC mode.

Steps to reproduce:
- open the file "building-IFC.FCstd" 
[ifc-building.zip](https://github.com/user-attachments/files/17266951/ifc-building.zip)

- switch to strict IFC mode
- walls disappear

When site object is created in the file, the conversion to strict IFC mode runs OK. When there is no building object in the file, the conversion to strict IFC mode runs OK.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: English/United States (en_US) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

BIM

### Anything else?

The issue has been confirmed by another Linux user:
https://forum.freecad.org/viewtopic.php?t=91178

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-06:
@yorikvanhavre FYI

luzpaz, 2024-10-06:
Per the forum thread, this was confirmed by @paullee0

yorikvanhavre, 2024-10-08:
Confirmed too. Having a deeper look...

semhustej, 2024-10-08:
> Confirmed too. Having a deeper look...
> 
> Ok. So what happens is this: The contents of the file are correctly converted to IFC and stored into the document. If you save the IFC file, you see the contents are there. The problem is, when switching to strict IFC, although the IFC data is created correctly, there is no object in the document to display it. I think here we should add some code to produce the site/building if needed after conversion. I'll prepare a PR

Great! I am not sure about that, but I think each "proper" (meaning good customs) IFC should have IfcSite and IfcBuildinga anyway. If FreeCAD creates that for the user, that is good.

yorikvanhavre, 2024-10-09:
It does, but you can switch that off in the preferences.

closed 2024-10-14 by yorikvanhavre

yorikvanhavre, 2024-11-18:
This should be addressed by https://github.com/FreeCAD/FreeCAD/pull/16948

PR #17093 (2024-10-14): BIM: Fixed missing object creation in IFC document conversion
Fixes #17031
Fixes #17032
commit 0b4510c9 (2024-10-14): BIM: Fixed missing object creation in IFC document conversion (#17093)

* BIM: Fixed missing object creation in IFC document conversion - fixes #17031

* BIM: fixed deleting of non-existing objects - fixes #17032
commit d82c2d6d (2024-10-14): BIM: Fixed missing object creation in IFC document conversion (#17093)

* BIM: Fixed missing object creation in IFC document conversion - fixes #17031

* BIM: fixed deleting of non-existing objects - fixes #17032