FreeCAD/FreeCAD #17034: PartDesign: Pad - Up to face Segmentation fault when selecting faces without assigned shape

Jarno-de-Wit, 2024-10-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating a Part Design pad using "up to shape", FreeCAD will SegFault if individual faces are selected (checkbox "Select all faces" = off), while no shape has been selected yet. I have verified this SegFault occurs both on Linux and MacOS in Release Candidate 2, as well as in the latest weekly build (tested on Linux only).

## How to reproduce:
1. Create an arbitrary solid, e.g. by creating a square pad
2. Create a new sketch
3. Open the pad dialog with this new sketch
4. Select "Up to shape" from the dropdown
5. Uncheck the "Select all faces" checkbox
6. Select any face on the existing solid

### Full version info

```shell
### System 1:
OS: macOS Mojave (10.14)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: C/Default (C) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus 0.4.65
  * Curves 0.6.40

### System 2:
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/unset/Qt default

### System 2 on latest weekly build:
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git) AppImage
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

### Subproject(s) affected?

PartDesign

### Anything else?

## Example file:
[PadSegfaultUpToShape.FCStd.zip](https://github.com/user-attachments/files/17267143/PadSegfaultUpToShape.FCStd.zip)
In this example file, execute steps 3-6 of the "How to reproduce" section to reproduce the steps.

## Terminal output of System 2 (Manjaro Linux) on RC2:
[LinuxSegfaultLog.txt](https://github.com/user-attachments/files/17267181/LinuxSegfaultLog.txt) (Also shown below)
```
Program received signal SIGSEGV, Segmentation fault.
[... 31 lines cut ...]
#31  0x7f6ba4e613eb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCAZQwBo6/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#32  /tmp/.mount_FreeCAZQwBo6/usr/bin/../lib/libQt5Widgets.so.5(+0x1c6e6f) [0x7f6ba4eb1e6f]
#33  /tmp/.mount_FreeCAZQwBo6/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca0c9) [0x7f6ba4eb50c9]
#34  0x7f6ba4e5b3ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAZQwBo6/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#35  0x7f6ba6794079 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAZQwBo6/usr/bin/../lib/libFreeCADGui.so+0x99
#36  0x7f6ba432eb28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAZQwBo6/usr/bin/../lib/libQt5Core.so.5+0x108
#37  0x7f6ba47bb3ec in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCAZQwBo6/usr/bin/../lib/libQt5Gui.so.5+0x84c
#38  0x7f6ba47910bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAZQwBo6/usr/bin/../lib/libQt5Gui.so.5+0xac
#39  /tmp/.mount_FreeCAZQwBo6/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x7f6b9e5abbea]
#40  /tmp/.mount_FreeCAZQwBo6/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x7f6ba030fad7]
#41  /tmp/.mount_FreeCAZQwBo6/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x7f6ba0312ac7]
#42  /tmp/.mount_FreeCAZQwBo6/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7f6ba03130ce]
#43  0x7f6ba4382a36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAZQwBo6/usr/bin/../lib/libQt5Core.so.5+0x66
#44  0x7f6ba432d70b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAZQwBo6/usr/bin/../lib/libQt5Core.so.5+0x11b
#45  0x7f6ba4335212 in QCoreApplication::exec() from /tmp/.mount_FreeCAZQwBo6/usr/bin/../lib/libQt5Core.so.5+0x82
#46  0x7f6ba66f34e1 in Gui::Application::runApplication() from /tmp/.mount_FreeCAZQwBo6/usr/bin/../lib/libFreeCADGui.so+0xa41
#47  /tmp/.mount_FreeCAZQwBo6/usr/bin/freecad(+0x789f) [0x55fd41d9d89f]
#48  /usr/lib/libc.so.6(+0x25e08) [0x7f6ba3c78e08]
#49  /usr/lib/libc.so.6(__libc_start_main+0x8c) [0x7f6ba3c78ecc]
#50  /tmp/.mount_FreeCAZQwBo6/usr/bin/freecad(+0x7c12) [0x55fd41d9dc12]
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2024-10-06:
@kadet1090 could you please take a look?

luzpaz, 2024-10-07:
@Jarno-de-Wit thank you for the very thorough report and testing 💪

FEA-eng, 2024-10-07:
It doesn't crash on Windows but there's:

```
19:49:42  <Exception> Access violation
19:49:42  0 Selection.cpp(156): Unhandled Base::Exception caught in selection observer: 
```

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

luzpaz, 2024-10-15:
cc @FlachyJoe

PR #17269 (2024-10-18): Fix uptoshape crash when selecting a face without assigned shape
Fix #17034
closed 2024-10-18 by chennes