FreeCAD/FreeCAD #17035: [1.0] Huge exception count when opening preferences window

shaise, 2024-10-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Not sure if its a bug or not:
When I debug freecad under Visual Studio, I get huge amount of exceptions (~12000 of them!) all are handled, yes, but I wonder if they can be avoided.
all these exceptions are from the following 2 functions:
```
bool Material::hasPhysicalProperty(const QString& name) const
{
    try {
        static_cast<void>(_physical.at(name));
    }
    catch (std::out_of_range const&) {
        return false;
    }
    return true;
}

bool Material::hasAppearanceProperty(const QString& name) const
{
    try {
        static_cast<void>(_appearance.at(name));
    }
    catch (std::out_of_range const&) {
        return false;
    }
    return true;
}
```
I wonder if there is a less harsh way of querying if there is a key in an std::map?
B.T.W: why are there so many callings to this function from the preference window?

shai

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 +5 (Git)
Build type: Release
Branch: effects3d
Hash: 98d761be7390f45801596bd0c232a2c6cb5fd40f
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * BIM_Workbench 2021.12.0
  * fasteners 0.5.29
  * Manipulator 1.5.7 (Disabled)
  * OpenTheme 2024.9.1
  * PieMenu 1.8.0 (Disabled)
  * Render 2024.8.4
  * SelectorToolbar (Disabled)
  * sheetmetal 0.4.26
  * TabBar (Disabled)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

benj5378, 2024-10-05:
This can absolutely be avoided. I believe `return _appearance.find(name) != _appearance.end()` can do the job. If possible, please test and make a PR

shaise, 2024-10-06:
I can easily place a PR. I'm just not sure how to test it, since I do not know what these function are for. I want to investigate why this function is called so many time first.

benj5378, 2024-10-06:
@shaise, make a PR and write that you haven't tested it. Then we'll find someone to test

shaise, 2024-10-06:
PR placed.
This seems to make freecad more responsive after startup. from about 2 seconds for the first action after creating a new document, it went down to a fraction of a second.

luzpaz, 2024-10-07:
Cc @davesrocketshop (since it touches Materials code)

davesrocketshop, 2024-10-07:
Good to know. Well done.

closed 2024-10-07 by shaise