FreeCAD/FreeCAD #17036: Sketcher: updating a spreadsheet dimension does not update an open sketch

eblanshey, 2024-10-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Create a spreadsheet with an alias/variable that represents a length dimension
2. Create a new part in Part Design, create a sketch with a line, set the formula to be the length dimension of the spreadsheet
3. While keeping the sketch open, switch to the spreadsheet and change the dimension. Save the document.
4. Return to the open sketch.

Expectation: the dimension is updated to the new dimension

Actual: the dimension is unchanged. You have to close the sketch and re-open it to continue sketching with the correct dimension. Pressing "CTRL+R" to refresh has no effect, either. 

This is especially a problem when windows are tiled such that the spreadsheet and sketches are both visible at the same time to be able to update variables in real-time.

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * Help 1.0.3
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-10-06:
What do you mean by 'return to the sketch'? If I activate the 3D view by clicking its tab, recomputing the sketch updates the dimension correctly.

hyarion, 2024-10-06:
I can reproduce this.
Sketches doesn't update automatically. It does however update when:
1. pressing esc if you have dimensions active
2. move a sketch point
3. close the sketch and open it again as the reporter describes

Here's a screen recording of points 1 and 2 (had to cut away number 3 due to 10MB file limit)

https://github.com/user-attachments/assets/caca2d46-84af-419b-8ffe-3fa5df934ab7

Reproduced in:
<details><summary>RC2</summary>
<p>

```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * A2plus 0.4.68
  * lattice2 1.0.0
  * CurvedShapes 1.0.10
  * sheetmetal 0.4.26
  * Curves 0.6.6
```

</p>
</details> 
<details><summary>Recent dev build</summary>
<p>

```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38888 +1 (Git)
Build type: Release
Branch: exception_wrapping
Hash: 1f6649ad3e5a52f49ef6d9ea109c64ad0aee8cca
Python 3.11.8, Qt 5.15.8, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * A2plus 0.4.68
  * lattice2 1.0.0
  * CurvedShapes 1.0.10
  * sheetmetal 0.4.26
  * Curves 0.6.6
```

</p>
</details>

eblanshey, 2024-10-06:
Thank you @hyarion. You're right in that it does seem that simply moving a dimension updates the sketch properly. And if I double click a dimension and open the popup, then close it, the sketch also updates. Would be nice to have the sketch update automatically as expected though. At least doing a "refresh" from the Edit menu should also update it, but that option is grayed out for me while in sketch mode.

hyarion, 2024-10-06:
I think I saw a bug some years ago that changing a field in a spreadsheet recomputed solids. IIRC that made it really difficult to work with spreadsheets when recomputing complex models.
The behavior that you are seeing could be an artifact of something like that, but it's just a guess.

The current behavior isn't especially nice either.

I don't know all the ins and outs here, but maybe it would be possible for the sketcher to update automatically if it is open?

eblanshey, 2024-10-06:
@hyarion if moving a single dimension does a full sketch recompute, hopefully it's not a big deal to do it when a value changes in the spreadsheet. At least enabling the Recompute button would be a good interim solution if there are performance issues doing it automatically.

luzpaz, 2025-01-25:
Can someone still confirm this on FreeCAD v1.1dev ?

eblanshey, 2025-01-27:
@luzpaz confirmed still happening in latest v1.1dev. There is a new bug happening, however, where pressing "backspace" while inside a cell dimension causes a popup to open temporarily in the sketcher, then closes it very quickly, and the cell leaves "typing mode". For example, if I have a dimension 100mm in the cell, and I delete one of the 0's to change the number, the cell stops allowing me to type, saving 10mm. The sketch does not update to the new 10mm value.

```
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39980 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 836ad91f3ddd3ec63b84b8e3f87a767d5465261d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * WebTools 1.0.0
  * Help 1.0.3
```

luzpaz, 2025-01-29:
> There is a new bug happening

Thanks @eblanshey can I trouble you to open a new ticket for this ?

eblanshey, 2025-02-01:
@luzpaz no problem, here: #19357

JohnOCFII, 2025-08-15:
> [@JohnOCFII](https://github.com/JohnOCFII) About info? TIA

```
OS: macOS 15.6
Architecture: arm64
Version: 1.1.0dev.42942 (Git)
Build date: 2025/08/15 15:09:23
Build type: Unknown
Branch: main
Hash: af83b6883eb0afae7d82aaaefc6a3991429c0a70
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * OpenTheme 2025.5.20
  * AddonManager 2025.8.15
  * manifest.json
  * Telemetry 1.0.2
```

closed 2026-09-08 by maxwxyz