FreeCAD/FreeCAD #17041: topological naming still have bug

kkeonline, 2024-10-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The attached video link shows that the topological naming error is still present in FreeCAD 1.0 RC2 and even nightly build version 38871 on September 30, 2024.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Anything else?

This video shows that the topological naming error is still present in FreeCAD 1.0 RC2 and even nightly build version 38871 on September 30, 2024.

[Video](https://youtu.be/gyTggFpGZmY)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-06:
I cannot reproduce on 
<details><summary>Details</summary>
<p>

```
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * Curves 0.6.48
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.24
  * Rocket 3.3.0
  * sheetmetal 0.4.26
  * Silk 0.1.5
  * Silk.backup1727166062.7227929 0.1.5 (Disabled)

``` 

</p>
</details> 

[image]

Please share the demo file and steps how to reproduce, ideally record a macro and upload it here.

kkeonline, 2024-10-06:
> I cannot reproduce on
> 
> Details
> ![grafik](https://private-user-images.githubusercontent.com/6246609/373927715-7e72742b-875d-4239-a70d-fe74dd2811d4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjgyMDgzOTEsIm5iZiI6MTcyODIwODA5MSwicGF0aCI6Ii82MjQ2NjA5LzM3MzkyNzcxNS03ZTcyNzQyYi04NzVkLTQyMzktYTcwZC1mZTc0ZGQyODExZDQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAwNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMDZUMDk0ODExWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YzQ3YzA0ZWI5MDIzZTIxNWFmYTE0OWU5N2Y2ZmZmZDgxMzI0MzRlOGYxZTZjZGUxZTRjY2YzYTlmZDZjZGI4MyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.KNLRFIPVsdV40R-6jZRBTUcsk9Hm-Ejq-D_ZKbMG4lA)
> 
> Please share the demo file and steps how to reproduce, ideally record a macro and upload it here.

Here is the steps:
1: Create sketch 1 draw a rectangle (any size)
2: Pad sketch 1 (any thick)
3. Create sketch 2 on face of pad draw any circle
4. Pocket sketch 2 (any depth)
5. Edit the sketch 1 create fillet all 4 corner of the rectangle + draw a circle in the center of rectangle (once click close editing, error will occur, the reference pane of sketch 2 will wrong.)

maxwxyz, 2024-10-06:
I did that but it works as expected, no issues. Can you share the file?

kkeonline, 2024-10-06:
> I did that but it works as expected, no issues. Can you share the file?

[topo_bug.zip](https://github.com/user-attachments/files/17268948/topo_bug.zip)

maxwxyz, 2024-10-06:
Works normally on my side. Can you try to reproduce from a fresh file and record a macro?

luzpaz, 2024-10-06:
@kkeonline can you update to latest dev version (yours is a 1+ weeks old) and try to reproduce again ?

7th-fret, 2024-10-06:
I can reproduce it following the steps snown on video.

> OS: Arch Linux (XFCE/xfce)
> Word size of FreeCAD-git: 64-bit
> Version: 1.1.0dev.38898 (Git)
> Build type: Release
> Branch: main
> Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
> Python 3.12.6, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
> Locale: English/United States (en_US)
> Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
> Installed mods: 
>   * fasteners 0.5.29
>   * OpticsWorkbench 1.0.24
>   * Curves 0.6.48
>   * OpenTheme 2024.5.16
>   * A2plus 0.4.67
>   * freecad.gears 1.3.0
>   * CurvedShapes 1.0.10
>   * sheetmetal 0.4.26

[image]

Krypto31337, 2024-10-06:
With 7th-fret's example, isn't it this?

https://github.com/FreeCAD/FreeCAD/issues/14720

In the 2nd op., if you just add the fillets and not the hole in the center it works.

luzpaz, 2024-10-06:
:eyes: #16803

kkeonline, 2024-10-06:
> @kkeonline can you update to latest dev version (yours is a 1+ weeks old) and try to reproduce again ?

I updated to the latest 38871 build and still the same error.

> With 7th-fret's example, isn't it this?
> 
> #14720
> 
> In the 2nd op., if you just add the fillets and not the pocket in the center it works with 38898.

I would say another word.
Create a fillet on a rectangle that have a hole will cause toponaming error.

[image]

kkeonline, 2024-10-06:
This also cause toponaming error

[image]

maxwxyz, 2024-10-06:
@CalligaroV FYI

kkeonline, 2024-10-06:
This is more simple edit showing tponaming error.
that not happen on other 3D software I mainly use.

[image]

CalligaroV, 2024-10-06:
Thanks @maxwxyz for the ping.

Haven't tried with main but the last example provided by @kkeonline can be reproduced also in LS3.

I think that this is one of the cases that #16803 may fix.
I'll work on that in the next days, hope I can come out with something in reasonable time.

kkeonline, 2024-12-30:
> Thanks @maxwxyz for the ping.
> 
> Haven't tried with main but the last example provided by @kkeonline can be reproduced also in LS3.
> 
> I think that this is one of the cases that #16803 may fix. I'll work on that in the next days, hope I can come out with something in reasonable time.

Look like #16803 doesn't fix this bug
It cause a segfault when modify the sketch and the patch #18548 fix the segfault
however nothing change from both patch #16803 and #18548
After modifing the sketch still need to fix with attachemnt editor, it not automatic find the new reference face name.