FreeCAD/FreeCAD #17043: Base.FreeCADError: No such command 'Arch_ToggleSubs'

realchrisolin, 2024-10-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I suspect this is more of a UX issue than an actual bug. When opening the BIMExample project:
Model window > Street level > right-click Entrance stairs > Toggle subcomponents throws an exception:

19:25:14  Traceback (most recent call last):
19:25:14    File "C:\Program Files\FreeCAD 1.0\Mod\BIM\ArchComponent.py", line 1525, in toggleSubcomponents
19:25:14      FreeCADGui.runCommand("Arch_ToggleSubs")
19:25:14  Base.FreeCADError: No such command 'Arch_ToggleSubs'

I skimmed through related classes and it looks like this is due to the Entrance stairs object not having any subcomponents. If anything, the messaging presented to the end-user should be improved.

### Full version info

```shell
1.0 RC2
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-06:
Please paste your full [About](https://wiki.freecad.org/About) info. TIA

maxwxyz, 2024-10-06:
@yorikvanhavre FYI

Roy-043, 2024-10-06:
The issue occurs if the BIM WB has not yet been loaded in the current FreeCAD session.
A potential solution can be found in Mod/Draft/draftviewproviders/view_draft_annotation.py:
https://github.com/FreeCAD/FreeCAD/blob/32e09d9554db26414697ed19e0189ccab4ea9029/src/Mod/Draft/draftviewproviders/view_draft_annotation.py#L227-L258

furgo16, 2024-10-06:
I can reproduce, I had always thought that the BIM Workbench was automatically loaded when a BIM file was opened (I'm not sure if this is either desirable or doable, though). I was surprised to see the [Toggle subcomponents](https://wiki.freecad.org/Arch_ToggleSubs) command available without having loaded the BIM Workbench.

To the OP, a better component to toggle subcomponents for could be <kbd>BIMExample</kbd> > <kbd>Site</kbd> > <kbd>Casa Martirana</kbd> > <kbd>Basement</kbd> > <kbd>Walls</kbd> > <kbd>Stone wall</kbd>

[image]

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git) Snap 1156
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * ezydxf
  * Ondsel-Lens 2024.9.13.02
  * Assembly4 0.50.15
```

yorikvanhavre, 2024-10-07:
No workbench is ever fully loaded when BIM starts indeed...
I think the best course of action here, is whenever we use a runCommand() in BIM (or anywhere else), first load the appropriate module (there is no harm loading it again if it was loaded already), in this case:
`FreeCADGui.addModule("bimcommands.BimArchUtils")`
But we could take the opportunity to do that more seriously throughout BIM...

furgo16, 2024-10-09:
> Or, we remove all these by-object context-menu options and replace them with proper, workbench-defined context menu commands. That would take care of everything automatically and maybe more cleanly?

Could you elaborate a bit more on this approach, in case someone would like to pick it up?

yorikvanhavre, 2024-10-10:
> Could you elaborate a bit more on this approach, in case someone would like to pick it up?

Basically, there are two ways to add items to the right-click context menu:

- In the workbench definition, ex: https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/BIM/InitGui.py#L606 but there you can only add FreeCAD commands, but it works in both the 3D view and the tree view.
- In the object's view provider, ex: https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/BIM/nativeifc/ifc_viewproviders.py#L86 but there you can only affect the tree view's context menu. It doesn't appear when right-clicking the 3D view

So my suggestion here would be:

1. Identify all places where viewprovier contextmenus are used
2. Turn all these into proper FreeCAD commands
3. Re-add them as workbench contextmenus

yorikvanhavre, 2024-10-10:
Point 1 above, there are quite a few...

```
ArchAxis.py
ArchAxisSystem.py
ArchBuildingPart.py
ArchBuilding.py
ArchComponent.py
ArchFloor.py
ArchGrid.py
ArchIFCView.py
ArchMaterial.py (3x)
ArchReference.py
ArchSchedule.py
ArchSectionPlane.py
ArchSite.py
ArchWall.p
ArchWindow.py
nativeifc/ifc_viewproviders.py (4x)
```

yorikvanhavre, 2024-10-11:
That's actually a good solution. And easy to implement too, it's just a matter of adding
```python
if FreeCADGui.activeWorkbench().GetClassName() != 'Gui::PythonWorkbench':
    return
```
at the beginning of each `setupContextMenu`