FreeCAD/FreeCAD #17044: Draft: duplicate snap and scale widgets in the notification area

furgo16, 2024-10-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

0. Prerequisites: neither the BIM nor Draft workbenches are set to "Auto-load" or have their checkboxes ticked under <kbd>Preferences</kbd> > <kbd>Workbenches</kbd>. If it makes a difference, there I also have <kbd>TabBar</kbd> selected as <kbd>Workbench selector type</kbd>
1. Start FreeCAD
2. Load the BIM Workbench
3. Load the Draft Workbench
4. Notice the duplicate snap and scale widgets in the notification area at the bottom of the screen:

[image]

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git) Snap 1156
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * ezydxf
  * Ondsel-Lens 2024.9.13.02
  * Assembly4 0.50.15
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

semhustej, 2024-10-06:
I am unable to reproduce the issue.
[image]

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: English/United States (en_US) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

```

maxwxyz, 2024-10-06:
Cannot reproduce on 
```
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * Curves 0.6.48
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.24
  * Rocket 3.3.0
  * sheetmetal 0.4.26
  * Silk 0.1.5
  * Silk.backup1727166062.7227929 0.1.5 (Disabled)

```

kaiwas, 2024-10-06:
Not confirm
`OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git) AppImage
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * lattice2 1.0.0
  * PitchedRoof
  * OpenTheme 2024.9.1
`

furgo16, 2024-10-06:
Thanks everyone for testing. I can reproduce this both on my snap and PPA package installations.

In order to trigger it, neither the BIM nor Draft workbenches can have their checkbox ticked under Preferences > Workbenches > Available Workbenches.

If they are, and the workbenches are put (pre-loaded?) on the tabbar, I cannot reproduce it. I'll highlight this better on the original description.

semhustej, 2024-10-06:
Thanks for clarification @furgo16 .
I can reproduce the issue now. I do not even have to do steps (2) and (3) to see the issue. It is immediately visible after FreeCAD restart:
[image]

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: English/United States (en_US) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

```

furgo16, 2024-10-06:
Thanks for testing!

semhustej, 2024-10-09:
I get different behavior with the current dev release. When Draft workbench is set as default and BIM workbench is auto-loaded, the widgets are duplicated.  Doesn't matter whether switching is by combobox or tabbar.
[image]

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: English/United States (en_US) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

```

Roy-043, 2025-03-27:
At step 3 in the test scenario the Draft WB is activated, immediately deactivated, and then activated again. That should not happen. As a result there are timing issues with the delayed display of the widgets.

Activation 1: The widgets are displayed with a delay
Deactivation: If the widgets exist they are removed, but they are not yet there, so nothing happens.
Activation 2: The widgets are still not there, so they are again triggered.

I'll have to look at the logic of the code. Moving conditional checks to the end of the delay may improve things.

PR #20491 (2025-03-28): Draft: fix status bar widgets display timing issue
Fixes #17044.

See:
https://github.com/FreeCAD/FreeCAD/issues/17044#issuecomment-2757637750
commit 4ac8f0b1 (2025-03-28): Draft: fix status bar widgets display timing issue

Fixes #17044.
closed 2025-03-28 by hyarion