FreeCAD/FreeCAD #17046: C++ DXF importer ignores normal (gc 210) of polylines

Roy-043, 2024-10-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The attached DXF file does not import properly with the C++ importer. Polylines with (210 0.0 0.0 -1.0) appear mirrored in the result. Gc 210 (the normal) is ignored.

[Seite.dxf.txt](https://github.com/user-attachments/files/17268771/Seite.dxf.txt) (remove .txt)

File in DwgCAD:
[image]
File in FreeCAD:
[image]

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-06:
@kpemartin any chance you can take a look at this ?

FEA-eng, 2026-02-08:
> [@FEA-eng](https://github.com/FEA-eng) I do not think this is related to settings. There seems to be a new issue caused by a v1.1 regression.

So it's even worse now. Don't you think that the new regression deserves a separate issue and maybe a blocker status?

maxwxyz, 2026-11-07:
Part of this is fixed in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/6cfaa17ad3e36f3e014224f7f2d8a0c903bb686a

The C++ importer applies the normal to 2D polylines. Other 2D entity types and the later block-handling problems are not covered.