FreeCAD/FreeCAD #17047: Assembly: error when using the Toggle Grounded command on a joint

furgo16, 2024-10-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Uncompress the test file [test file](https://github.com/user-attachments/files/17268821/TestAssembly.zip) and open it with FreeCAD 
2. Double-click on the `Assembly` item in the document tree to activate the assembly and to load the Assembly Workbench
3. Select `Cuve002` on the document tree
4. Click the [Toggle Grounded](https://wiki.freecad.org/Assembly_ToggleGrounded) command, or alternatively press the <kbd>G</kbd> key on the keyboard
5. Select `Joints` > `GroundedJoint` on the document tree
6. Click the [Toggle Grounded](https://wiki.freecad.org/Assembly_ToggleGrounded) command, or alternatively press the <kbd>G</kbd> key on the keyboard
7. The following stacktrace appears on the Report panel:

```python-traceback
Running the Python command 'Assembly_ToggleGrounded' failed:
Traceback (most recent call last):
  File "/snap/freecad/1157/usr/Mod/Assembly/CommandCreateJoint.py", line 562, in Activated
    createGroundedJoint(moving_part)
  File "/snap/freecad/1157/usr/Mod/Assembly/CommandCreateJoint.py", line 493, in createGroundedJoint
    Gui.doCommand(commands)
  File "<string>", line 5, in <module>
  File "/snap/freecad/1157/usr/Mod/Assembly/JointObject.py", line 1095, in __init__
    joint.Placement = obj_to_ground.Placement

'Assembly.JointGroup' object has no attribute 'Placement'
```

Note: I only later realized later that this is not the expected workflow to toggle a grounded joint. The component needs to be selected (`Cube001`) instead of the joint.

Nevertheless, a warning pointing the (unexperienced) user to select a component instead of a joint might be better than an unhandled error.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git) Snap 1157
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * ezydxf
  * Ondsel-Lens 2024.9.13.02
  * Assembly4 0.50.15
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-10-06:
Unrelated, but strangely enough, the last line of the traceback is missing the raised exception's name (`AttributeError`?). I had to check twice that it wasn't a copy & paste error of mine.

luzpaz, 2025-05-17:
@PaddleStroke any thoughts ?