FreeCAD/FreeCAD #17048: Core: Segfault on deleting an object with MultiTransform

pieterhijma, 2024-10-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Deleting a document object that has a PartDesign multitransform with a mirrored transform crashes FreeCAD.

To reproduce:

- Create a new empty document
- Create a part
- Using PartDesign, create a body
- Create a sketch, for example on the XY plane, and for example a rectangle
- Pad the rectangle
- Select the pad and create a multitransform
- Right click in transformations to add a mirrored transformation
- For the plane select the base XY plane (the same plane of the sketch)
<br>

- Copy the part (select it and do Ctrl-c)
- Make sure all dependencies are selected and click ok
<br>

- Create a new empty document
- In this document, paste the part from the clipboard (Ctrl-v)
- Select the part, press Delete, and click on Yes to delete the contents as well

In the stacktrace below, `getTree()` evaluates to `nullptr`.

```
Thread 1 "FreeCAD" received signal SIGSEGV, Segmentation fault.
0x00007ffff5cf60e2 in Gui::DocumentObjectItem::testStatus (this=0x55555bf3b400, resetStatus=false, icon1=..., icon2=...) at freecad-main/src/Gui/Tree.cpp:5391
5391	        getTree()->initViewItemOption(&opt);

5386
5387	#if QT_VERSION < QT_VERSION_CHECK(6,0,0)
5388	        int w = getTree()->viewOptions().decorationSize.width();
5389	#else
5390	        QStyleOptionViewItem opt;
5391	        getTree()->initViewItemOption(&opt);
5392	        int w = opt.decorationSize.width();
5393	#endif
5394
5395	        QPixmap pxOn, pxOff;
```

<details>
<summary>Stacktrace</summary>
<pre>
#0  0x00007ffff5cf60e2 in Gui::DocumentObjectItem::testStatus (this=0x55555bf3b400, resetStatus=false, icon1=..., icon2=...) at /home/pieter/current/fpa/freecad-main/src/Gui/Tree.cpp:5391
#1  0x00007ffff5cff16e in Gui::DocumentObjectData::testStatus (this=0x55555e0cf0f0, resetStatus=false) at /home/pieter/current/fpa/freecad-main/src/Gui/Tree.cpp:356
#2  0x00007ffff5cf0d54 in Gui::DocumentItem::testStatus (this=0x55555de68410) at /home/pieter/current/fpa/freecad-main/src/Gui/Tree.cpp:4645
#3  0x00007ffff5ce4cd0 in Gui::TreeWidget::onUpdateStatus (this=0x555555e54980) at /home/pieter/current/fpa/freecad-main/src/Gui/Tree.cpp:3043
#4  0x00007ffff5cd3450 in Gui::TreeWidget::_updateStatus (this=0x555555e54980, delay=false) at /home/pieter/current/fpa/freecad-main/src/Gui/Tree.cpp:968
#5  0x00007ffff5ce7789 in Gui::TreeWidget::onItemSelectionChanged (this=0x555555e54980) at /home/pieter/current/fpa/freecad-main/src/Gui/Tree.cpp:3402
#6  0x00007ffff5d27d1e in QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (Gui::TreeWidget::*)()>::call(void (Gui::TreeWidget::*)(), Gui::TreeWidget*, void**) (f=(void (Gui::TreeWidget::*)(Gui::TreeWidget * const)) 0x7ffff5ce76a4 <Gui::TreeWidget::onItemSelectionChanged()>, o=0x555555e54980, arg=0x7fffffff92d8) at /usr/include/qt6/QtCore/qobjectdefs_impl.h:145
#7  0x00007ffff5d21d60 in QtPrivate::FunctionPointer<void (Gui::TreeWidget::*)()>::call<QtPrivate::List<>, void>(void (Gui::TreeWidget::*)(), Gui::TreeWidget*, void**) (f=(void (Gui::TreeWidget::*)(Gui::TreeWidget * const)) 0x7ffff5ce76a4 <Gui::TreeWidget::onItemSelectionChanged()>, o=0x555555e54980, arg=0x7fffffff92d8) at /usr/include/qt6/QtCore/qobjectdefs_impl.h:182
#8  0x00007ffff5d18cb1 in QtPrivate::QCallableObject<void (Gui::TreeWidget::*)(), QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) (which=1, this_=0x555555f03dd0, r=0x555555e54980, a=0x7fffffff92d8, ret=0x0) at /usr/include/qt6/QtCore/qobjectdefs_impl.h:553
#9  0x00007fffefda3457 in QtPrivate::QSlotObjectBase::call (this=<optimized out>, r=<optimized out>, a=<optimized out>, this=<optimized out>, r=<optimized out>, a=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qobjectdefs_impl.h:469
#10 doActivate<false> (sender=<optimized out>, signal_index=<optimized out>, argv=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qobject.cpp:4086
#11 0x00007ffff107f365 in QTreeWidget::itemSelectionChanged (this=0x555555e54980) at /usr/src/debug/qt6-base/build/src/widgets/Widgets_autogen/include/moc_qtreewidget.cpp:414
#12 QTreeWidgetPrivate::selectionChanged (this=0x555555df2110, selected=..., deselected=...) at /usr/src/debug/qt6-base/qtbase/src/widgets/itemviews/qtreewidget.cpp:2394
#13 0x00007fffefda3457 in QtPrivate::QSlotObjectBase::call (this=<optimized out>, r=<optimized out>, a=<optimized out>, this=<optimized out>, r=<optimized out>, a=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qobjectdefs_impl.h:469
#14 doActivate<false> (sender=<optimized out>, signal_index=<optimized out>, argv=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qobject.cpp:4086
#15 0x00007fffeff5f65b in QItemSelectionModel::selectionChanged (this=0x555555ef7190, _t1=..., _t2=...) at /usr/src/debug/qt6-base/build/src/corelib/Core_autogen/include/moc_qitemselectionmodel.cpp:492
#16 QItemSelectionModelPrivate::rowsAboutToBeInserted (this=0x555555ef71b0, parent=..., start=0, end=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/corelib/itemmodels/qitemselectionmodel.cpp:825
#17 0x00007fffefda3457 in QtPrivate::QSlotObjectBase::call (this=<optimized out>, r=<optimized out>, a=<optimized out>, this=<optimized out>, r=<optimized out>, a=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qobjectdefs_impl.h:469
#18 doActivate<false> (sender=<optimized out>, signal_index=<optimized out>, argv=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qobject.cpp:4086
#19 0x00007fffeff545a7 in QAbstractItemModel::rowsAboutToBeInserted (this=0x555555ef5000, _t1=..., _t2=<optimized out>, _t3=<optimized out>, _t4=...) at /usr/src/debug/qt6-base/build/src/corelib/Core_autogen/include/moc_qabstractitemmodel.cpp:1049
[... 102 lines cut ...]
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git)
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.12.6, Qt 6.7.3, Coin 4.0.3, Vtk , OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Installed mods: 
  * OpenTheme 2024.5.16
  * osh-autodoc-workbench 0.2.3
  * Ondsel-Lens 2024.9.13.02
```

### Subproject(s) affected?

Core

### Anything else?

Issue https://github.com/FreeCAD/FreeCAD/issues/16688 may be related because it involves Parts, copies, and deletion as well.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-06:
@pieterhijma any chance you can upload an example file for convenient testing ?

pieterhijma, 2024-10-06:
@luzpaz, yes (rename to `test-delete-multitransform.FCStd`): 
[test-delete-multitransform.zip](https://github.com/user-attachments/files/17269764/test-delete-multitransform.zip)

If you delete the part and press yes to delete all contents, it should crash.

luzpaz, 2024-10-06:
Thanks. 

Can't reproduce on 
```
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git) AppImage
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Classic/Qt default
```

luzpaz, 2024-10-06:
@pieterhijma just noticed you don't have VTK version in your About info

> Python 3.12.6, Qt 6.7.3, Coin 4.0.3, **Vtk** , OCC 7.8.1

AIRCAP, 2024-10-06:
cannot reproduce no crash on 
```
OS: Ubuntu 20.04.6 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38896 (Git)
Build type: Unknown
Branch: main
Hash: 596fa2856b60e3262aba7852625d38e1f8ec7cd6
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CurvedShapes 1.0.10
  * CfdOF 1.27.10
  * Curves 0.6.48
  * AirPlaneDesign 0.4.1
```

pieterhijma, 2024-10-06:
> @pieterhijma just noticed you don't have VTK version in your About info
> 
> > Python 3.12.6, Qt 6.7.3, Coin 4.0.3, **Vtk** , OCC 7.8.1

Weird it is not reported, it is 9.3.1. 

Strange that it can't be reproduced. Actually, I tried it again and I couldn't reproduce on my system. Then I tried a couple of times and it crashed again, so perhaps it's not completely deterministic.

In any case, following the steps seems to be completely reproducible. If still not reproducible, I suspect the Qt 5/6 difference.

zbynekwinkler, 2024-10-07:
I was able to follow all the steps (several times) and have not encountered any crashes.

```
OS: Debian GNU/Linux 12 (bookworm)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38896 +4 (Git)
Build type: Debug
Branch: docker
Hash: a9de3243101505a7f803c267f85f020ec59b477f
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

SpaceBison, 2025-01-16:
I encountered this in my own project and consistently reproduce it on the example.

```
OS: Manjaro Linux (GNOME/gnome-xorg/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/qt6ct-style
Installed mods: 
  * ThreadProfile 1.92.0
```

FEA-eng, 2025-10-22:
No crash or errors on Windows.

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/14 15:14:51
Build type: Release
Branch: main
Hash: 41ce37233d64c81c2f105de085635d4dcd669943
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
```

pieterhijma, 2025-10-24:
I cannot reproduce it myself any longer. @SpaceBison, can you still reproduce it?

pieterhijma, 2025-10-28:
Thanks, closing.

closed 2025-10-28 by pieterhijma