FreeCAD/FreeCAD #17049: Making a circle and entering the dimension in the same operations generates an overconstrained shape.

GAZ082, 2024-10-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Making a circle and entering the dimension in the same operations generates an over-constrained shape.

https://github.com/user-attachments/assets/b0fae021-35ca-4e8a-8904-fdec3325f128

### Full version info

```shell
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git) AppImage
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * OpenTheme 2024.9.1
  * QuickMeasure 2022.10.28
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

NewStorm, 2024-10-06:
I cannot reproduce your issue on macOS with version 1.1.0dev.38871. However, it could be of interest that different constraints are set for me (see attached Image).

```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion

```
[image]

Krypto31337, 2024-10-06:
Works fine here, performed exactly as you shown.  

Like NewStorm commented, there's something up with your constraints as mine look like the ones in that comment, except a 5mm constraint on diameter.

OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git) AppImage
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * sheetmetal 0.4.26
  * freecad.gears 1.3.0

maxwxyz, 2024-10-06:
I remember an autoconstraint issue about creating this overconstrained issue when choosing the origin but cannot find it at the moment. Does it also happen when you create it not at the origin?

NewStorm, 2024-10-06:
I can not reproduce the issue with a featureless body but GAZ082s body already has a pad.
If this feature is a cylinder, that would be a strong indication for #14736 or ?

GAZ082, 2024-10-06:
Hi folks for testing.

1. Only happens if i anchor in the origin or in another point and then perform the dimensioning in the same operation.
2. Happens in an empty sketch as well.

https://github.com/user-attachments/assets/cb68d149-80ab-4a78-b883-44e8086de7fd

Macro:
```# -*- coding: utf-8 -*-

# Macro Begin: /home/gabriel/.local/share/FreeCAD/Macro/sasa.FCMacro +++++++++++++++++++++++++++++++++++++++++++++++++
import FreeCAD
import PartDesign
import PartDesignGui
import Sketcher

# Gui.runCommand('Std_DlgMacroRecord',0)
### Begin command PartDesign_Body
App.activeDocument().addObject('PartDesign::Body','Body')
App.ActiveDocument.getObject('Body').Label = 'Body'
# Gui.activateView('Gui::View3DInventor', True)
# Gui.activeView().setActiveObject('pdbody', App.activeDocument().Body)
# Gui.Selection.clearSelection()
# Gui.Selection.addSelection(App.ActiveDocument.Body)
App.ActiveDocument.recompute()
### End command PartDesign_Body
# Gui.Selection.addSelection('Unnamed3','Body')
# Gui.runCommand('PartDesign_NewSketch',0)
# Gui.Selection.clearSelection()
# Gui.Selection.addSelection('Unnamed3','Body','Origin.XY_Plane.',26.258,32.5,0)
App.getDocument('Unnamed3').getObject('Body').newObject('Sketcher::SketchObject','Sketch')
App.getDocument('Unnamed3').getObject('Sketch').AttachmentSupport = (App.getDocument('Unnamed3').getObject('XY_Plane'),[''])
App.getDocument('Unnamed3').getObject('Sketch').MapMode = 'FlatFace'
App.ActiveDocument.recompute()
# Gui.getDocument('Unnamed3').setEdit(App.getDocument('Unnamed3').getObject('Body'), 0, 'Sketch.')
# ActiveSketch = App.getDocument('Unnamed3').getObject('Sketch')
# tv = Show.TempoVis(App.ActiveDocument, tag= ActiveSketch.ViewObject.TypeId)
# ActiveSketch.ViewObject.TempoVis = tv
# if ActiveSketch.ViewObject.EditingWorkbench:
#   tv.activateWorkbench(ActiveSketch.ViewObject.EditingWorkbench)
# if ActiveSketch.ViewObject.HideDependent:
#   tv.hide(tv.get_all_dependent(App.getDocument('Unnamed3').getObject('Body'), 'Sketch.'))
# if ActiveSketch.ViewObject.ShowSupport:
#   tv.show([ref[0] for ref in ActiveSketch.AttachmentSupport if not ref[0].isDerivedFrom("PartDesign::Plane")])
# if ActiveSketch.ViewObject.ShowLinks:
#   tv.show([ref[0] for ref in ActiveSketch.ExternalGeometry])
# tv.sketchClipPlane(ActiveSketch, ActiveSketch.ViewObject.SectionView)
# tv.hide(ActiveSketch)
# del(tv)
# del(ActiveSketch)
# 
# ActiveSketch = App.getDocument('Unnamed3').getObject('Sketch')
# if ActiveSketch.ViewObject.RestoreCamera:
#   ActiveSketch.ViewObject.TempoVis.saveCamera()
#   if ActiveSketch.ViewObject.ForceOrtho:
#     ActiveSketch.ViewObject.Document.ActiveView.setCameraType('Orthographic')
# 
# Gui.Selection.clearSelection()
# Gui.runCommand('Sketcher_CompCreateConic',0)
lastGeoId = len(ActiveSketch.Geometry)

geoList = []
geoList.append(Part.Circle(App.Vector(0.000000, 0.000000, 0.000000), App.Vector(0.000000, 0.000000, 1.000000), 1.000000))
App.getDocument('Unnamed3').getObject('Sketch').addGeometry(geoList,False)
del geoList

constraintList = []
constraintList = []
constraintList.append(Sketcher.Constraint('Coincident', 0, 3, -1, 1))
constraintList.append(Sketcher.Constraint('PointOnObject', -1, 1, 0))
[... 50 lines cut ...]
App.ActiveDocument.recompute()
# Gui.runCommand('Sketcher_CompCreateConic',0)
lastGeoId = len(ActiveSketch.Geometry)

geoList = []
geoList.append(Part.Circle(App.Vector(-47.848164, 21.265984, 0.000000), App.Vector(0.000000, 0.000000, 1.000000), 12.340956))
App.getDocument('Unnamed3').getObject('Sketch').addGeometry(geoList,False)
del geoList

constraintList = []
App.getDocument('Unnamed3').getObject('Sketch').addConstraint(Sketcher.Constraint('Coincident', 2, 3, 1, 1))

App.ActiveDocument.recompute()
# Gui.runCommand('Sketcher_Dimension',0)
App.getDocument('Unnamed3').getObject('Sketch').addConstraint(Sketcher.Constraint('Diameter',2,24.681912)) 
# Gui.Selection.addSelection('Unnamed3','Body','Sketch.Edge3',-37.0889,27.3105,0)
App.getDocument('Unnamed3').getObject('Sketch').setDatum(3,App.Units.Quantity('24.680000 mm'))
App.ActiveDocument.recompute()
# Gui.Selection.clearSelection()
App.getDocument('Unnamed3').getObject('Sketch').movePoint(1,1,App.Vector(-29.746902,30.000162,0),0)
App.ActiveDocument.recompute()
# Gui.Selection.addSelection('Unnamed3','Body','Sketch.Edge3',-33.5602,18.2641,0.008,False)
# Gui.Selection.clearSelection()
# Macro End: /home/gabriel/.local/share/FreeCAD/Macro/sasa.FCMacro +++++++++++++++++++++++++++++++++++++++++++++++++
```

GAZ082, 2024-10-06:
Related:

When i do a Box, if the dimension is close enough to the origin or the point, triggers a constraint.

Then i tested with a circle. If the diameter is big enough, there is no additional constrain that triggers the error, but if it is close (1,2 mm) it happens.

https://github.com/user-attachments/assets/24f725b7-fc86-43b9-a568-cbf5226d701b

maxwxyz, 2024-10-06:
@AjinkyaDahale @FlachyJoe FYI

Krypto31337, 2024-10-06:
For me, a 2mm centered circle works fine but a 1mm centered circle will trigger the same error as shown in the latest video from GAZ082.

Try zooming in a bit before making the small 1mm circle. If I do that, I don't get the error.

NewStorm, 2024-10-06:
I have tested this with various zoom levels and dimensions.
At a zoomlevel 14,78m(irrelevant) x 5522,10mm it breaks at around 53,3.
That factor of 1e-3 is consistent with tests on different levels and dimensions

kriswilk, 2024-10-06:
> I have tested this with various zoom levels and dimensions.

I don't think the zoom level is important What actually seems to govern this bug is the "pick radius" (Preferences -> General -> Selection). With the default setting of 16 px, and the default sketch zoom level, I can't create a 3mm circle by dimensioning immediately. However, by lowering the pick radius to 2 px, the same operations succeeds in the same viewport.

As @NewStorm noticed, if you zoom in or out, the critical diameter for dimensioning changes. That's a consequence of where the pick radius kicks in.

Basically, something about the dimensioning logic is (directly or indirectly) using the pick radius to make a decision when it clearly shouldn't.

https://github.com/user-attachments/assets/a4229245-8834-40fa-ab9b-403b6d06fed8

AjinkyaDahale, 2024-10-06:
@PaddleStroke could you take a look, since this relates to OVP?

PaddleStroke, 2024-10-07:
My current branch is at 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6 and I cannot reproduce. Perhaps it has been introduced since. I'll rebase to give it a try.

luzpaz, 2024-10-15:
@PaddleStroke any luck?

closed 2025-08-22 by maxwxyz