FreeCAD/FreeCAD #17050: Step import does not present a progress indication...

leoheck, 2024-10-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When importing a step file.

It shows this popup... then there is this `Show progress bar when importing` option...
I don't see any progress indication anywhere. 
Is this working?

[image]

The step files tested are big with multiple objects inside. So it takes time to import the step file. 

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git) AppImage
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.4.26
  * InventorLoader 1.4.0
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.0.24
  * lattice2 1.0.0
  * kicadStepUpMod 11.3.9
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-06:
could you share a .STEP file to test? TIA

leoheck, 2024-10-07:
I cannot share the step files I am creating but I created one new for you @maxwxyz 
[big_step-my_design.zip](https://github.com/user-attachments/files/17279115/big_step-my_design.zip)

luzpaz, 2024-10-07:
Confirmed on  
```
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git) AppImage
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

leoheck, 2024-10-07:
The `Reduce number of objects` seems not to be working too but I am not sure on how this feature works. I wanted to have this feature enabled since I have some duplicates.

leoheck, 2024-10-07:
> Confirmed on

I am not sure it is a regression, I've never seen this progress bar... 
I would say it is just a non implemented feature.

luzpaz, 2024-10-14:
@yorikvanhavre can you weigh-in ?

FEA-eng, 2025-10-22:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/14 15:14:51
Build type: Release
Branch: main
Hash: 41ce37233d64c81c2f105de085635d4dcd669943
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
```

luzpaz, 2025-11-28:
@ipatch does this still not work ? (asking since you've been looking at the STEP code lately.)

luzpaz, 2025-11-28:
Should we hide it in that case and open a feature request ?

ipatch, 2025-11-29:
> Should we hide it in that case and open a feature request ?

doesn't sound like a terrible idea, just doing a quick grep through `./src/Mod/Import/App` for `progress` yields,

```
╰─λ rg --fixed-strings -i --hidden -g '!*.ts'   -- 'progress' ./
./ImportOCAF2.cpp
105:    defaultOptions.showProgress = settings.getShowProgress();
523:    sequencer = options.showProgress ? &seq : nullptr;

./ReaderGltf.cpp
28:#include <Message_ProgressRange.hxx>
63:    Standard_Boolean ret = aReader.Perform(filename, Message_ProgressRange());

./ImportOCAF2.h
73:    bool showProgress = false;
104:    void setShowProgress(bool enable)
106:        options.showProgress = enable;

./WriterGltf.cpp
28:#include <Message_ProgressRange.hxx>
55:    Standard_Boolean ret = aWriter.Perform(hDoc, aMetadata, Message_ProgressRange());

./AppImportPy.cpp
45:#include <Message_ProgressRange.hxx>

./SCL/essa_par.py
46:            current_param, progress = process_nested_parent_str2(nv)
50:            k += progress + 1

./SCL/Utils.py
54:            current_param, progress = process_nested_parent_str(nv)
57:            k += progress + 1
╭─capin ssh m1mac in /opt/code/fcgit/fcsrc/src/Mod/Import/App (ipatch.tshoot.25759↑1 ✚2⚑19)
╰─λ pwd
/opt/code/fcgit/fcsrc/src/Mod/Import/App
```

closed 2026-04-27 by hyarion