FreeCAD/FreeCAD #17052: BIM WB - PathArray Losing Properties When Saved

point-droid, 2024-10-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the attached I have created a Beam, a path to create an array and a custom material. The beam is set to the custom material and is then used to create a PathArray. The PathArray is coloured correctly, from the custom material, until I save, close and reopen the file at which point the elements in the PathArray revert to the FreeCAD default colours. Deleting the PathArray and recreating it solves the problem until I save, close, reopen again. I don't know if it's just colour or other properties that are also lost.

[temp.zip](https://github.com/user-attachments/files/17270134/temp.zip)

### Full version info

```shell
OS: openSUSE Leap 15.5 (KDE//usr/share/xsessions/plasma5)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * parts_library
```

### Subproject(s) affected?

None

### Anything else?

https://forum.freecad.org/viewtopic.php?t=91157

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-06:
@paullee0 do you mind trying to reproduce ?

semhustej, 2024-10-07:
I can reproduce the problem.

1. tried as described in the bug report - confirm
2. tried creating Draft_OrthoArray from column and followed the steps from the bug report - confirm

Plus I tried creating Draft_OrthoArray from column and than changing the material of the base object. The change does not propagate the objects in the array.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: English/United States (en_US) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

```

paullee0, 2024-10-08:
Thanks @semhustej.

Out of curiosity, I tested, tweak the object a little bit and has below finding:-

1. Changed the Transparency of the object to any figure say 90 (not sure if other figure works) and turned it back to 0
2. If the colour turns to the chosen material colour, save the model file
3. Reload the model file, the colour property is restored

Theory :  
1. There are 2 Properties Input - 1/ Shape Appearance Transparency, 2/ Transparency
2. Maybe they are out-of-sync / in conflict which result in colour restoration failed

yorikvanhavre, 2024-10-08:
In any case the PathArray object does have the correct color stored:

[image]

The beam object too, and it's correctly rendered:

[image]

So there must be a problem within the PathArray object...

furgo16, 2024-10-09:
Could this be related to https://github.com/FreeCAD/FreeCAD/issues/15503?

yorikvanhavre, 2024-10-10:
> Could this be related to https://github.com/FreeCAD/FreeCAD/issues/15503?

I don't think so (but I could very well be wrong), I think this one is linked to the use of App:Links while #15503 not. 

But this needs further testing (ex. If it works OK with non-BIM objects, if other App::Link objects suffer from the same, etc)

PR #18123 (2024-11-29): Draft: fix color of arrays
If `get_diffuse_color` returned a list with a single color, the list was wrongly multiplied. Regression introduced in V0.21, but in that version the error was masked by how Core handled the diffuse color.

Fixes #17052.
commit 298e6921 (2024-11-29): Draft: fix color of arrays

If `get_diffuse_color` returned a list with a single color, the list was wrongly multiplied. Regression introduced in V0.21, but in that version the error was masked by how Core handled the diffuse color.

Fixes #17052.
commit d9a4f8a5 (2024-11-29): Draft: fix color of arrays (#18123)

If `get_diffuse_color` returned a list with a single color, the list was wrongly multiplied. Regression introduced in V0.21, but in that version the error was masked by how Core handled the diffuse color.

Fixes #17052.
commit 1372cdb5 (2024-11-29): Draft: fix color of arrays (#18123)

If `get_diffuse_color` returned a list with a single color, the list was wrongly multiplied. Regression introduced in V0.21, but in that version the error was masked by how Core handled the diffuse color.

Fixes #17052.
closed 2024-11-29 by Roy-043