FreeCAD/FreeCAD #17054: Hyperlinks in About > Libraries open empty tabs in the application

maxwxyz, 2024-10-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The hyperlinks in the about dialog open empty tabs in the FreeCAD application, instead of the system's web browser.
I guess this is due to the removal of the web features @chennes ? Are there other locations which need a workaround?
[image]

### Full version info

```shell
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * Curves 0.6.48
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.24
  * Rocket 3.3.0
  * sheetmetal 0.4.26
  * Silk 0.1.5
  * Silk.backup1727166062.7227929 0.1.5 (Disabled)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-06:
Can confirm. When I click on one of the links, I see a small dialog pop-up for a second then disappear. Then it output the following to the Report View:

```
17:16:33  QTextBrowser: No document for file:////tmp/.mount_FreeCAjVSdO9/usr/share/doc/FreeCAD//ThirdPartyLibraries.html#_TocCoin3D
17:16:39  QTextBrowser: No document for file:////tmp/.mount_FreeCAjVSdO9/usr/share/doc/FreeCAD//ThirdPartyLibraries.html#_TocEigen
17:16:48  QTextBrowser: No document for file:////tmp/.mount_FreeCAjVSdO9/usr/share/doc/FreeCAD//ThirdPartyLibraries.html#_TocKDL
17:16:51  QTextBrowser: No document for file:////tmp/.mount_FreeCAjVSdO9/usr/share/doc/FreeCAD//ThirdPartyLibraries.html#_TocOCCT
17:16:57  QTextBrowser: No document for file:////tmp/.mount_FreeCAjVSdO9/usr/share/doc/FreeCAD//ThirdPartyLibraries.html#_TocXercesC
```

```
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git) AppImage
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Classic/Qt default
```

NewStorm, 2024-10-16:
I'm currently working on this bug and it can be fixed by editing just a few lines but this leads to another question.
The current logic is to show the listing of the various used open source components.
The links refer to different sections in ThirdPartyLibraries.html with basically the same information.
This seems redundant to me. 
Are there any reasons to not show the ThirdPartyLibraries.html at the first place ?

PR #17337 (2024-10-25): Splashscreen: Show ThirdPartyLibraries.html at About/Libraries - Fixes #17054
Currently, the 3rd party libraries information is contained in the ThirdPartyLibraries.html and get additionally generated in the About/Libraries Tab. 
With this pull request, the version strings get injected into the html via cmake and instead of generating the libraries listing, the html gets shown instead.
This removes redundancies, duplicate source code and splits data from logic.
It also fixes #17054
closed 2024-10-25 by chennes