FreeCAD/FreeCAD #17056: Crash in on Open CSV (drag-n-drop+from start)

hyarion, 2024-10-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD crashes when opening a CSV files from either:
* drag and drop
* clicking a csv file in the start window under recent files

FreeCAD can load the files successfully when loading from the menu.

File to test with:
[test.csv](https://github.com/user-attachments/files/17270702/test.csv)

Drag and drop crash
---

1. open freecad
2. drag the attached csv file to the main window

Crash!

Screen recording:

https://github.com/user-attachments/assets/62df1717-51fc-40b7-a8bc-d3f3d7a2da09

Open-from-start crash
---
1. open freecad
2. select a csv file from the recent files

Crash!

Screen recording:

https://github.com/user-attachments/assets/2c7bdfc5-da76-41df-843e-b8190f9b5b9d

### Full version info

```shell
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * A2plus 0.4.68
  * lattice2 1.0.0
  * CurvedShapes 1.0.10
  * sheetmetal 0.4.26
  * Curves 0.6.6

and

OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38888 +2 (Git)
Build type: Release
Branch: exception_wrapping
Hash: 6885f17a51e7a190e23821380ef8f5d281964c0c
Python 3.11.8, Qt 5.15.8, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * A2plus 0.4.68
  * lattice2 1.0.0
  * CurvedShapes 1.0.10
  * sheetmetal 0.4.26
  * Curves 0.6.6
```

### Subproject(s) affected?

Spreadsheet

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hyarion, 2024-10-06:
The issue here is that spreadsheet tries to get the active document but no document are loaded with a new freecad, so it crashes instead:
https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Spreadsheet/Gui/AppSpreadsheetGui.cpp#L107

As I see it there's too ways to solve this:
1. Create a new document if there isn't one already, or
2. Make the actions for these action use open instead of import.

The first option is easy to implement and fixes this issue, but it feels wrong to create a file in a function called insert.

The second option would change the behavior for all imports. Do we want to call open on all non-.FCStd-files?

Suggestions?

AIRCAP, 2024-10-07:
suggestion. if a particular import requires a doc to function, it should check if a doc is existing and if not throw an exception, not crash.

this exception could then be caught by the file open dialog (or the function called by open recent, probably the same) and notify or ask the user (This import requires an active document. Do you want to start a new document?)

AIRCAP, 2024-10-07:
or automagically just trigger "new document" and call the opening again

but either way, not the actual importer function should do that, the global "file open" function could do that (as opposed to an error message).

while the actual import function should just throw an exception if it can't

PR #17084 (2024-10-14): Fix crashes when importing and opening .csv files on macOS
fixes #17056 

This PR does the following:
* Combines open and insert spreadsheet to reduce code duplication
* Creates new document on import of docName does not exist as required by [MainWindow](https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/MainWindow.cpp#L2415):
    > `// if the passed document name doesn't exist the module should create it, if needed`
* Let recent file cards in start open files instead of importing them. This also makes start open the recent files in the same way as [CmdStdOpen](https://github.com/FreeCAD/Fre [... cut]