FreeCAD/FreeCAD #17059: Improvement of Python API -- Python implementations have no __repr__

AIRCAP, 2024-10-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As discussed in #development:

> @bensay: @CorvusCorax idk if an issue already exists... but please open an issue on GitHub requesting implementation of `__repr__` across objects in all of FreeCAD. It's actually a very big issue, making it very difficult to use Python Console.
> 
> https://docs.python.org/3/reference/datamodel.html#object.__repr__
> 
> Like even this
> 
> >>> Gui.ActiveDocument
> <GUI Document object at 0x5791140ea400>
> 
> If __repr__ was implemented, it would be
> 
> >>> Gui.ActiveDocument
> <GUI Document MyNewTable>
> 
> Really bad user experience
> 

We also found out that `Gui.Control().activeDialog()` only returns true or false instead of a python representation of the dialog to interact with and `Control().getActiveTaskDialog()` will only ever give "Task Dialog" as a string - probably because no dialog ever overrides the default for this.

This issue could act as a meta-issue for general improvements and housecleaning of the Python API in case more are found - in any way this is post 1.0 work and not super urgent.

### Full version info

```shell
all current versions
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

benj5378, 2024-10-06:
`Gui.Control.activeTaskDialog()` returns `Task Dialog` as `__repr__` for the class. It would be much better if it returned `<Task Dialog Sketcher Edit>` for instance

benj5378, 2025-02-07:
@tritao, is this something for you? Preferably, compiltion/tests should fail if no implemented __repr__

luzpaz, 2025-09-30:
Hi @pyro9, any interest in taking this on? Since your FreeCAD tutorials heavily utilize the python console these days... this could really soup up the python UX ? No pressure, was just wondering if you'd be interested. 😅

mnesarco, 2025-09-30:
> cc [@mnesarco](https://github.com/mnesarco) as well, who has been looking into things in this area.

Yeah another area to improve..... 1.2 will have some progress with python integration to improve QOL for devs... there are so many areas that need some care.

luzpaz, 2025-09-30:
Can someone show an example of the correct use of `__repr__` per this ticket in the FreeCAD source code ? 

We should also create a link on the developer handbook repo for devs to start incorporating it

pyro9, 2025-10-01:
@luzpaz Yes, I'd like to take a look at this. Both for the Python representation of some core objects and to get FeaturePython to allow and use a __repr__ method from the Python side.

pyro9, 2025-10-03:
@luzpaz So far I've just made a few __repr__ more informative. I'm still peeling the onion on this one.

chennes, 2025-10-03:
(this shouldn't have been auto-closed, this is a pretty large work-in-progress... reopened, but let me know if there's a more useful way to track this work, @pyro9)

PR #24376 (2025-10-03): Repr fix

Make the repr for Document indicate the Name and Label of the Document
Make the repr for the Gui Document show the associated Document Name
repr for a task dialog shows the associated Document name. Also formatted consistent with Python as in:
<Task Dialog for 'Test3' >

fixes #17059