FreeCAD/FreeCAD #17061: Some settings inside preferences are not updated on startup

byteofsoren, 2024-10-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It seams like certain parameters inside the preferences dialog are not updated while FreeCAD starts.
I have found two such issues, both related to the tree view but perhaps there are more off them.

How to reproduce:
Try change the two following settings to some thing else then default.

- preferences -> Display -> Advanced -> Tree View - FontSize
- preferences -> Display -> UI -> Tree View - Icon size

(The last one wont change the icon size but I think that is a know issue)
Then in the preferences dialog press the OK button.
Close FreeCAD any way you want and then start FreeCAD again.

Now when you start FreeCAD again the text size or the icon size are not sett to what you wanted but to the default sizes.
However if you open the preferences dialog then the value the user has set is still present but pressing the OK button is not going to update the tree view text size because it has not changed.

Right now the possible work around is to open the preferences and change the tree view font size and press apply change it and then press OK and do that every time you restart FreeCAD.
  

### Full version info

```shell
OS: Arch Linux (awesome)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git) AppImage
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * kicadStepUpMod 11.3.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

byteofsoren, 2024-10-14:
> Anyone able to reproduce this ?

Is this perhaps an issue on my side?

philjynx, 2024-10-27:
Yes, same here. Lord knows how the search facility works on github, I just posted the same two issues as I could not find them. They've just both been closed as dupicate.
What I cannot tell from your posts, or mine, or the email I received saying closed as duplicate is whether there is any developer interest in these issues.

luca259, 2024-11-08:
Same issue. 

```
OS: macOS 13.7
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * ProDarkThemePreferencePack 1.0.0
  * Nodes 0.1.36
  * Dracula 0.0.5
  * fasteners 0.5.29
  * CurvedShapes 1.0.12
  * sheetmetal 0.2.60
  * ExplodedAssembly
  * Help 1.0.3
  * Curves 0.6.51

```

FreeWilly2022, 2025-06-20:
I tried this in both 1.0.1 and 1.1 dev. The `preferences -> Display -> UI -> Tree View` - Icon size works in both as expected.

The `preferences -> Display -> Advanced -> Tree View` - FontSize sometimes updates immediately in 1.1 dev but sometimes not, but does update in 1.0.1. But when I try to reset to default it did not reset until I restarted FreeCAD.

```
OS: macOS 15.5
Architecture: arm64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * FusedFilamentDesign 0.25.220
```
 
```
OS: macOS 15.5
Architecture: arm64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/06/09 17:51:59
Build type: Release
Branch: main
Hash: 12b4610b499a9756f19bbb85f59fe16e5fa135db
Python 3.11.13, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: OpenSCAD/Free Turntable/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 72/93.5/1
Installed mods: 
  * FusedFilamentDesign 0.25.220
```