FreeCAD/FreeCAD #17062: Crash

kotvkvante, 2024-10-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Placing single point in new Sketch. 
2. Leave Sketch 
3. Crash
 

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38115 (Git)
Build type: Release
Branch: main
Hash: c4e55e3855b8a39543371a775ffdbf6153d36c2b
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Sketcher

### Anything else?

https://youtu.be/Zw9Rwpzrb_E

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-10-07:
This build is quite outdated, please try in the newest one from here: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

It shouldn't happen there.

luzpaz, 2024-10-07:
Please download the weekly build. If you can still reproduce then comment on this issue.

closed 2024-10-07 by luzpaz