FreeCAD/FreeCAD #17066: TechDraw: TechDraw_ActiveView - units mismatch in crop width and crop height

furgo16, 2024-10-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Forum thread: https://forum.freecad.org/viewtopic.php?p=765757#p765757

When using the [TechDraw_ActiveView](https://wiki.freecad.org/TechDraw_ActiveView) command to insert a snapshot (bitmap) of the current 3D view, the image size is hardcoded to 1280 x 1024 px.

1. The default crop width and crop height in the dialog are the default image's x, y resolution (1280 x 1024 px) divided by 10 (128.00 x 102.40 mm). Units are then mm, without any conversion. This seems to be the first error.
   [image]
1. The (cropped) Width, Height properties have no units when viewed through the property editor:
   [image]

@WandererFan's comment from that thread:

> In the UI, the image size fields are QuantitySpinBoxes. Not sure why this change was done, but the corresponding Properties were not changed from PropertyFloat to PropertyLength, so now we have a mis-match between the UI and the Properties - one with units, and one without. This will have to be sorted out.
>
> [...] I need to study the options in the extended ui. There is something wonky in the centering of the image (I think).

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git) Snap 1157
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * ezydxf
  * Ondsel-Lens 2024.9.13.02
  * Assembly4 0.50.15
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2026-01-03:
Still an issue. This might be a good first issue for someone starting in the project.

FEA-eng, 2026-10-17:
Still relevant

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261013 (Git shallow)
Build date: 2026/10/13 19:55:30
Build type: Release
Branch: (HEAD detached at 50daee777)
Hash: 50daee777321b5085accd1f57b57f1a7c1781cae
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 617.14/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```

ZazaXinCi, 2026-10-24:
Hi, I'd like to take this issue. The ActiveView crop dialog shows millimetres, but Width and Height are plain floats with no unit, so the property editor shows bare numbers. My plan is to switch them to PropertyLength, keep the stored values unchanged, and add the migration for old documents. Can I take this issue?

WandererFan, 2026-10-24:
You can certainly take this issue.  I've assigned it to you.