FreeCAD/FreeCAD #17070: Sketcher: Can no longer create a sketch from three points

theosib, 2024-10-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm trying to create a sketch defined by three points. In the screenshot, you can see the three points I'm trying to use, marked in green. When I select these and click the sketch button, nothing happens.

[image]
[rhombicuboctahedron.FCStd.zip](https://github.com/user-attachments/files/17280279/rhombicuboctahedron.FCStd.zip)

### Full version info

```shell
OS: macOS 14.7
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git)
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * fasteners 0.5.29
  * Curves 0.6.44
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

theosib, 2024-10-07:
It doesn't look like the attachment worked right. I'll try again here.
[rhombicuboctahedron.FCStd.zip](https://github.com/user-attachments/files/17280312/rhombicuboctahedron.FCStd.zip)

theosib, 2024-10-07:
I tried fusing the objects together and set "refine" to true. After that, I was able to make a sketch. So I have a workaround, but I think there's still a regression.

Roy-043, 2024-10-07:
The issue is caused by the use of Links. It also occurs in V0,21.

Roy-043, 2024-10-07:
May be a duplicate of #17022.

theosib, 2025-08-11:
I did a quick test where I made a cube and some links to the cube, selected three points, and attached a sketch. It worked!