FreeCAD/FreeCAD #17071: BIM: Recompute issue related to Roof subtraction

Roy-043, 2024-10-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the attached file a roof has been subtracted from two wall. But the result is only correct for one of the walls. Getting the other wall to update requires a certain recompute sequence.

1. Open the file.
2. Select "OG Wände" ("2nd floor walls").
3. Right-click and select "Recompute object" from the Tree view context menu.
4. The roof is now correctly subtracted. Also openings for widows and doors show.
5. Invoke Std_Refresh to recompute the objects that now require this.
6. Result: Everything seems OK now.
7. Right-click the document and select "Mark to recompute".
8. Invoke Std_Refresh.
9. Result: There is a "OG Wände has no solid" warning which seems incorrect.

If you just open the file and only follow steps 7 and 8 "OG Wände" does not update.

File (remove .zip):
[Dachtest2.FCStd.zip](https://github.com/user-attachments/files/17280188/Dachtest2.FCStd.zip)

Forum post:
https://forum.freecad.org/viewtopic.php?p=782628&sid=2a377495891004b423aa00946b9cb970#p782628

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

paullee0, 2024-10-07:
Confirmed.  Roof do have some recompute() problem.

Recompute the Roof first, then the Wall, there is no more problem.

kaiwas, 2024-10-07:
I use the ForceRecompute macro and its upgrade, which recalculates selected objects.
I have already gotten into the habit of recalculating the walls (and the roof)
```

import FreeCADGui as Gui

objs = Gui.Selection.getSelection()
for obj in objs:
    obj.recompute()
```
[image]

kaiwas, 2024-10-07:
The error occurs even if you change the order.
try recalculating the roof at the first stage.
After that, the wall can no longer be repaired.

https://github.com/user-attachments/assets/0840f94c-b2d2-45ea-bd7b-c93d2700c42b

I am not completely sure, but I think the problem is in the sketch from which the walls are made. Perhaps it is not closed.

I have been told many times to make sketches as simple as possible, and this is the very case where I would like to agree with this.

kaiwas, 2024-10-08:
Surprisingly, despite the error, this object (not a solid) has volume. 
[image]

yorikvanhavre, 2024-10-08:
OCCT might still be able to calculate a volume from a non-solid object. 

But even after the "OG Wände has no solid" message, The shape is still solid:
```python
shp.Solids
[<Solid object at 0x55e8f2889380>]
```
The message is printed at https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/BIM/ArchComponent.py#L951 but it shouldn't pop up in that case, because we have a solid... Maybe this message was printed at an intermediary stage?