FreeCAD/FreeCAD #17072: Sketcher: Polyline attached to external geometry results in "wire not closed"

theosib, 2024-10-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As I'm sure you know, "wire not closed" is a plague. I just wanted to provide you with an example of a very simple case where attaching to external geometry results in a closed wire that Part thinks is not closed.

What I did was import the three points that you see in the screenshot. Then I made a polyline to close them into a triangle. Part workbench thinks it's not closed. What I think is happening is that the start and end points are both attached to the same external geometry point, but the extruder is not smart enough to understand that this means they're also indirectly coincident with each other.

[image]
[rhombicuboctahedron.FCStd.zip](https://github.com/user-attachments/files/17280504/rhombicuboctahedron.FCStd.zip)

### Full version info

```shell
OS: macOS 14.7
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git)
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * fasteners 0.5.29
  * Curves 0.6.44
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

theosib, 2024-10-07:
The obvious workaround is to draw a triangle that is not attached to the external points and then make each triangle vertex coincident with a corresponding external point.

theosib, 2024-10-08:
I loaded this design into 0.21.2, and it has the same problem.

I have actually tinkered with trying to trick the sketcher into not closing a wire by making endpoints indirectly coincident. For instance, I can make an open polyline with four segments and then constrain the segments until the open endpoints land on top of each other. FreeCAD is actually surprisingly good at detecting these cases and realizing that the line is indirectly closed.

So I was very surprised that it didn't close the wire in this case since the open endpoints are both constrained to the very same external geometry point. This should be much easier to detect as closed than the example I just described above.

BTW, it seems like my attachment didn't work or something. I don't see a download link in the original post, so I'm attaching it here.
[rhombicuboctahedron.FCStd.zip](https://github.com/user-attachments/files/17299813/rhombicuboctahedron.FCStd.zip)

mwganson, 2024-10-10:
There is a sketcher bug that is intermittent where wires are incorrectly shown not to be closed.  I have fond that by moving some of the edges or vertices it can be resolved, but this could be challenging to do with links to external geometry since they are fixed.  Perhaps try deleting one constraint, moving that edge, and then reapplying the coincident constraint.

bereldhuin, 2024-12-03:
Hi,

I have exactly the same problem (Freecad 1.1.0dev rev 39416 on Ubuntu 24.04).
The problem appears in the last sketch (Sketch0002) for which I cannot create a pocket.

[Ex770.zip](https://github.com/user-attachments/files/17997503/Ex770.zip)

mwganson, 2024-12-03:
You must toggle the external geometry to construction mode.  Why is this normal mode by default?

maxwxyz, 2024-12-04:
@mwganson to be consistent with the other tools. The point tool was switched too a few months ago.
Is there still a bug (initial issue) or can it be closed?

mwganson, 2024-12-04:
Defaulting to normal will create too much confusion because it is a departure from existing behavior.   Watch how many "Why can't I pad this sketch?" questions appear on the help board.

In addition to creating confusion, the most common use case for external geometry is to be a reference and not for use as normal geometry.  Defaults should be based on most common use case.  New points should be construction mode by default, too, if they're not already.

MattN92705, 2025-06-06:
Can confirm this bug still exists in latest dev version (42006).

Simple test - sketch a rectangle, extrude it.  add another sketch, pull in reference to one side of the first rectangle, draw a rectangle constrained to that segment.  attempt to extrude results in "wire not closed".

[image]

OS: macOS 15.5
Architecture: arm64
Version: 1.1.0dev.42006 (Git)
Build date: 2025/06/02 16:32:08
Build type: Release
Branch: main
Hash: db1b5a877314f8d21514680421649fd9dc65acde
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Rounded Arcball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 72/108/1
Installed mods: 
  * OpenTheme 2025.5.20
  * lattice2 1.0.0
  * Telemetry 1.0.1
  * Curves 0.6.63

maxwxyz, 2025-06-06:
@MattN92705 same solution as here https://github.com/FreeCAD/FreeCAD/issues/17072#issuecomment-2515772530
Has nothing to do with this issue.

Luciniki, 2025-06-15:
> [@mwganson](https://github.com/mwganson) to be consistent with the other tools. The point tool was switched too a few months ago. Is there still a bug (initial issue) or can it be closed?

I would suggest changing at least the color of the external geometry to distinguish whether it is of a constructive or reference type.