FreeCAD/FreeCAD #17074: Sketcher, create slot, failure with constraints.

thomaselo, 2024-10-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the RC! and RC2 and the latest weekly builds, when creating a slot in Sketcher, the resulting slot is either overconstrained or underconstrained and results in an error when closing the sketch.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Danish/Denmark (da_DK)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-10-07:
Strange. I can draw a fully constrained slot using OVP but it fails to pad with "Wire is not closed".

[slot.FCStd.zip](https://github.com/user-attachments/files/17282723/slot.FCStd.zip)

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

Krypto31337, 2024-10-07:
With self-drawn slots, I can't reproduce either of these using OVP for dimensions or performing a dimension with the tools later.

With FEA-eng's drawing, if I do nothing to the sketch I get the same error during pad.  If anything is done to the sketch, it's fine. The first time, I constrained the slot against an axis and it padded fine. Restarted again and this time just TRIED to move the already constrained sketch and that fixed it for a pad.

OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git) AppImage
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * sheetmetal 0.4.26
  * freecad.gears 1.3.0

Roy-043, 2024-10-08:
Related: #16239.

FEA-eng, 2024-10-15:
@maxwxyz This is pretty bad for 1.0. Can it be a blocker?

wwmayer, 2024-10-15:
A slot now is a compound of four wires. Each wire has a single edge but these edges don't share common vertexes. This means the slot has eight open vertexes.

A correct slot should be a wire of four edges and two adjacent edges should share their vertexes.

wwmayer, 2024-10-15:
The broken part is TopoShape::makeElementWires. The success of the function seems to depend on the position or size of the edges. Sometimes it's able to form a single wire and sometimes it's not able to realize that two edges are connected and then it starts to create a wire for every single edge. The latter then leads to non-closed wires that can't be padded.

wwmayer, 2024-10-15:
Creating a single wire fails because the distance between two end points is slightly too high so that `BRepBuilderAPI_MakeWire` refuses to add an edge. So, the problem must be the geometry creation somewhere.

wwmayer, 2024-10-15:
Somewhere the shape creation in `SketchObject::buildShape()` is flawed. When I add there the old method from v0.21 then the shape is correctly created.
```
void SketchObject::buildShape()
{
    auto shape = solvedSketch.toShape();
    Shape.setValue(shape);
}
```

wwmayer, 2024-10-15:
Got it!

There are two geometry sets used by the sketcher.

The class SketchObject stores a list of geometries as they are created by the user. These geometries cannot be used for the shape creation.

The class Sketch keeps a separate list with the output geometries after the solver was running on the input geometries and the constraints. These are the correct geometries to create the shape.

PaddleStroke, 2024-10-16:
So this issue is now closed?

wwmayer, 2024-10-16:
thomaselo should still confirm that  the issue is fixed.

luzpaz, 2024-10-16:
cc @thomaselo

maxwxyz, 2024-10-17:
Dots this also fix other issues with overconstraint sketches by autoconstraint or is this limited to the slot feature?

FEA-eng, 2024-10-19:
Creating a slot with OVP seems to work normally now but try this with _Auto remove redundant_ enabled:
1. Create a new sketch.
2. Draw a slot manually selecting the origin as the first point and some location on the X axis as the second point (the third one doesn't matter).
3. Sketch is underconstrained. Exit it. Now it's reported as overconstrained.

maxwxyz, 2024-10-19:
Removed blocker as discussed in today's developer meeting.
Is the leftover here now the same as: https://github.com/FreeCAD/FreeCAD/issues/15797 ?

thomaselo, 2024-10-22:
I now testet create slot in newest weekly build 39024 and could not 
replicate the bug.

Den 16-10-2024 kl. 17:13 skrev wwmayer:

>
> thomaselo should still confirm that the issue is fixed.
--------------DeoQJ0bk8uAoZroHB0cwqrib
Content-Type: text/html; charset="UTF-8"
Content-Transfer-Encoding: 8bit

<!DOCTYPE html>
<html>
  <head>
    <meta http-equiv="Content-Type" content="text/html; charset=UTF-8">
  </head>
  <body>
    <p>I now testet create slot in newest weekly build 39024 and could
      not replicate the bug.<br>
    </p>
    <div class="moz-cite-prefix">Den 16-10-2024 kl. 17:13 skrev wwmayer:<br>
    </div>
    <blockquote type="cite"
***@***.***">
      <meta http-equiv="Content-Type" content="text/html; charset=UTF-8">
      <p dir="auto">thomaselo should still confirm that the issue is
        fixed.</p>
      <p
style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br>
href="https://github.com/FreeCAD/FreeCAD/issues/17074#issuecomment-2417127081"
          moz-do-not-send="true">view it on GitHub</a>, or <a
          moz-do-not-send="true">unsubscribe</a>.<br>
src="https://github.com/notifications/beacon/ACBPPPU4QPNWO7LQGNF6NYDZ3Z67ZA5CNFSM6AAAAABPQG5HS6WGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTUQCJXKS.gif"
          height="1" width="1" alt="" moz-do-not-send="true"><span
style="color: transparent; font-size: 0; display: none; visibility: hidden; overflow: hidden; opacity: 0; width: 0; height: 0; max-width: 0; max-height: 0; mso-hide: all">Message
          ID: <span>&lt;FreeCAD/FreeCAD/issues/17074/2417127081</span><span>@</span><span>github</span><span>.</span><span>com&gt;</span></span></p>
      <script type="application/ld+json">[
{
"target": "https://github.com/FreeCAD/FreeCAD/issues/17074#issuecomment-2417127081",
"url": "https://github.com/FreeCAD/FreeCAD/issues/17074#issuecomment-2417127081",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]</script>
    </blockquote>
  </body>
</html>

--------------DeoQJ0bk8uAoZroHB0cwqrib--

luzpaz, 2024-10-22:
Thanks @thomaselo, closing. 
Note: FEA-eng's remainder bug is actually #15797

wwmayer, 2024-10-22:
Unfortunately, it's not fixed. FEA-eng posted a reliable way to get the over-constrained issue. I can reproduce it with the latest version from today: 0b0bb183236