FreeCAD/FreeCAD #17075: Improvements to VarSets UI/UX in the expression editor

kriswilk, 2024-10-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The new VarSet UI within the expression editor is a good start. Issue #15555 touches on a bug in the current implementation. I'd like to open a more general discussion about possible changes to this UI.

I suggest the following COSMETIC changes to this dialog:

- Change "New Property" to "Name" to be consistent with the "Add Property" dialog.
- Change "Show variable sets" to "Add to variable set..." to clarify the action that this UI takes.
- Move this check box beside the OK/Cancel buttons.
- Set the default state of the checkbox to UNCHECKED. It currently remembers its state, which I think is undesirable.
- When checked, cause the new fields to pop-in BELOW, preserving the location of the expression entry box.
- Remove the "Variable Sets" group box label.
- Remove the "Info" section completely. It doesn't indicate anything that the field names/contents don't already. Furthermore, it's not internationalized.

I suggest the following FUNCTIONAL changes to this dialog:

- The field "Variable Set" should be pre-populated:
    - The last populated value, OR
    - If using the dialog for the first time, the first available VarSet (often there's only one).
- The field "Group" should be pre-populated:
    - The last populated value, OR
    - If using the dialog for the first time, the first available group from the pre-populated VarSet (often just "Base").
- When the check mark is activated, the focus should go immediately to the "Name" field for quick entry.
- Ideally, offer to prefix the group name as in the "Add Property" dialog.

I've compiled a quick proof-of-concept of the COSMETIC changes only. If others are amenable to these changes, I can tidy up the code and submit a PR.

@pieterhijma is probably most qualified to make the functional changes if he has the time to do so. I'll be looking at the code and could take it on if I get more familiar with it...

https://github.com/user-attachments/assets/ff2f8a99-fe59-49c1-9adb-b9dc329ade3a

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kriswilk, 2024-10-08:
I've added screenshots of the existing and proposed UIs. I will copy them here for completeness...

Existing:

[image]

Proposed:

[image]

kriswilk, 2024-10-08:
One thing to be aware of is that the existing UI presents warnings/errors in the "Info" area. For example if the name is missing or invalid.

With the proposed simplified UI, these warnings would need to be converted to a message box on "OK". Or there could be other visual cues such as red text (see how Spreadsheet aliases do it), tooltips, etc...

pieterhijma, 2025-04-02:
I like the suggestions. @FreeCAD/design-working-group, could you take a look at this as well?

luzpaz, 2025-04-08:
This is great feedback, the problem is there are way too many issues being addressed in the ticket so it's hard to follow and track.

closed 2025-08-27 by kadet1090