FreeCAD/FreeCAD #17076: External SMESH dependency

bkmgit, 2024-10-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD has an embedded version of [SMESH](), but there are a number of other versions available online, for example:
- https://github.com/SalomePlatform/smesh
- https://github.com/trelau/smesh which does not build for OCCT 7.8 https://github.com/trelau/SMESH/issues/77

Should the bundled version of Smesh be considered as a separate one from the above two or is compatibility with one of these upstreams planned?  This would help with packaging of FreeCAD dependencies on linux distributions.

### Full version info

```shell
Current head of git repository https://github.com/FreeCAD/FreeCAD/commit/8078e77169e3f78c1f7d14362c8e3b4c4239918a
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-10-07:
@marioalexis84 and @looooo FYI

marioalexis84, 2024-10-08:
> This would help with packaging of FreeCAD dependencies on linux distributions.

FreeCAD should compile with version 9 of SMESH, see: https://github.com/FreeCAD/FreeCAD/commit/135525a8066703059603be1a395986327e635568

> I think to switch to the freecad internal smesh in the future 

Ok @looooo, I will start preparing the code to migrate the objects that use the old implementation to the new one.

looooo, 2024-10-08:
@marioalexis84 that's good. But please allow the old interface to exist for some time. The transition for the conda-forge packages will need some time and I don't think this will happen very soon. Just speaking about the target for the next release which comes after 1.0.

marioalexis84, 2024-11-19:
@maxwxyz  I think this can be closed

closed 2024-11-20 by maxwxyz