FreeCAD/FreeCAD #17078: [PartDesign] Fillet and Chamfer makes part loose the color

domi4484, 2024-10-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Draw a part
2. Change the color (for example with `Random color`)
3. Apply `Fillet` or `Chamfer` to an edge
4. Note that color is changed back to normal gray

[image]

### Full version info

```shell
OS: openSUSE Tumbleweed (KDE/plasma6)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38922 (Git)
Build type: Debug
Branch: main
Hash: 8078e77169e3f78c1f7d14362c8e3b4c4239918a
Python 3.11.10, Qt 5.15.15, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * sheetmetal 0.4.18 (Disabled)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-10-07:
Duplicate of or at least related to #15503 ?

domi4484, 2024-10-07:
@FEA-eng probably yes by looking at other closed duplicates. Tagging 15503 with `Topic: Color` (and maybe rename a bit) would probably help finding it.

closed 2024-10-07 by domi4484