FreeCAD/FreeCAD #17081: Sketcher appears to delete all constraints when accidentally deleting external geometry reference then pressing undo

AnyOldName3, 2024-10-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Sometimes, undoing the deletion of an external geometry reference in a sketch will remove all constraints. They may come back when closing and reopening the sketch.

Repro steps:
* Press *Create New* on the Start page.
* Go to the Part Design workbench.
* Create a body.
* Create a sketch.
* Draw and constrain a circle.
* Close the sketch.
* Create another sketch in the same plane.
* Add the circle as an external geometry reference.
* Draw a rectangle whose corners lie on the circle.
* That leaves 1 DoF, so constrain one of the lengths.
* Delete the circle external geometry reference.
* Press Ctrl+Z for undo.
* Observe that the circle external geometry reference is back, but all constraints are gone, i.e. the *Constraints* list is empty, and you've got 16 DoF(s).
* Close the sketch.
* Open it again.
* Hopefully observe that the constraints are back.

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
[/code]
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AIRCAP, 2024-10-07:
there also seems to be a possibly related issue where when you have a broken sketch and you "delete external reference constraints" in the diagnosis tool, it actually removes all constraints.

luzpaz, 2024-10-14:
@AnyOldName3 can you reproduce on RC2 ? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc2

closed 2024-10-14 by luzpaz