FreeCAD/FreeCAD #17083: [CAM] Can't cancel out of the file system dialog when choosing ToolBit Library editor

brandoneubank, 2024-10-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When selecting the ToolBit Library editor, the filesystem cancel button closes the window, but it pops up again. The menubar options seem to work, but clicking int he UI returns the system error sound and doesn't remove the open file dialog.

### Full version info

```shell
OS: macOS 15.0
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
Installed mods: 
  * ExtremeProDark 2.9.0 (Disabled)
  * sheetmetal 0.3.16
  * ExtremeProLight 1.6.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-14:
Hi @opensourcecnc, could  you reproduce this issues?

opensourcecnc, 2024-10-14:
> Hi @opensourcecnc, could you reproduce this issues?

I'll take look

luzpaz, 2024-10-14:
@johnocfii would you mind testing on macOS ? TIA

JohnOCFII, 2024-10-14:
> @JohnOCFII would you mind testing on macOS ? TIA

I tried, but you can call the ToolBit Library editor in a number of different ways -- and I didn't see this particular problem.   I have seen issues that seemed to vary (Mac specific) based on the location of the Tools directories, which I've often wondered if they were Mac file access permissions related.

If the OP could enter the specific steps taken to enter the Tool Bit editor, I might be able to replicate directly.

JohnOCFII, 2024-10-15:
Thanks for the additional info.  I was able to recreate the issue.  I have seen this before myself (and could have sworn I opened an issue, but can't find it).

```
OS: macOS 15.0.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38946 (Git)
Build type: Release
Branch: main
Hash: e478192aa5393de9b7df190341a043f79ec6ac49
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

Adding some screen shots to tell the story:

- Launch FreeCAD
- Enter CAM workbench
- Select ToolBit Dock from menu bar:

[image]

- This dialog box appears (with a very generic question mark icon)
[image]

- Select "Yes" and this operating system file system dialog box appears.  I believe when I tested this on Linux, there are text instructions here -- but no instruction here on macOS.  If the user selects `Cancel` nothing happens.  Separately (when I had MY issue here) I wasn't sure if I needed to enter a new directory name, etc.  The default expected answer is to select `Open` but that isn't really what I'd expect.  

[image]

- After figuring out that I needed other press `Open` I am presented with a series of dialog boxes, showing what is happening.

[image]

xtemp09, 2025-01-12:
The commit https://github.com/FreeCAD/FreeCAD/commit/c1202c3afaaa71a7ea1383af13efd6b53646c108 introduces the infinite loop.

			
	
https://github.com/FreeCAD/FreeCAD/blob/7829fcc4888bce067fe002a4d857a4953d2482fc/src/Mod/CAM/Path/Tool/Gui/BitLibrary.py#L87-L91

For strange reason, I cannot find infinite loops anywhere in *.py files of _FreeCAD_.

JohnOCFII, 2025-01-12:
> * Select "Yes" and this operating system file system dialog box appears.  I believe when I tested this on Linux, there are text instructions here -- but no instruction here on macOS.  If the user [...]

FYI -- this comment is related to this (recently added) issue: https://github.com/FreeCAD/FreeCAD/issues/18887

xtemp09, 2025-01-13:
@sliptonic, can I propose an idea of remaking this? I think that in this case wizard is the best workflow[^1]. If the paths are interdependent, then a detailslist[^2] or  a window with line edits are likely to be a better choice.

[^1]: https://learn.microsoft.com/en-us/power-platform/guidance/creator-kit/subwaynav
[^2]: https://learn.microsoft.com/en-us/power-platform/guidance/creator-kit/detailslist

I would do it in _C++/Qt_, but this code is in _Python_, I don't know how to replace or modify that.

PR #21425 (2025-06-02): CAM: Integration of the tool and shape editors from BTL (Better Tool Library)
This very large PR essentially replaces CAM Tool Management completely, as part of the [Better Tool Library](https://github.com/knipknap/better-tool-library/) integration.

## Relevant Issues

Discussion: #9466

I also went through all open CAM issues to find any that may be resolved or otherwise affected:

- Resolves: #8201
- Resolves: #8526
- Resolves: #9625
- Resolves: #11522
- Resolves: #12712
- Likely resolves: #12823, but should be tested to be sure
- Possibly resolves #13603, to be tested
- Resolves: #17083
-  [... cut]