FreeCAD/FreeCAD #17085: Extrusion to face can only be done in a certain order

herrdeh, 2024-10-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Given is a feature "Verst Rohr". It needs to get an interior enforcement "Pad" through the whole part. This can only created, if "Skz00" is padded to face6 and Skz01 is padded to face7 - as done in file "bugverd01-working".

Attempts to create it vice-versa (Skz00 padded to face7 and Skz01 padded to face6) result in odd failures, with only fractions of the whole body available for further proceedings.

Using a (partial) shape binder does not work at all, throwing "Must not intersect sketch".

Cheers,
Wolf

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git)
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/Classic/Qt default
Installed mods: 
  * 3D_Printing_Tools
  * Assembly4 0.50.12
  * dodo 1.0.1
  * ExtremeProLight 1.6.0
  * Curves 0.6.29
  * BOLTSFC
  * nurbs
  * animation
  * IconThemes
  * lattice2 1.0.0
  * ProDarkThemePreferencePack 1.0.0 (Disabled)
```

### Subproject(s) affected?

None

### Anything else?

This is somehow related to bug16175, as the given solution is a workaround for a not-present "pad-to-both-directions" feature.

[wd.zip](https://github.com/user-attachments/files/17286953/wd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2024-10-09:
Confirmed but I can't find the cause. UpToShape works well with 2 faces or all faces.

AIRCAP, 2024-10-10:
A workaround is: don't use the sketch to pad, make a subshapebinder on the sketch and then use that

why?  I don't know

herrdeh, 2024-10-10:
> but subshapebinders magically fix a lot of things - for example if you make
> a section of a part and the wire is not closed - make a subshapebinder of
> it and that will be considered closed.

I tried that first, but it didn't work with a sub shape binder.

FlachyJoe, 2024-10-10:
I'm working on a double side up-to-shape which solve this issue and more.

herrdeh, 2024-10-10:
Great, that's really good news!

Thanks to everybody who puts effort into FreeCAD and made it as impressive
as it already is.

AIRCAP, 2024-10-10:
yeah its weird. it worked with the subshape binder - until i triggered a complete recompute of the document, when it again didn't work. 
on the pro side, I can trigger the bug also with a fixed dimension sketch, it's not just up to face that has this

AIRCAP, 2024-10-10:
I tried to reproduce it with a simpler part - the same thing happens without any "toFace" - these are two simpler one dimensional fixed length pads with much simpler sketches 
[bugverd03-fail.zip](https://github.com/user-attachments/files/17333854/bugverd03-fail.zip)
on a simpler pipe (skipped the second loft) and the same thing happens

AIRCAP, 2024-10-10:
@FlachyJoe just tested it with OCCT 7.3 instead of OCCT 7.6 
[image]
on 7.3 the shape does NOT disappear, instead just all surfaces have inverted normals ;)

AIRCAP, 2024-10-10:
the part then also fails BOP checks with "Invalid Tolerance value" -- i think this might be related to tolerances not being set correctly by FreeCAD

FlachyJoe, 2024-10-10:
**Where** are tolerances set by FreeCAD :interrobang:

AIRCAP, 2024-10-11:
wherever in the code a TopoDS_Shape is created by calling a generator function in OCCT you are supposed to specify the tolerances for the surfaces, edges and vertices respectively.  in most places it's not set, so they default to Precision::Confusion() - which depending on the part's size and shape can be too low or too high. if a part is created through python for example in 

https://github.com/FreeCAD/FreeCAD/blob/e478192aa5393de9b7df190341a043f79ec6ac49/src/Mod/Part/App/TopoShapeWirePyImp.cpp#L392-L420

the function at least allows to set tolerances but defaults to the lowest possible value or just a random hardcoded number -- there are probably a hundred or more places in FreeCAD code like this - what is suppose to happen is to set the tolerance to a sensible value based on the size of the shape - a 10km large part cannot have nanometer precision and a micrometer large part should not have mm sized precision, or the underlying OCCT functions will fail - badly (they do)

in general the tolerance values Freecad sets are often too low which is the reason for ~~hacks~~ stopgap measures like this: #17119

AIRCAP, 2024-10-11:
although that hack might still be needed due to actual bugs in OCCT ;)

FlachyJoe, 2024-10-11:
@AIRCAP for reference, if don't see this already: https://opencascade.blogspot.com/2009/02/topology-and-geometry-in-open-cascade_09.html

AIRCAP, 2024-10-17:
Ok, new development. I just checked. When I export the two shapes before and after the pad as separate brep solids, and then use a Part Fuse, the same failure happens - in main

but if a fuzzy value is applied to the fusion, then it succeeds. so this whole thing is indeed a tolerance issue

AIRCAP, 2024-10-17:
[shape1.zip](https://github.com/user-attachments/files/17413845/shape1.zip)
here is the example - if I do the boolean fuse with 0 tolerance (main) it fails, if I do it with fuzzy fusion ( #17119 ) then it succeeds.

The fuzzy patc was supposed to also fix partdesign but the issue is PartDesign uses TopoShape.makeElementBoolean() and that in its header explicitly sets tolerance to 0.0 as a default value - I think I can catch the 0.0 and apply autofuzzy and that should fix this - and possibly many other issues in PartDesign

FlachyJoe, 2024-10-17:
@AIRCAP well done! 2 points:

- you can retrieve the shape computed by the feature with `Part.show(App.ActiveDocument.Pad.AddSubShape)`
- Tolerance and Fuzzy value are two different things in OCC, tolerance is set per sub-element when fuzzy is used for boolean operation. You can mix tolerances in a unique shape but the fuzzy is set for the whole operation.

AIRCAP, 2024-10-17:
> @AIRCAP well done! 2 points:
>
>     * you can retrieve the shape computed by the feature with `Part.show(App.ActiveDocument.Pad.AddSubShape)`
>
>     * Tolerance and Fuzzy value are two different things [...]

yes, but FreeCAD in TopoShape.h actually uses the term "tolerance" for Fizzy setting. - all the boolean functions in Part/App/TopoShape.h have an optional parameter tolerance (defaults to 0.0) which when set actually sets the Fuzzy value for the operation. thats what I meant.  Yes this is not the inherent tolerance inside the shape (but it affects the inherent tolerance in the result if applied)

since part tolerances are set incorrectly (too low) all over FreeCAD, this fuzzy tolerance is needed for booleans to succeed in some not obvious cases.

FlachyJoe, 2024-10-17:
I think we should give more attention to tolerances (the OCC ones). The problem with boolean operation you solve with the fuzzy value could be only the tail of the seaworm.

PR #17119 (2024-10-21): Fix boolean operations
Fix #17159
Fix #17150
Fix #17085
Fix #16081 
Fix #15825
Fix #15615
can alleviate #11083 ( fixes with nonstandard setting for boolean fuzzy > 1000 - but that's too high to cut out very delicate parts from very large blocks )
Fix #9666
Allows workaround for #17338 

OCCT has issues with boolean operations failing when parts are touching but not intersecting. This includes but is not limited to the PartDesign helix , which is notorious for producing results that do not pass BOP tests or even fail completely.

The suspected problem is too narrow precision. Part precision sho [... cut]
closed 2024-10-21 by yorikvanhavre