FreeCAD/FreeCAD #17086: Job creation not possible in the path workbench

GMellar, 2024-10-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After creating a body I opened the path workbench and to create a job. The dialog opens but after selection and confirmation there is the following error: 

> 
> 07:14:12  pyException: Traceback (most recent call last):
>   File "/opt/freecad/Mod/Path/Path/Tool/Gui/Controller.py", line 62, in attach
>     vobj.setEditorMode("ShapeColor", mode)
> <class 'AttributeError'>: Property container has no property 'ShapeColor'
> 07:14:12  Job.ERROR: Property container has no property 'ShapeColor'
> 07:14:12  Traceback (most recent call last):
> 07:14:12    File "/opt/freecad/Mod/Path/Path/Main/Gui/Job.py", line 1644, in Create
>     obj.ViewObject.Proxy = ViewProvider(obj.ViewObject)
>                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> 07:14:12    File "/opt/freecad/Mod/Path/Path/Main/Gui/Job.py", line 97, in __init__
>     vobj.setEditorMode("ShapeColor", mode)
> 07:14:12  AttributeError: Property container has no property 'ShapeColor'

The dialog closes an no job has been created. 

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD losgelöst bei 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * FEM_FrontISTR 0.1.0
  * Assembly4 0.50.15
  * Plot 2022.4.17
  * Silk 0.1.5
  * fasteners 0.5.29
  * CurvedShapes 1.0.10
  * Curves 0.6.48
  * A2plus 0.4.68
  * Ship 2022.4.11
  * Reinforcement
  * sheetmetal 0.4.26
  * OpticsWorkbench 1.0.24
  * toSketch 1.0.1
  * EM 2.1.1
  * CfdOF 1.27.10
  * InventorLoader 1.4.0
  * Nodes 0.1.36
  * FEMbyGEN 2.2.0
  * kicadStepUpMod 11.3.9
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-09:
Can you share a demo file and steps how to reproduce?
Does this work in the stable 0.21.2 version?

luzpaz, 2024-10-09:
@mlampert can you reproduce per chance ?

Syres916, 2024-10-09:
@GMellar there's something specific that you're not telling us about your 1.0RC2 build, why would it have _Path_ workbench when it was replaced by _CAM_ months ago? Even the path to _Path_ nolonger exists _/opt/freecad/Mod/Path/Path_
Please retry with the [AppImage](https://github.com/FreeCAD/FreeCAD/releases/tag/1.0rc2) for your architecture and I'm sure you'll find the error doesn't appear.

luzpaz, 2024-10-09:
@Syres916 nice catch

GMellar, 2024-10-10:
> @GMellar there's something specific that you're not telling us about your 1.0RC2 build, why would it have _Path_ workbench when it was replaced by _CAM_ months ago? Even the path to _Path_ nolonger [...]

The Version was right and it had the Path workbench because I built FreeCAD myself and did make install into the old directory which had the pre 1.0 workbenches.

closed 2024-10-11 by luzpaz