FreeCAD/FreeCAD #17087: [BIM] It is not possible to add a new wall section if a segment is added to a sketch.

kaiwas, 2024-10-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It is impossible to add a new wall section if the wall is built using the wall tool. This problem depends on the direction of building the new section.

1. (Problem) We build a wall and add a segment to the sketch from the bottom up.
2. (Works) We build a wall and add a segment from the top down.
3. (Works) We build a sketch. We make a wall out of it. Further editing does not cause problems for any directions of new sketch segments.
Video demonstration.

https://github.com/user-attachments/assets/d025f819-cabc-4294-85fd-64d07777aef9

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38898 (Git) AppImage
Build type: Release
Branch: main
Hash: 32e09d9554db26414697ed19e0189ccab4ea9029
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * lattice2 1.0.0
  * PitchedRoof
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-10-08:
Ok confirmed...
The weird thing is, if I edit the sketch again, and move the coincident point, then after that it works...

kaiwas, 2024-10-08:
> I'm also unable to reproduce that each time. Sometimes yes, sometimes no

I don't know how to help.
But if you have any ideas on how I can perform additional tests, please let me know.

semhustej, 2024-10-08:
I can confirm the issue too.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: English/United States (en_US) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

```

maxwxyz, 2024-10-08:
Is this a regression from 0.21.2 stable?

kaiwas, 2024-10-08:
> The problem only occurs if the new segment is almost perfectly vertical

probably another example is the roof cutting theme, because there are also a lot of vertical and horizontal lines in the sketcher.

ps. I will be able to test version 0.21.1, but a little later. my computer was taken away by the kids )

semhustej, 2024-10-08:
I confirm that I am not able to reproduce the issue in 0.21.2 in my quick test.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: C/Default (C) [ OS: English/United States (en_US) ]

```

paullee0, 2024-10-09:
@yorikvanhavre 

Something related to your earlier finding and remark in ArchWall.py ? -

https://github.com/FreeCAD/FreeCAD/blob/a9bff78974f301100134a13c636e23aa91be5bcf/src/Mod/BIM/ArchWall.py#L539

paullee0, 2024-10-09:
> This is primarily a Sketcher problem. If no new wall segment is added there is a small gap in the sketch. You can verify this by using [Sketcher_ValidateSketch](https://wiki.freecad.org/Sketcher_ValidateSketch).
> 
> The problem only occurs if the new segment is almost perfectly vertical. If you start from the endpoint of the existing segment and hold down Ctrl (angle snap) while creating the vertical segment it happens every time. If you don't snap you need a steady hand to reproduce consistently.
> 
> Creating a perfectly vertical segment in the other direction is less likely, which explains why the problem does not, or very rarely, occurs then.
> 
> Of course it is contradictory that there is a gap at a corner with a coincident constraint. But we see this 'failure to converge' in other cases as well: perfectly horizontal slot and geometry created on top of external geometry are two recent examples that come to mind. I get the idea that the solver sometimes 'thinks' some geometry is already OK and simply stops too early.
> 
> #16239 #16178 https://forum.freecad.org/viewtopic.php?t=91147

I read a few time but do not fully understand :)

But ArchWall should not fail

kaiwas, 2024-10-09:
> @kaiwas Can you post the 'perfectly' wrong model file, thanks :)

It wasn't difficult )

To check, I opened this file in version 0.21.1 and the wall is also destroyed there.
Remove the zip extension.
[Break_Wall.FCStd.zip](https://github.com/user-attachments/files/17316737/Break_Wall.FCStd.zip)

paullee0, 2024-10-09:
It happens I created one, finding with below codes:-
```

>>> s=Gui.Selection.getSelection()[0]
>>> s.Shape
<Compound object at 0x55ee6732b130>
>>> s.Shape.Wires
[<Wire object at 0x55ee640c81c0>, <Wire object at 0x55ee6743aaa0>]
>>> s.Shape.Edges
[<Edge object at 0x55ee697f5c70>, <Edge object at 0x55ee67545940>]
>>> cs=Part.getSortedClusters(s.Shape.Edges)
>>> cs
[[<Edge object at 0x55ee6753b6b0>, <Edge object at 0x55ee67357ca0>]]

>>> for e in s.Shape.Edges:
...   print( e.Vertexes[0].Point)
...   print( e.Vertexes[1].Point)
... 
Vector (-74.00718688964844, 7.443048000335693, 0.0)
Vector (850.7616577148438, 41.05305480957031, 0.0)
Vector (699.930176, 878.611023, 0.0)
Vector (850.761658, 41.05305499999997, 0.0)
>>> 
```

Sketch return 2 Edges (as expected), 2 Wires (Unexpected, expected 1 wire), 

Part.getSortedClusters return 1 group of Edges (the 2 edges are connected)

Print the X,Y,Z of the 2 edges, note the 2 end points are slightly off !

Vector (-74.00718688964844, 7.443048000335693, 0.0)
Vector (**850.761657**7148438, **41.0530548**0957031, 0.0)
Vector (699.930176, 878.611023, 0.0)
Vector (**850.761658**, **41.0530549**9999997, 0.0)

kaiwas, 2024-10-09:
Maybe the sketcher needs a fix like this? To remove the excessive precision.
https://github.com/FreeCAD/FreeCAD/pull/17119

Why doesn't this happen in 0.21? Were there no gaps or was the construction less precise?

yorikvanhavre, 2024-10-10:
Indeed we can reproduce the problem without BIM so it's a sketcher issue. I'm not sure it has to do with precision, because in @paullee0's example above:

```python
Vector (-74.00718688964844, 7.443048000335693, 0.0)
Vector (850.7616577148438, 41.05305480957031, 0.0)
Vector (699.930176, 878.611023, 0.0)
Vector (850.761658, 41.05305499999997, 0.0)
```
The gap does not seem to be caused by a difference of precision between p1 and p2, but really a different position in space.

I have no idea how to debug this further.. Maybe @AjinkyaDahale or @PaddleStroke have an idea?

paullee0, 2024-10-10:
@yorikvanhavre  Part maybe part of the problem, as it was shown Part.getSortedClusters() reported the seemingly disconnected edges connected (code and result repeated below).

Maybe I should also check the Sketch.Geometry in the Sketcher if they agreed with the Shape output of the Sketch.Shape.Edges.

```
>>> cs=Part.getSortedClusters(s.Shape.Edges)
>>> cs
[[<Edge object at 0x55ee6753b6b0>, <Edge object at 0x55ee67357ca0>]]

```

yorikvanhavre, 2024-10-10:
> Maybe I should also check the Sketch.Geometry in the Sketcher if they agreed with the Shape output of the Sketch.Shape.Edges.

The question is if we're able to build a wire from that cluster. If yes, then we could implement a workaround here...

paullee0, 2024-10-10:
OK, tested.  To answer Yorik's question :
- No, not able to build a wire from cluster
- either by Part.getsortedClusters(s.Shape.Edges)
- nor by Part.getsortedClusters(s.Geometry[0].toShape(), s.Geometry[1].toShape()]
- both return message as ArchWall.py do : "Part.OCCError: BRep_API: command not done"
- both the geometry and edges output by wire indicates they 2 edges are disconnected

So, it seems to me:-
- Sketcher 'end' the coincident constraints algorithm 'too soon' that the 2 ends not yet meet
- Part.getsortedClusters() mistakenly considered those 2 edges as connected
- (But Sketch.Wires return 2 Wires instead of 1 - so 'it' know the coincident constraint is not done yet)

Details of the test -

```
>>> s.Shape.Edges
[<Edge object at 0x564bb59835e0>, <Edge object at 0x564bb6983ea0>]
>>> s.Shape.Wires
[<Wire object at 0x564bb59f34a0>, <Wire object at 0x564bb6974b90>]
>>> cs=Part.getSortedClusters(s.Shape.Edges)
>>> cs
[[<Edge object at 0x564bb59835e0>, <Edge object at 0x564bb59a7ad0>]]
>>> for e in s.Shape.Edges:
...    print( e.Vertexes[0].Point)
...    print( e.Vertexes[1].Point)
... 
Vector (-74.00718688964844, 7.443048000335693, 0.0)
Vector (850.7616577148438, 41.05305480957031, 0.0)
Vector (699.930176, 878.611023, 0.0)
Vector (850.761658, 41.05305499999997, 0.0)
>>> Part.Wire(cs[0])
Traceback (most recent call last):
  File "<input>", line 1, in <module>
Part.OCCError: BRep_API: command not done

>>> s.Geometry
[<Line segment (-74.0072,7.44305,0) (850.762,41.0531,0) >, <Line segment (699.93,878.611,0) (850.762,41.0531,0) >]
>>> sg0=s.Geometry[0]
>>> sg0.StartPoint
Vector (-74.00718688964844, 7.443048000335693, 0.0)
>>> sg0.EndPoint
Vector (850.7616577148438, 41.05305480957031, 0.0)
>>> sg1=s.Geometry[1]
>>> sg1.StartPoint
Vector (699.930176, 878.611023, 0.0)
>>> sg1.EndPoint
Vector (850.761658, 41.05305499999997, 0.0)
>>> 
>>> e0=sg0.toShape()
>>> e1=sg1.toShape()
>>> Part.Wire([e0,e1])
Traceback (most recent call last):
  File "<input>", line 1, in <module>
Part.OCCError: BRep_API: command not done

```

yorikvanhavre, 2024-10-11:
I think we need the help of a Sketcher specialist. I don't see clearly how easily fixable that is, of if we need to start on a workaround

AjinkyaDahale, 2024-10-11:
Looks like it's related to #16904.

kaiwas, 2024-10-13:
I checked the wall construction on the wizard build and it seems the problem is more apparent. (made about 10 attempts)

```
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38946 (Git) AppImage
Build type: Release
Branch: main
Hash: e478192aa5393de9b7df190341a043f79ec6ac49
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * lattice2 1.0.0
  * PitchedRoof
  * OpenTheme 2024.9.1

```

kaiwas, 2024-10-13:
Why does the problem only appear if the wall was created with the wall tool?
If you create a sketch first, and only then a wall - there is no problem. I made more than 10 attempts, but could not get the error.

But as soon as I make a wall, and then correct the sketch - the problem occurs on the first try.

https://github.com/user-attachments/assets/d1e3a198-3cd3-4d0e-8f8a-8bdd29adf3eb

paullee0, 2024-10-13:
> Why does the problem only appear if the wall was created with the wall tool? If you create a sketch first, and only then a wall - there is no problem. I made more than 10 attempts, but could not get the error.
> 
> But as soon as I make a wall, and then correct the sketch - the problem occurs on the first try.
> vokoscreenNG-2024-10-13_11-36-04.mp4
> 
> Maybe somewhere in the code of the wall tool the order of actions should be changed?

At 16s, could you try not moving the 1st edge before you add the 2nd one?  I think moving the any point in the Sketch 'cure' the problem.  Thanks.

kaiwas, 2024-10-13:
> At 16s, could you try not moving the 1st edge before you add the 2nd one? I think moving the any point in the Sketch 'cure' the problem. Thanks.

Yes. I just increased the length of the segment because then the wall would be too short.

Okay. Here are 5 more attempts without moving the sketch (I'm sure if there were 20+ of them, everything would be similar)

The 6th attempt I use the wall tool (the problem immediately arises)

https://github.com/user-attachments/assets/bccb8b21-1fba-49a0-a342-d873263226cc

paullee0, 2024-10-13:
> > At 16s, could you try not moving the 1st edge before you add the 2nd one? I think moving the any point in the Sketch 'cure' the problem. Thanks.
> 
> Yes. I just increased the length of the segment because then the wall would be too short.
> 
> Okay. Here are 5 more attempts without moving the sketch (I'm sure if there were 20+ of them, everything would be similar)
> 
> The 6th attempt I use the wall tool (the problem immediately arises)
> vokoscreenNG-2024-10-13_12-00-25.mp4

Thanks, interesting, though still a Sketcher's problem because under the hood the code just ask Sketcher to add the line.  But your test probably narrow down which code may bring the problem.

@yorikvanhavre, see if this shed some light on this problem.

Thanks all :)

kaiwas, 2024-10-13:
> I never use the ArchWall intuitive tool to draw Wall, just directLY build a (Arch)Sketch to represent a Arch layout with all the constraints then Wall, CurtainWall, Slab etc. so never experience this problem :LOL

In version 0.21 this problem did not occur.
I also always build a sketch first, and in fact I stumbled upon the problem by accident when we were discussing on the forum "Arch_Remove - is this a toponaming problem?"

yorikvanhavre, 2024-10-15:
> @yorikvanhavre, see if this shed some light on this problem.

Not really... If you create a wall with the wall tool, then a sketch with only one line, I can't see any difference between the two sketches. I don't know what we should do different here...

As for the way to "cure" the sketch, I see that too, but it just looks to me this forces a recalculation of the entire sketch, and I can't find a way to trigger that from the outside...

Roy-043, 2024-10-16:
May be solved by #17273?

kaiwas, 2024-10-19:
> May be solved by #17273?

Checked in weekly build 38991 (October 19th). The issue is not resolved.

paullee0, 2024-10-19:
Checked git 38991 also, on Fedora 40, not fixed unfortunately :)

```
OS: Fedora Linux 40 (Workstation Edition) (GNOME/gnome/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git) AppImage
Build type: Release
Branch: main
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Link to SketchArchLite

```

yorikvanhavre, 2024-10-21:
I cannot reproduce the problem anymore with current main branch code.

luzpaz, 2024-10-21:
@yorikvanhavre About info for posterity ?

kaiwas, 2024-10-22:
I confirm. The issue is not detected in today's build.

https://github.com/user-attachments/assets/ff7796fd-3922-408b-aee0-b16271ca215b

BUT! If I use SketchArch, the issue still reproduces. @paullee0 

https://github.com/user-attachments/assets/c55478d2-6cc2-4f50-89ac-9547e911407e

```
OS: Manjaro Linux (KDE/plasma/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git) AppImage
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * lattice2 1.0.0
  * PitchedRoof
  * OpenTheme 2024.9.1
```

yorikvanhavre, 2024-10-22:
Closing then as it's confirmed fixed by 2 people :)

paullee0, 2024-10-22:
> I confirm. The issue is not detected in today's build.
> vokoscreenNG-2024-10-22_08-45-25.mp4
> 
> BUT! If I use SketchArch, the issue still reproduces. @paullee0
> vokoscreenNG-2024-10-22_08-43-02.mp4
> 

Unfortunately here, not ok even without SketchArch add-on :)

```
OS: Fedora Linux 40 (Workstation Edition) (GNOME/gnome/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git) AppImage
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

[image]

semhustej, 2024-10-22:
I can reproduce the issue too. But only if the second segment is **not** perpendicular to the first one. When second segment is perpendicular, everything is OK.

https://github.com/user-attachments/assets/f03ee536-40a3-453c-80f6-a7ea4b13186a

```
18:05:01  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39024-conda-Windows-x86_64-py311\Mod\BIM\ArchWall.py", line 289, in execute
    extdata = self.getExtrusionData(obj)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39024-conda-Windows-x86_64-py311\Mod\BIM\ArchWall.py", line 1042, in getExtrusionData
    w2 = DraftGeomUtils.offsetWire(wire, dvec,
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39024-conda-Windows-x86_64-py311\Mod\Draft\draftgeoutils\offsets.py", line 214, in offsetWire
    wire = Part.Wire(edges)
           ^^^^^^^^^^^^^^^^
<class 'Part.OCCError'>: BRep_API: command not done
18:05:01  Wall: BRep_API: command not done
18:05:01  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39024-conda-Windows-x86_64-py311\Mod\BIM\ArchWall.py", line 289, in execute
    extdata = self.getExtrusionData(obj)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39024-conda-Windows-x86_64-py311\Mod\BIM\ArchWall.py", line 1042, in getExtrusionData
    w2 = DraftGeomUtils.offsetWire(wire, dvec,
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39024-conda-Windows-x86_64-py311\Mod\Draft\draftgeoutils\offsets.py", line 214, in offsetWire
    wire = Part.Wire(edges)
           ^^^^^^^^^^^^^^^^
<class 'Part.OCCError'>: BRep_API: command not done
18:05:01  Wall: BRep_API: command not done

```

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git)
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

```

maxwxyz, 2024-10-22:
reopening.
@yorikvanhavre your about info is at an older state than the latest two and the current weekly.

kaiwas, 2024-10-22:
> Unfortunately here, not ok even without SketchArch add-on :)

This is really true (I confirm). I didn't even have time to rejoice (((

Roy-043, 2024-10-23:
I can also still reproduce the error.
Note that you should not snap to the grid when testing.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39024 (Git)
Build type: Release
Branch: main
Hash: cf8ad6637351819dfa07d49cfd65d511fe14a084
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

yorikvanhavre, 2024-10-24:
I still cannot reproduce anymore, snapping to the grid or not...

```
OS: Debian GNU/Linux trixie/sid (GNOME/gnome/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39029 (Git)
Build type: Unknown
Branch: main
Hash: 0b0bb183236b40476bbb5f71dc66561f4ee56f01
Python 3.12.6, Qt 5.15.13, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * dxf-library
  * OpenTheme 2024.7.3
  * parts_library
  * freecad.gears 1.3.0
  * OpenDark 2024.3.13
  * CADExchanger
  * BIM021 2021.12.0
  * ExtremeProDark 2.9.0
```

kaiwas, 2024-10-24:
> 39029
> Build type: Unknown

We will wait for this version for testing. Thanks for the news.

paullee0, 2024-10-27:
Same in latest git, AppImage on Fedora 40

```
OS: Fedora Linux 40 (Workstation Edition) (GNOME/gnome/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git) AppImage
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

```

[image]

semhustej, 2024-10-27:
I can also reproduce 100% of time with the second segment not being 90 degrees to the first one.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git)
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

```

yorikvanhavre, 2024-10-29:
Ok I could reproduce it too indeed when the base wall's endpoints are not round values (in line with @Roy-043's comment about not snapping).

The crazy thing is, if you create a copy of the "bad" shape with: `Part.show(myWall.Base.Shape.Wires[0])` the copy is a valid wire and making a wall from it works.

yorikvanhavre, 2025-02-06:
That special case is there for @paullee0 's archsketch... Maybe we can treat that differently?

paullee0, 2025-02-06:
I am not sure I understand.

Both Sketch and ArchSketch are treated equally.

Do you mean obj.Base.GeometryFacadeList, or Part.getSortedClusters(), is problematic?

If there is a gap between 2 edges, Part.getSortedClusters() still sort them in a cluster for unknown reason, the latter have inherent problem?

Roy-043, 2025-02-08:
Open the attached file (remove the .zip extension).
Run this code:
```python
sketch = App.ActiveDocument.WallTrace
edges = [geom.Geometry.toShape() for geom in obj.GeometryFacadeList]
for edge in edges:
   [v.Point for v in edge.Vertexes]
```

Result:
```
[Vector (-8.93274016024598, 4.284031046131648, 0.0), Vector (780.8034614747439, 4.284031046131648, 0.0)]
[Vector (780.803461, 4.284031, 0.0), Vector (780.803461, 372.784374, 0.0)]

```

As you can see there is a small gap between the end of the first edge and the start of the second. But the (outward) shape of the sketch is a single wire.

[wall-trace-gap.FCStd.zip](https://github.com/user-attachments/files/18721977/wall-trace-gap.FCStd.zip)

paullee0, 2025-02-09:
Thanks, there seems to be associated Sketcher and Part.getSortedCluster() problem, random output cases (see details in https://github.com/FreeCAD/FreeCAD/issues/17087#issuecomment-2402994709).

The two (2) edges in Sketch geometry should be **connected with Coincident constraint**. However, in Sketch.Geometry, there is **still gap in the 2 'coincident' vertexes in Geometry**, then :-
 
1. Sketch.Shape.Wires return **2 Wires**
- Wires[0]/[1].Edges return seemingly 'coincident' Vertexes but **with gap**
- But Part.**getSortedCluster() return 1 cluster** (i.e. suggesting 1 wire)
3. Sketch.Shape.Wires return **1 Wire**
- Wires[0].Edges return seemingly 'coincident' Vertexes **but with No gap**
- But Part.**getSortedCluster() return 1 cluster** (i.e. suggesting 1 wire) 

Not sure if any expert in Sketcher and Part.getSortedCluster() is dealing with this.

On our side, we can test the above irregularity and do something.

paullee0, 2025-03-22:
Add occasion raised up by @psommerfeld on FC Forum -
https://forum.freecad.org/viewtopic.php?p=816741#p816741

paullee0, 2025-03-22:
After all these cases, I still believe this is Sketcher solver + Part (Part.getSortedCluster() particularly) problem : 
- https://github.com/FreeCAD/FreeCAD/issues/17087#issuecomment-2646047996
- https://github.com/FreeCAD/FreeCAD/issues/17087#issuecomment-2402994709

@PaddleStroke, seems you are on a lot of Sketcher issues, any view?  Or anybody you may suggest?

Thanks in advance.

psommerfeld, 2025-03-22:
One more project with the same failure, in case it's helpful, at https://forum.freecad.org/viewtopic.php?p=817689#p817689

Roy-043, 2025-06-22:
For reference: forum topic about using `Part.ShapeFix` to maybe address this issue:
https://forum.freecad.org/viewtopic.php?t=97846

Roy-043, 2025-12-04:
In the file I have previously posted, https://github.com/FreeCAD/FreeCAD/issues/17087#issuecomment-2645920009, the gap problem can be fixed with this code:
```python
doc = App.ActiveDocument
sketch = doc.WallTrace
sketch.touch()
sketch.solve()
doc.recompute()
```