FreeCAD/FreeCAD #17088: FreeCAD 1.0 RC2 crash on Mac M1.

fubyo, 2024-10-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

How to reproduce the crash:

1) Start a fresh FreeCAD 1.0 RC2 installation on M1 Mac.
2) Create new Project.
3) Add a sketch on the XY-Plane.
4) Add a line.
5) Add a point somewhere on the sketch (unrelated to the line).
6) Click "Close" in the "Tasks" pane.

On my system, this crashes it everytime.

### Full version info

```shell
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-08:
Could you provide an example file until it crashes? Is this also happening with the stable 0.21.2?
@JohnOCFII could you try to reproduce this?

fubyo, 2024-10-08:
> Could you provide an example file until it crashes? Is this also happening with the stable 0.21.2? @JohnOCFII could you try to reproduce this?

I have attached a test project. Once you open it, you just need to click on the sketch and FreeCAD crashes.
I did not get the chance to test this on the stable version yet.

[Test.FCStd.zip](https://github.com/user-attachments/files/17295639/Test.FCStd.zip)

fubyo, 2024-10-08:
I just tested with the latest stable version. The issue is not there.

JohnOCFII, 2024-10-08:
Also tested - no crash here either.

```
OS: macOS 15.0.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * Curves 0.6.47
```

closed 2024-10-08 by maxwxyz