FreeCAD/FreeCAD #17089: Crash in BIM

Muetdhiver-the-Wintermute, 2024-10-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD crashes when mooving camera / zooming in BIM 
[Запись экрана от 2024-10-08 11-25-37.webm](https://github.com/user-attachments/assets/96f0b2cb-ea08-4534-b6bd-38f82449bbfc)

### Full version info

```shell
OS: Manjaro Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: makepkg
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
```

### Subproject(s) affected?

None

### Anything else?

Coin warning in glxglue_isdirect(): Couldn't get current GLX context.
/usr/lib/freecad/Mod/BIM/ArchAxis.py:401: SyntaxWarning: invalid escape sequence '\['
  pts = re.findall("point \[(.*?)\]",buf)[0]
/usr/lib/freecad/Mod/BIM/ArchSite.py:70: SyntaxWarning: invalid escape sequence '\['
  buf = re.findall("point \[(.*?)\]",buf)
/usr/lib/freecad/Mod/BIM/ArchSectionPlane.py:737: SyntaxWarning: invalid escape sequence '\#'
  marker = re.findall("<line x1=.*?stroke=\"\#ffffff\".*?\/>",svg)
/usr/lib/freecad/Mod/BIM/ArchSectionPlane.py:753: SyntaxWarning: invalid escape sequence '\#'
  svg = re.sub("<line x1=.*?stroke=\"\#ffffff\".*?\/>","",svg,count=1)
/usr/lib/freecad/Mod/BIM/ArchSectionPlane.py:781: SyntaxWarning: invalid escape sequence '\?'
  svg = re.sub("<\?xml.*?>","",svg,flags=re.MULTILINE|re.DOTALL)
/usr/lib/freecad/Mod/BIM/ArchSectionPlane.py:783: SyntaxWarning: invalid escape sequence '\/'
  svg = re.sub("<\/svg>","",svg,flags=re.MULTILINE|re.DOTALL)
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3d1d0) [0x7f32c9c4c1d0]
#1  0x7f32cd04dee4 in Gui::LinkView::linkGetElementPicked(SoPickedPoint const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) const from /usr/lib/freecad/lib/libFreeCADGui.so+0x8d4
#2  0x7f32cd04f15b in Gui::ViewProviderLink::getElementPicked(SoPickedPoint const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) const from /usr/lib/freecad/lib/libFreeCADGui.so+0x12b
#3  0x7f32cce6234f in Gui::SoFCUnifiedSelection::getPickedList(SoHandleEventAction*, bool) const from /usr/lib/freecad/lib/libFreeCADGui.so+0x59f
#4  0x7f32cce66740 in Gui::SoFCUnifiedSelection::handleEvent(SoHandleEventAction*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x1f0
#5  0x7f32c8ab537c in SoAction::traverse(SoNode*) from /usr/lib/libCoin.so.80+0x6c
#6  0x7f32c8c8843b in SoChildList::traverse(SoAction*, int, int) from /usr/lib/libCoin.so.80+0xfb
#7  0x7f32c8cd7486 in SoGroup::doAction(SoAction*) from /usr/lib/libCoin.so.80+0x56
#8  0x7f32c8cee26f in SoSeparator::doAction(SoAction*) from /usr/lib/libCoin.so.80+0x2f
#9  0x7f32c8ab537c in SoAction::traverse(SoNode*) from /usr/lib/libCoin.so.80+0x6c
#10  0x7f32c8abc564 in SoHandleEventAction::beginTraversal(SoNode*) from /usr/lib/libCoin.so.80+0xc4
#11  0x7f32c8aac551 in SoAction::apply(SoNode*) from /usr/lib/libCoin.so.80+0xd1
#12  0x7f32c8ca1359 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /usr/lib/libCoin.so.80+0x79
#13  0x7f32ccf43352 in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x22
#14  0x7f32ccf52d4e in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x5e
#15  0x7f32ccf6121d in Gui::NavigationStyle::processSoEvent(SoEvent const*) from /usr/lib/freecad/lib/libFreeCADGui.so+0xdd
#16  0x7f32ccf62f2b in Gui::CADNavigationStyle::processSoEvent(SoEvent const*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x22b
#17  0x7f32ccf6143c in Gui::NavigationStyle::processEvent(SoEvent const*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x6c
#18  0x7f32ccf9627f in Gui::View3DInventorViewer::processSoEvent(SoEvent const*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x6f
#19  0x7f32ccf3c985 in SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x165
#20  0x7f32c9744ef8 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x118
#21  0x7f32ca8fc8bc in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x8c
#22  0x7f32ca901c92 in QApplication::notify(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0xf32
#23  0x7f32ccaa1c89 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x99
#24  0x7f32c9745aa8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#25  0x7f32ca8f3f9d in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /usr/lib/libQt6Widgets.so.6+0x21d
#26  /usr/lib/libQt6Widgets.so.6(+0x1635c2) [0x7f32ca9635c2]
#27  /usr/lib/libQt6Widgets.so.6(+0x164c20) [0x7f32ca964c20]
#28  0x7f32ca8fc8cc in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x9c
#29  0x7f32ccaa1c89 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad/lib/libFreeCADGui.so+0x99
#30  0x7f32c9745aa8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x158
#31  0x7f32c9f78140 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /usr/lib/libQt6Gui.so.6+0x8d0
#32  0x7f32c9fe7724 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Gui.so.6+0x304
#33  /usr/lib/libQt6Gui.so.6(+0x6041e4) [0x7f32ca4041e4]
#34  /usr/lib/libglib-2.0.so.0(+0x5d299) [0x7f32c81cd299]
#35  /usr/lib/libglib-2.0.so.0(+0xbfec7) [0x7f32c822fec7]
#36  /usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x35) [0x7f32c81cc795]
#37  0x7f32c99a82bd in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x8d
#38  0x7f32c974ff66 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x1a6
#39  0x7f32c974a11d in QCoreApplication::exec() from /usr/lib/libQt6Core.so.6+0x8d
#40  0x7f32cc9aa2a6 in Gui::Application::runApplication() from /usr/lib/freecad/lib/libFreeCADGui.so+0xb46
#41  FreeCAD(+0x6d0c) [0x5653ec3ccd0c]
#42  /usr/lib/libc.so.6(+0x25e08) [0x7f32c9c34e08]
#43  /usr/lib/libc.so.6(__libc_start_main+0x8c) [0x7f32c9c34ecc]
#44  FreeCAD(+0x7105) [0x5653ec3cd105]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-10-08:
Perhaps a duplicate of https://github.com/FreeCAD/FreeCAD/issues/16232?

yorikvanhavre, 2024-10-08:
> Perhaps a duplicate of https://github.com/FreeCAD/FreeCAD/issues/16232?

That's possible... It does not seem to be related to BIM anyway.

@Muetdhiver-the-Wintermute can you try an RC2 build? It should contain the fix above. https://github.com/FreeCAD/FreeCAD/releases/tag/1.0rc2

maxwxyz, 2024-10-08:
Closing as duplicate

closed 2024-10-08 by maxwxyz