FreeCAD/FreeCAD #17090: TechDraw: detail view ignores hatch pattern

furgo16, 2024-10-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]

The attached test file and the screenshot should illustrate this: when creating a detail view from a e.g. a section cut face (`from Section A - A`) with a hatch pattern, that hatch pattern does not appear in the detail view (`Detail 1`).

[TestAssembly.zip](https://github.com/user-attachments/files/17289630/TestAssembly.zip)

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38924 (Git) Snap 1159
Build type: Release
Branch: main
Hash: 13eeb7b8c23cb245a431bc8a825127570bf77c73
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * ezydxf
  * Ondsel-Lens 2024.9.13.02
  * Assembly4 0.50.15
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-08:
@WandererFan FYI

WandererFan, 2024-10-08:
There is currently no way to link a face in the detail view with a face (and hatch) in the base view.   Would need to add code to find any faces within the detail area on the base view and then match them to faces in the detail view (which may be a different shape).

Can't see this as a bug.

zambetti, 2026-10-20:
Having recently needed detail views to inherit hatching from section views, I pointed OAI-Astra at this issue and it worked (at least for my use case):

https://github.com/zambetti/FreeCAD/blob/experiment/ai-techdraw-detail-hatching/EXPERIMENT_DETAIL_HATCHING.md

I'm aware of FreeCAD's AI policy, so I'm only sharing this as a possible reference for a proper fix.