FreeCAD/FreeCAD #17094: instant crash - edit-select all

graemejw, 2024-10-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When choosing select all from the edit menu freecad instantly crashes

### Full version info

```shell
OS: macOS 15.0
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * FreeCAD-themes 2024.7.24
  * OpenTheme 2024.8.17
  * lattice2 1.0.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-08:
Cannot reproduce on Win.
@JohnOCFII can you confirm on macOS?
```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.48
  * fasteners 0.5.29
  * freecad.gears 1.2.0
  * Manipulator 1.5.7

```

JohnOCFII, 2024-10-08:
> Cannot reproduce on Win.
> @JohnOCFII can you confirm on macOS?

Crashes on version 38871.  Does NOT crash with 38923.

(For some reason - while I got a crash report first time through - I'm not getting one now on 38871, even if the program exits.)

```
OS: macOS 15.0.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * Curves 0.6.47
```

```
OS: macOS 15.0.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * Curves 0.6.47
```

maxwxyz, 2024-10-08:
I'll close this then as fixed.

closed 2024-10-08 by maxwxyz