FreeCAD/FreeCAD #17095: Crash die kleine ist okayon MAC OS  Freecad RC2

rolfeisenhut, 2024-10-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Create new project, create body, create a sketch.

Click on show x-axis oder y-axis or y-axis in tree view for show/hidding the axis, 
It results in a  crash....

### Full version info

```shell
OS: macOS 15.0.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-10-09:
@johnocfii  can you reproduce on macos ? TIA

hyarion, 2024-10-09:
It sounds like a duplicate of https://github.com/FreeCAD/FreeCAD/issues/16820

hyarion, 2024-10-09:
<details><summary>The crash is indeed 16820</summary>
<p>

```
Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libTKBRep.7.7.2.dylib         	       0x14e71cc3c BRep_Tool::Curve(TopoDS_Edge const&, TopLoc_Location&, double&, double&) + 64
1   libTKBRep.7.7.2.dylib         	       0x14e72c9e0 BRepAdaptor_Curve::Initialize(TopoDS_Edge const&) + 188
2   libTKBRep.7.7.2.dylib         	       0x14e72c8b4 BRepAdaptor_Curve::BRepAdaptor_Curve(TopoDS_Edge const&) + 160
3   Measure.so                    	       0x127ed9258 Measure::Measurement::length() const + 308
4   MeasureGui.so                 	       0x153918038 MeasureGui::QuickMeasure::printResult() + 1480
5   MeasureGui.so                 	       0x153917428 MeasureGui::QuickMeasure::processSelection() + 52
6   libQt5Core.5.15.13.dylib      	       0x102561e84 0x10241c000 + 1334916
7   libQt5Core.5.15.13.dylib      	       0x102545094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
8   CoreFoundation                	       0x18a4ae46c __CFRunLoopDoSource0 + 176
9   CoreFoundation                	       0x18a4ae1dc __CFRunLoopDoSources0 + 244
10  CoreFoundation                	       0x18a4acdc8 __CFRunLoopRun + 828
11  CoreFoundation                	       0x18a4ac434 CFRunLoopRunSpecific + 608
```

</p>
</details> 

This has been fixed and does not cause a crash in dev builds afterwards.

@luzpaz We can close this issue.

closed 2024-10-09 by maxwxyz