FreeCAD/FreeCAD #17097: view provider not attached

eerica860tw, 2024-10-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Problem:
Every time I click "create shape binder", "sub-shape binder" or "datum" button, a bunch of errors appear from Report view saying that "view provider not attached" and nothing was created.
Why is this happening and how should I fix it?
I have already tried latest version of FreeCAD and tested in a new document but in vain.
Could someone please help?
[image]

Here is a video to recreate the error
https://www.youtube.com/watch?v=hsmULMFUuY4

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git)
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Chinese/Taiwan (zh_TW)
Installed mods: 
  * freecad.gears 1.3.0
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-09:
I cannot reproduce.
Please provide a demo file. Does this work in the stable 0.21.2?
```
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * Curves 0.6.48
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.24
  * Rocket 3.3.0
  * sheetmetal 0.4.26
  * Silk 0.1.5
  * Silk.backup1727166062.7227929 0.1.5 (Disabled)
```

maxwxyz, 2024-10-09:
@CalligaroV could this be TNP related?

CalligaroV, 2024-10-09:
Hi @maxwxyz and @eerica860tw!

Probably yes, it may be TNP related but I can't say for sure without reproducing the issue and looking at the code.
Moreover I never used much the (Sub)ShapeBinder function (yeah, in this case I'm a terrible FC user :confounded:) therfor I'll need some time before giving an useful feedback.

ATM, given what I'm following and what's in my cue probably I could be able to start testing/studying this one at some point during the next week.

eerica860tw, 2024-10-10:
Hi @maxwxyz  and @CalligaroV !
I think I found the problem.
Looks like everyone else didn't encountered this problem, and the dev version is pretty much identical.
Thus I couldn't help but think that maybe the problem is caused by runtime environment.
And the major difference between your environment and mine is system language (I am Chinese user).

So I moved the whole dev version program from the original download path which contains Chinese char to root of my F: drive,
everything works as it should now!
[image]

Many thanks to you two! I have been stuck in this problem form a long time, it's just like a pain in the ass lol
Thank you!

maxwxyz, 2024-10-10:
but then there still seems to be an issue?
@adrianinsaval is this a limitation that FC cannot be run on system paths with Chinese characters?

luzpaz, 2024-10-11:
Ticket needs a rename

adrianinsaval, 2024-10-12:
> but then there still seems to be an issue? @adrianinsaval is this a limitation that FC cannot be run on system paths with Chinese characters?

I don't know but sounds like a bug to me since OP claims it worked in previous version.

What is not clear to me is if the issue really comes from installing freecad into a directoy with chinese characters or it's about the file being on a directory with chinese characters.

In any case free software should be for everyone and should not stop working just because you use a different language, we should work on fixing this.

luzpaz, 2024-10-14:
Anyone want to try tackling this issue ?

eerica860tw, 2024-10-14:
> I don't know but sounds like a bug to me since OP claims it worked in previous version.
>
> What is not clear to me is if the issue really comes from installing freecad into a directoy with chinese [...]

Hi @adrianinsaval , I installed both version on different path, and tests if I could create datum plane or shape binder normally.
Here is the results:
1. Install stable version in "Chinese chars contained path": datum, shape binder work fine.
2. Install stable version in "English chars only path": everything works fine.
3. Put dev version folder under "Chinese chars contained path" and runs freecad.exe: datum, shape binder failed
4. Put dev version folder under "English chars only path" and runs freecad.exe: datum, shape binder work fine

eerica860tw, 2024-10-15:
Hi @adrianinsaval, I didn't saved or loaded any file during the tests. I just opened FreeCAD, created new document and tested those functions.

luzpaz, 2024-10-16:
Related: 
* https://github.com/FreeCAD/FreeCAD/pull/17265

maxwxyz, 2024-10-19:
@eerica860tw could you try this if it works in the RT linkstage fork https://github.com/realthunder/FreeCAD/tree/LinkStage3

WandererFan, 2024-10-20:
FWIW, I tried the following:
copy fc install to ~/Target/我喜欢吃苹果
copy DevCreated.FCStd to ~/Documents/FreeCAD/Support/Languages/床前明月光

run FC from ~/Target/我喜欢吃苹果
load DevCreated.FCStd from ~/Documents/FreeCAD/Support/Languages/床前明月光

create datum plane - OK
create subshapebinder - OK

Windows only problem?  

OS: Linux Mint 21.2 (X-Cinnamon/cinnamon/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 +1 (Git)
Build type: debug
Branch: LeaderLineStyle17307
Hash: 3c0cd49411cf45e2caed22fabfda48d5cd2295fa
Python 3.10.12, Qt 5.15.2, Coin 4.0.0, Vtk , OCC 7.8.0
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.21

wwmayer, 2024-11-08:
It's the same problem as in #14352

adrianinsaval, 2024-11-09:
meaning it is solved or meaning the same change should be made somewhere else in the code?

PR #17759 (2024-11-09): Material: Replace YAML::LoadFile() with YAML::Load()
Fixes #17097
commit 9126e7e9 (2024-11-09): Material: Replace YAML::LoadFile() with YAML::Load()

YAML::LoadFile() cannot be used on Windows because it doesn't support UTF-16 to load a file from an arbitrary location

This fixes #17097
commit df971b10 (2024-11-09): Material: Replace YAML::LoadFile() with YAML::Load()

YAML::LoadFile() cannot be used on Windows because it doesn't support UTF-16 to load a file from an arbitrary location

This fixes #17097
closed 2024-11-09 by chennes