FreeCAD/FreeCAD #17098: Problem importing STEP files

Presleypxab, 2024-10-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When trying to import an assembly in STEP format into FreeCAD, an error occurs and the import does not complete.

### Full version info

```shell
<class 'NameError'>: name 'PyExc_FC_AbortIOException' is not defined
09:45:17  QPixmap::setMask() mask size differs from pixmap size
09:45:17  QPixmap::setMask() mask size differs from pixmap size
09:45:17  QPixmap::setMask() mask size differs from pixmap size
09:45:17  QPixmap::setMask() mask size differs from pixmap size
09:45:17  QPixmap::setMask() mask size differs from pixmap size
09:45:17  QPixmap::setMask() mask size differs from pixmap size
09:45:17  QPixmap::setMask() mask size differs from pixmap size
09:45:17  QPixmap::setMask() mask size differs from pixmap size
09:45:17  QPixmap::setMask() mask size differs from pixmap size
09:45:17  QPixmap::setMask() mask size differs from pixmap size
09:45:17  QPixmap::setMask() mask size differs from pixmap size
09:45:17  QPixmap::setMask() mask size differs from pixmap size
09:45:17  QPixmap::setMask() mask size differs from pixmap size
09:45:17  QPixmap::setMask() mask size differs from pixmap size
09:45:17  QPixmap::setMask() mask size differs from pixmap size
09:45:17  QPixmap::setMask() mask size differs from pixmap size
09:45:17  QPixmap::setMask() mask size differs from pixmap size
09:45:17  QPixmap::setMask() mask size differs from pixmap size
09:45:17  QPixmap::setMask() mask size differs from pixmap size
09:45:17  QPixmap::setMask() mask size differs from pixmap size
09:45:17  QPixmap::setMask() mask size differs from pixmap size
09:45:17  QPixmap::setMask() mask size differs from pixmap size
09:45:17  QPixmap::setMask() mask size differs from pixmap size
09:45:17  QPixmap::setMask() mask size differs from pixmap size
09:45:17  QPixmap::setMask() mask size differs from pixmap size
09:45:17  QPixmap::setMask() mask size differs from pixmap size
11:12:52  pyException: Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\site-packages\freecad\module_io.py", line 12, in OpenInsertObject
    except PyExc_FC_AbortIOException:
           ^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'NameError'>: name 'PyExc_FC_AbortIOException' is not defined
```

### Subproject(s) affected?

File formats

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-10-08:
Please supply your version info and attach an example file.

Version info:
Help → About FreeCAD → Press "Copy to clipboard" button → Paste in "Full version info" field.

Presleypxab, 2024-10-08:
Hi,

The version is 1.0 RC2

luzpaz, 2024-10-08:
@Presleypxab 
Please don't respond via email. Instead:  

1. Click https://github.com/FreeCAD/FreeCAD/issues/17098
1. Open FreeCAD 
1. Paste the full [About](https://wiki.freecad.org/About) info. We ask this for a reason. 
1. zip up your example Step file and drag/drop it into the github textbox for us to test.

Presleypxab, 2024-10-09:
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vt
[Botão-fina2.zip](https://github.com/user-attachments/files/17301261/Botao-fina2.zip)
k 9.2.6, OCC 7.7.2
Locale: Portuguese/Brazil (pt_BR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * ModernUI
  * PieMenu 1.8.1

Presleypxab, 2024-10-09:
[Botão-fina2.zip](https://github.com/user-attachments/files/17301268/Botao-fina2.zip)

Presleypxab, 2024-10-16:
I uninstalled and installed it again and the same problem occurred when trying to import the STEP file. The error follows again.

23:13:21  pyException: Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 1.0\bin\Lib\site-packages\freecad\module_io.py", line 12, in OpenInsertObject
    except PyExc_FC_AbortIOException:
           ^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'NameError'>: name 'PyExc_FC_AbortIOException' is not defined

OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Brazil (pt_BR)
Stylesheet/Theme/QtStyle: unset/classic/Qt default

Here is the file I tried to open
[12.0124-12.0124.step.zip](https://github.com/user-attachments/files/17388335/12.0124-12.0124.step.zip)

luzpaz, 2024-10-16:
I can import fine with no errors/warnings

[image]

```
OS: Manjaro Linux (KDE/plasma/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38977 (Git) AppImage
Build type: Release
Branch: main
Hash: facd8ca6e0303cf96297a7a7398185770c1a1e0b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

luzpaz, 2024-10-16:
@Presleypxab can you try the weekly build ?  (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

Presleypxab, 2024-10-17:
@luzpaz 
Hi, so in FC 0.22 the file opens normally, but in FC1.0 it gives this error. I'll try this weekly compilation to see if it works. One question. To try these compilations do I have to uninstall the FC that is installed?

Presleypxab, 2024-10-17:
@luzpaz 
I tested the new build and it worked correctly. Top

closed 2024-10-17 by luzpaz

Presleypxab, 2024-10-17:
@luzpaz 

OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38977 (Git)
Build type: Release
Branch: main
Hash: facd8ca6e0303cf96297a7a7398185770c1a1e0b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Brazil (pt_BR)
Stylesheet/Theme/QtStyle: unset/classic/Qt default
Installed mods: 
  * A2plus 0.4.68
  * OpenTheme 2024.9.1
  * sheetmetal 0.4.26

Presleypxab, 2024-10-17:
I found another problem in this latest version. Should I post it here or do I have to open another topic?

Presleypxab, 2024-10-17:
@luzpaz 

I'm new to Freecad, I thought the project was amazing, and I'm willing to help with whatever is necessary. I have some questions regarding the handling of the program in some applications. I've done a lot of research but haven't found anything satisfactory. I'm coming from SolidWorks and there are some implementations that I don't know if they already exist, but I think they're valid. Can I look into this with you or should I ask someone else?

luzpaz, 2024-10-17:
> I found another problem in this latest version. Should I post it here or do I have to open another topic?

Open a new ticket, please. TIA

> I have some questions regarding the handling of the program in some applications. I've done a lot of research but haven't found anything satisfactory. I'm coming from SolidWorks and there are some [...]

The forum or discord is genuinely the best place for your inquiries. It will save you a lot of time and 'elbow grease' getting informed directly from veteran FC users. Welcome, I hope you like it and stay as long as you like.

Presleypxab, 2024-10-18:
@luzpaz 
Thanks for the welcome. Is there an open group on Discord?

luzpaz, 2024-10-18:
@Presleypxab https://discord.gg/w2cTKGzccC