FreeCAD/FreeCAD #17103: OpenSCAD file import not supported on MacOS (1.0 RC2)

alidaf, 2024-10-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Trying to open an OpenSCAD file, a dialog appears stating that the file is unsupported. The workbench is linked to the executable and loaded.
MacOS 13.6.9

### Full version info

```shell
1.0.0RC2 38806 (git)
```

### Subproject(s) affected?

OpenSCAD

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

benj5378, 2024-10-08:
Please with send the file you tried to import

maxwxyz, 2024-10-09:
Please add the full [About](https://wiki.freecad.org/About) info to the issue (`Help > About FreeCAD > Copy to clipboard`).

alidaf, 2024-10-09:
OS: macOS 13.6.9
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * lattice2 1.0.0

alidaf, 2024-10-09:
> Please with send the file you tried to import

This happens with every file I try. It may be noteworthy that on "file -> open", the default supported file list does not include .scad files. I select "all files" and then choose the file to load. I've attached an example.

[Keyring Container v2.1.scad.zip](https://github.com/user-attachments/files/17304594/Keyring.Container.v2.1.scad.zip)

luzpaz, 2024-10-09:
@alidaf can you also try the [alternate openscad importer](https://github.com/KeithSloan/OpenSCAD_Alt_Import) via the [addon manager](https://wiki.freecad.org/AddonManager ?

benj5378, 2024-10-09:
I can confirm, that the usual `Open...` or `Import...` under `File` workflow doesn't work with OpenSCAD files.

Opening it in to FreeCAD might be by first creating a new document, then open the OpenSCAD workbench to the press add OpenSCAD element and choose `Open...` to open a file. (Your specific file in this case fails, because of some imports?)

<details>
<summary>
Import warning messages
</summary>
```
16:36:52  WARNING: Token 'DOT' defined, but not used
16:36:52  WARNING: Token 'WORD' defined, but not used
16:36:52  WARNING: There are 2 unused tokens
16:36:52  End processing CSG file
16:37:14  WARNING: Can't open include file 'BOSL2/std.scad'.
WARNING: Can't open include file 'BOSL2/threading.scad'.
WARNING: Ignoring unknown module 'cyl' in file fc-349368-463457-000004.scad, line 131
WARNING: Ignoring unknown module 'cyl' in file fc-349368-463457-000004.scad, line 138
WARNING: Ignoring unknown module 'cyl' in file fc-349368-463457-000004.scad, line 148
WARNING: Ignoring unknown module 'cyl' in file fc-349368-463457-000004.scad, line 178
WARNING: Ignoring unknown module 'threaded_rod' in file fc-349368-463457-000004.scad, line 186
WARNING: Ignoring unknown module 'cyl' in file fc-349368-463457-000004.scad, line 196
WARNING: Ignoring unknown module 'threaded_rod' in file fc-349368-463457-000004.scad, line 230
WARNING: Ignoring unknown module 'cyl' in file fc-349368-463457-000004.scad, line 237
WARNING: Ignoring unknown module 'cyl' in file fc-349368-463457-000004.scad, line 247
WARNING: Ignoring unknown module 'cyl' in file fc-349368-463457-000004.scad, line 254
WARNING: Ignoring unknown module 'cyl' in file fc-349368-463457-000004.scad, line 264
WARNING: Ignoring unknown module 'cyl' in file fc-349368-463457-000004.scad, line 295
WARNING: Ignoring unknown module 'cyl' in file fc-349368-463457-000004.scad, line 302
WARNING: Ignoring unknown module 'cyl' in file fc-349368-463457-000004.scad, line 323
WARNING: Ignoring unknown module 'cyl' in file fc-349368-463457-000004.scad, line 329
WARNING: Ignoring unknown module 'cyl' in file fc-349368-463457-000004.scad, line 338
WARNING: Ignoring unknown module 'cuboid' in file fc-349368-463457-000004.scad, line 346
WARNING: Ignoring unknown module 'cyl' in file fc-349368-463457-000004.scad, line 352
```
</details>

I think the issue here can be defined by the unintuitive workflow. For OpenSCAD files, one should be able to press `Open...` or `Import...` in FreeCAD under `File` menu like with all other files - and then FreeCAD should under the hood load the OpenSCAD workbench to create the new object in an empty document.

alidaf, 2024-10-09:
I use the BOSL2 library a lot and it is installed in the proper library folder.

maxwxyz, 2024-10-09:
does it work in the stable 0.21.2 version?

luzpaz, 2024-10-09:
@KeithSloan do you have a moment to test this issue ?

ibh2322, 2024-10-09:
I had the same issue on open OpenSCAD file, but i got a ModuleNotFoundError: module named 'ply' and when i installed that i can open OpenSCAD files from Open... or Import... under File

OS: Salix 15.0 (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.Unknown
Build type: Release
Python 3.9.20, Qt 5.15.3, Coin 4.0.1, Vtk 9.0.3, OCC 7.3.0
Locale: Swedish/Sweden (sv_SE)
Stylesheet/Theme/QtStyle: unset/unset/Qt default

KeithSloan, 2024-10-09:
Sorry on Vacation at present not back in UK till 15th / 16th.

Do things work with FC RC1.0? thought I tried a minimum test of the Alternate Importer with FC RC1.0 but not 100% sure.

KeithSloan, 2024-10-09:
My advice would be to try the following to add in tracking down the problem

1) Try a csg file ( Open scad file in OpenSCAD and export a csg version). This would at least establish if the problem is with a csg import or the Workbench creating a csg file under the covers. All importers create a csg file and then import that.

2) Is the workbench preference that points the OpenSCAD executable set correctly, think it may need to be something like /Applications/OpenSCAD.app or /Applications/OpenSCAD.app/bin/openscad, cannot remember without access to my machine.

3) Are the libraries used by the OpenSCAD executable compatible with FC RC 2.0. Need to check level of python especially.
This would be my most likely cause of the issue.

Hope this helps.

KeithSloan, 2024-10-15:
@ibh2322. Thanks for the info.

As far as the Alternative Importer, I have added ply to the metadata.txt and package.xml sections.

It used to be the case that FC used the ply library.

myxxmikeyxx, 2025-01-13:
Hello. I had this same issue on my mac. Intel based. 
```
OS: macOS 14.2.1
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Alternate_OpenSCAD 1.0.0
```

As you can see I install Alternative OpenSCAD as well and still was unable to import anything.
Files I tried:
- CSG
- SCAD
- OFF
- 3MF

Error when importing 3MF:
```
13:03:46  pyException: Traceback (most recent call last):
  File "<string>", line 2, in <module>
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/freecad/module_io.py", line 16, in OpenInsertObject
    getattr(importerModule, importMethod)(objectPath, *importArgs, **importKwargs)
<class 'RuntimeError'>: Zip file consistency problem. Failure while reading zip file central directory
```

Steps I tried:

1. Open FreeCAD and try doing just open file. 
2. Tired this on all the above listed files that were exported from OpenSCAD-2021.01

Other:
1. Open FreeCAD and create new project
2. File > import > select file
3. Tried all the other options to import the different files, still no luck.

Original OpenSCAD file [link here](https://www.printables.com/model/34992-pegstr-pegboard-wizard/files)
Opened that file in OpenSCAD and then tried multiple different export options. I am able to open those files back up in OpenSCAD.

The tutorial I am following [link is here](https://forums.autodesk.com/t5/fusion-manage-ideas/how-to-import-openscad-file-into-fusion360-as-a-solid-body/idi-p/10174834)

Reason I am trying to do this is so that I can use fusion 360 for the file. I have some coworkers that have pegboard at their desk and fusion 360 is what they have used in the past to edit objects and import a mesh and converting it gives that top image on the post results and is hard to edit.

FEA-eng, 2026-05-01:
Needs testing on macOS.

KeithSloan, 2026-11-08:
See latest update to https://github.com/KeithSloan/OpenSCAD_Alt_Import

https://github.com/KeithSloan/OpenSCAD_Alt_Import/blob/master/UPDATE_README.md