FreeCAD/FreeCAD #17104: Core: Autocomplete does not work on properties of linked object

pieterhijma, 2024-10-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Given a link (for example Part001) to some object that has a custom property (for example a Part with a property `MyProp`), autocompletion does not work in expressions that try to access `<<Part001>>.MyProp`.

### Full version info

```shell
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38925 (Git)
Build type: Release
Branch: main
Hash: 12c6fc2920d083c7407419f8b9ae3651798c005a
Python 3.12.6, Qt 6.7.3, Coin 4.0.3, Vtk , OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Installed mods: 
  * OpenTheme 2024.5.16
  * osh-autodoc-workbench 0.2.3
  * Ondsel-Lens 2024.9.13.02
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-10-08:
Related (or duplicate?) issue: https://github.com/FreeCAD/FreeCAD/issues/12252

pieterhijma, 2025-08-09:
Yes, I think so, except that I believe that this issue better describes the general issue. It happens to occur in spreadsheets as well, but I don't think it is a spreadsheet problem, but more a core problem. Maybe we can link the other issue in as a manifestation of this more generic issue? Then if at some point we have a fix for it, we can also check if the fix works in Spreadsheet.

luzpaz, 2025-12-19:
This is a workflow disruptor. Anyone able to tackle this before the 1.1 release ?