FreeCAD/FreeCAD #17105: Path_Inspect tool can't be dragged

jes, 2024-10-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In Ubuntu 24.04 LTS, using FreeCAD 0.21.2 from Snap, when I open the Path_Inspect tool, it *works* properly, but I can't see all of my toolpaths because it is stuck in the middle of the screen and I can't move it. It is fixed so that it is in the middle of the FreeCAD view, and trying to drag the Path_Inspect window drags the whole FreeCAD window behind it.

### Full version info

```shell
[code]
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) Snap 908
Build type: Release
Branch: tag: 0.21.2
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.7, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
[/code]
```

### Subproject(s) affected?

CAM

### Anything else?

I didn't manage to include the pointer in the video.

I am dragging the titlebar of the Path_Inspect window, and it first un-maximises FreeCAD and then drags the entire FreeCAD window around with it.

[Screencast from 2024-10-08 16-12-32.webm](https://github.com/user-attachments/assets/32b346f1-3290-4c14-8aab-d6c3d080d2ae)

I wonder if the window needs to be marked as "not a modal" or something?

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-09:
@jes could you try this with the 1.0 release candidate or dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

jes, 2024-10-09:
Yes, the bug is there the same in the "latest/edge" version of the snap:

  OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
  Word size of FreeCAD: 64-bit
  Version: 1.1.0dev.38926 (Git) Snap 1160
  Build type: Release
  Branch: main
  Hash: a9bff78974f301100134a13c636e23aa91be5bcf
  Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
  Locale: English/United Kingdom (en_GB)
  Stylesheet/Theme/QtStyle: unset/FreeCAD Dark/Fusion

dresco, 2024-12-14:
Hi @jes, this is likely to be the Gnome "attach modal dialogs" setting?

You can check with;
`gsettings get org.gnome.mutter attach-modal-dialogs`

And disable this behaviour with;
`gsettings set org.gnome.mutter attach-modal-dialogs false`

luzpaz, 2024-12-16:
@jes please confirm

sliptonic, 2025-05-19:
I can't duplicate this.  Closing.  OP can re-open if it is still a problem

closed 2025-05-19 by sliptonic

jes, 2025-05-20:
Yes the bug is still there. Ubuntu 24.04 LTS.

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.38641 +468 (Git) Snap 1248
Build type: Release
Branch: tag: 1.0.0
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * freecad.gears 1.3.0
```

FreeCAD is not the only program that does this, it may be an Ubuntu mis-feature rather than a FreeCAD bug.

dresco, 2025-05-20:
@jes  

It's a Gnome feature, did you verify the **attach-modal-dialogs** setting from earlier comment?
You can either change it with gsettings, or from the GUI with the GNOME Tweaks tool..

jes, 2025-05-20:
Oh yeah, that actually did fix it, thanks! I somehow missed that comment.

jes, 2025-05-20:
So maybe the fix is to make Path Inspect not be a modal?