FreeCAD/FreeCAD #17110: Material: Materials editor RMB context menu is not working

semhustej, 2024-10-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In Materials editor, when right clicking on existing material and using option "Inherit from" and "Inherit new material", neither of  these options from the RMB menu seem to work.

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: English/United States (en_US) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

The issue was confirmed by two other users, one in this discussion: https://forum.freecad.org/viewtopic.php?t=91242

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-10-08:
I can confirm it.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

@davesrocketshop FYI

luzpaz, 2024-10-31:
@davesrocketshop any chance this can be addressed for 1.0 or should we mark post-1.0 ?

davesrocketshop, 2024-10-31:
Well, the simplest fix would be to disable it for now. I wouldn't want to do any more for 1.0 this close to release.

luzpaz, 2024-11-01:
Thanks! BTW, the PR shouldn't really close this ticket in that case, right?

davesrocketshop, 2024-11-01:
It probably should. I think the whole editor needs a top to bottom review by the UX team. That's a whole separate issue

semhustej, 2024-11-01:
> Well, the simplest fix would be to disable it for now. I wouldn't want to do any more for 1.0 this close to release.

I think that makes sense. From what I understand all the options the RMB context menu provides can be achieved in different way from the materials editor.

PR #17599 (2024-11-04): Materials: Disable editor right click menu
The right click context menu in the materials editor was implemented as a place holder but is currently non-functional. It should be disabled for 1.0. This will be revisited post 1.0 as all editor functionality is reviewed and improved.

Fixes #17110

[image]
commit 14306801 (2024-11-04): Materials: Disable editor right click menu

The right click context menu in the materials editor was implemented
as a place holder but is currently non-functional. It should be
disabled for 1.0. This will be revisited post 1.0 as all editor
functionality is reviewed and improved.

Fixes #17110
commit 364ae2ca (2024-11-04): Materials: Disable editor right click menu

The right click context menu in the materials editor was implemented
as a place holder but is currently non-functional. It should be
disabled for 1.0. This will be revisited post 1.0 as all editor
functionality is reviewed and improved.

Fixes #17110
closed 2024-11-04 by yorikvanhavre