FreeCAD/FreeCAD #17112: Property View - Fresh Install - No modificaitons

jrb0411, 2024-10-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Fresh installation of ver 1.0.0RC2. no modifications to preferences. The Property View window has very poor contrast.  See attached image.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus 0.4.65
  * Assembly4 0.50.15
  * BIM021 2021.12.0
  * ExplodedAssembly
  * woodworking 0.21.2.33771
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-10-09:
cannot reproduce. Did you restart FreeCAD?
```
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38871 (Git)
Build type: Release
Branch: main
Hash: 813ee36cb80e0d29c2c9b991a7b2fee2c61f5cf6
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * Curves 0.6.48
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.24
  * Rocket 3.3.0
  * sheetmetal 0.4.26
  * Silk 0.1.5
  * Silk.backup1727166062.7227929 0.1.5 (Disabled)

```
[image]

goleon69, 2024-10-10:
same here on Mac, just installed the RC2 and open a previous file done with 0.21.

[image]

maxwxyz, 2024-10-10:
@JohnOCFII same on your side?

MisterMakerNL, 2024-10-10:
This is the background of Qtreeview, it is set to 2e2e2e, I have checked the stylesheet if 2e2e2e was used anywhere and it is not. 
It could still be caused by a RGBA color, but I doubt it.
I also cannot reproduce it, and that often means it is some QTstyle issue.
Recently there was an update with QT style in windows.
So updating windows 11 would be a good idea, as I see it is outdated.
For the mac that is a bit difficult. Maybe Qtreeview isn't set correctly in main and it is taken from Qtstyle, which in Linux would be supplied by the system and maybe mac and windows11 style doesn't set this color. Even QT says windows11 style isn't complete and recommend fusion soooo that is kinda annoying.

goleon69, 2024-10-10:
Not sure can help but I just checked again and after I open the add-on manager and update the data, I got a series of errors and then tried to draw a new object and the Property view was ok, then re-open the old file done on 0.21 and the property manager now looks ok.

[image]

JohnOCFII, 2024-10-12:
> @JohnOCFII same on your side?

I was attempting to test this, and was switching from OpenLight to FreeCAD Classic.  I hit `Apply` in Preferences, and had a crash.  After restarting, it appears the theme change to FreeCAD Classic **did** take hold, and the theming and text look OK to me.

```
OS: macOS 15.0.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * Curves 0.6.50
```

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [21171]
Path:                  /Applications/Other Apps/FreeCAD-38923.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-10-12 12:32:53.5564 -0500
OS Version:            macOS 15.0.1 (24A348)
Report Version:        12
Anonymous UUID:        E1A9B3E8-17C0-3635-633F-31D076EC9805

Sleep/Wake UUID:       283ED146-87C4-4210-836D-0AF0B9C09B29

Time Awake Since Boot: 220000 seconds
Time Since Wake:       19991 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00006a0eeec178f8
Exception Codes:       0x0000000000000001, 0x00006a0eeec178f8

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [21171]

VM Region Info: 0x6a0eeec178f8 is not in any region.  Bytes after previous region: 11058714605817  
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      MALLOC_NANO              600000000000-600020000000 [512.0M] rw-/rwx SM=PRV  
--->  
      UNUSED SPACE AT END

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libQt5Widgets.5.15.13.dylib   	       0x103d5ee90 0x103ce0000 + 519824
1   libQt5Widgets.5.15.13.dylib   	       0x103d52100 0x103ce0000 + 467200
2   libQt5Core.5.15.13.dylib      	       0x1045e1094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
3   libQt5Widgets.5.15.13.dylib   	       0x103d0a67c QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) + 1872
4   libQt5Widgets.5.15.13.dylib   	       0x103cee264 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 232
5   libQt5Core.5.15.13.dylib      	       0x1045e1094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
6   libQt5Core.5.15.13.dylib      	       0x1045e1094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
7   CoreFoundation                	       0x1982a63a8 __CFRunLoopDoSource0 + 176
8   CoreFoundation                	       0x1982a610c __CFRunLoopDoSources0 + 244
9   CoreFoundation                	       0x1982a4cf8 __CFRunLoopRun + 840
10  CoreFoundation                	       0x1982a4334 CFRunLoopRunSpecific + 572
11  HIToolbox                     	       0x1a36dd0cc RunCurrentEventLoopInMode + 292
12  HIToolbox                     	       0x1a36e2d1c ReceiveNextEventCommon + 220
13  HIToolbox                     	       0x1a36e3020 _BlockUntilNextEventMatchingListInModeWithFilter + 76
14  AppKit                        	       0x19bde8a70 _DPSNextEvent + 660
15  AppKit                        	       0x19c70e7b8 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 688
16  AppKit                        	       0x19c016920 -[NSApplication _doModalLoop:peek:] + 216
17  AppKit                        	       0x19c321708 __33-[NSApplication runModalSession:]_block_invoke_2 + 84
18  AppKit                        	       0x19c321698 __33-[NSApplication runModalSession:]_block_invoke + 112
19  AppKit                        	       0x19c014e28 _NSTryRunModal + 100
20  AppKit                        	       0x19c321568 -[NSApplication runModalSession:] + 148
21  libqcocoa.dylib               	       0x108f1c3ac 0x108ef0000 + 181164
22  libFreeCADGui.dylib           	       0x104b2405c Gui::Command::_invoke(int, bool) + 516
23  libFreeCADGui.dylib           	       0x104b23e1c Gui::Command::invoke(int, Gui::Command::TriggerSource) + 348
24  libQt5Core.5.15.13.dylib      	       0x1045fde84 0x1044b8000 + 1334916
25  libQt5Core.5.15.13.dylib      	       0x1045e1094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
26  CoreFoundation                	       0x1982a63a8 __CFRunLoopDoSource0 + 176
27  CoreFoundation                	       0x1982a610c __CFRunLoopDoSources0 + 244
28  CoreFoundation                	       0x1982a4cf8 __CFRunLoopRun + 840
29  CoreFoundation                	       0x1982a4334 CFRunLoopRunSpecific + 572
30  HIToolbox                     	       0x1a36dd0cc RunCurrentEventLoopInMode + 292
31  HIToolbox                     	       0x1a36e2d1c ReceiveNextEventCommon + 220
32  HIToolbox                     	       0x1a36e3020 _BlockUntilNextEventMatchingListInModeWithFilter + 76
33  AppKit                        	       0x19bde8a70 _DPSNextEvent + 660
34  AppKit                        	       0x19c70e7b8 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 688
35  AppKit                        	       0x19bddbb7c -[NSApplication run] + 480
36  libqcocoa.dylib               	       0x108f1c4d0 0x108ef0000 + 181456
37  freecad                       	       0x1025d7184 main + 5128
38  dyld                          	       0x197e3c274 start + 2840
```
[image]

JohnOCFII, 2024-10-12:
Seems OK for me.  Switched from OpenLight to FreeCAD Classic.  If the problem persists for the OP, perhaps rename `user.cfg` and `system.cfg` in the Configuration directory and try again.  https://wiki.freecad.org/Start_up_and_Configuration

```
OS: macOS 15.0.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38923 (Git)
Build type: Release
Branch: main
Hash: d20cb9e6ee198beb2bfd7e72d3dec0a575e3f28c
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * Curves 0.6.50
```
[image]

luzpaz, 2024-10-17:
cc @jrb0411

jrb0411, 2024-10-21:
I fresh installation seems to have corrected the issue.  I did choose the classic scheme for display.  Apologies for the delay in response. Hurricane Milton and all.  Thanks for the input.  Here is my current installation:

OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.48
  * ExplodedAssembly
  * Woodworking 0.21.2.33771

maxwxyz, 2024-10-21:
Thanks for the test! Closing this.

closed 2024-10-21 by maxwxyz